espressif/arduino-esp32 #1251: Question: Can I get IP of client connected to softAP?

lwoodbury, 2018-03-23:
I have been using a UDP (OSC) multicast on a Huzzah ESP32 with softAP connection, but experiencing a lot of packet loss. Seems multicast over Wifi is bad from what I have read so trying to do unicast, but I don't want to have to specify the IP of the remote device. Can I just get the IP of a client that connects to softAP somehow? 

There seems to be an 8266 example of how to do it here, but I can't get it to work on ESP32: [https://github.com/esp8266/Arduino/issues/2100](url)

Thanks!

moneypython, 2018-04-08:
Running into the same needs. Looks like ESP32 uses a custom fork of the ESP8266 wifi code, and is missing parts of the event system.

closed 2019-08-15 by stale[bot]

lwoodbury, 2019-08-16:
Well, I don't know if its been solved and I still need the functionality, can we keep it open or note the solution?

atanisoft, 2019-08-16:
@lwoodbury If/When this PR https://github.com/espressif/esp-idf/pull/3605 makes it to an esp-idf release it should be possible. Without this the IP is unfortunately not available.