espressif/arduino-esp32 #1252: EEPROMClass readString Bug

michygit, 2018-03-23:
Hi! Is this bug?

String EEPROMClass::readString (int address)
{
  if (address < 0 || address > _size)
    return String(0);

  uint16_t len;
  for (len = 0; len <= _size; len++)
    if (_data[address + len] == 0)
      break;

  if (address + len > _size)
    return String(0);

  char value[len + 1];
  memcpy((uint8_t*) value, _data + address, len);
  value[len + 1] = 0; <- ????
  return String(value);
}

baoshuai9527, 2018-03-24:
String terminator 0x00

jasonhillier, 2018-06-30:
*bump*

I was having an issue with EEPROM.readString() always putting an extra junk byte at the end. It looks like this is a bug!

Specifically, these lines:
Assume len=10, then
```
char value[len + 1]; //SIZE is 11
memcpy((uint8_t*) value, _data + address, len); //we READ 10 bytes
value[len + 1] = 0; <- //WAIT! we are writing the null terminator at index 11!, that is the 12th byte in the array! We are writing out-of-bounds of the array! this means value[len]=? because it is a byte from random memory!
```

PR #2439 (2019-02-14): Fixed missing null terminator in EEPROM.readString()
Greetings,

This patch corrects an issue with `EEPROM.readString(address, value, maxLen)` not terminating the read string. The resulting symptoms can be garbage characters when used.

A similar issue, #1608, corrected the overload function `EEPROM.readString(address)`, but did not address this function.

Additionally, this patch should close #1252.

Thanks,
-Kevin
closed 2019-02-14 by me-no-dev