espressif/arduino-esp32 #1253: A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('S')

jfornech, 2018-03-23:
### Hardware:
Board:							?Widora Air V6?
Core Installation/update date:			?20/mar/2018?
IDE name:						?Arduino IDE? ?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('S')

### Sketch:
```cpp
*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```
Please, how can i fix this error ?
Thanks

beegee-tokyo, 2018-03-24:
Looks like your computer cannot see the ESP32
- Are you on MAC? Check [Wiki](http://desire.giesecke.tk/index.php/2018/01/30/esp32-not-showing-in-serial-port-mac-desktop-laptop-problem/)
- Do you have the proper USB driver installed? CP2104 is the driver you need to look for [Silabs drivers](https://www.silabs.com/products/development-tools/software/usb-to-uart-bridge-vcp-drivers)
- Did you try different USB cable or different USB port?

glkeeney1951, 2018-04-04:
I got the same error with and Espressif ESP32-WROOM-32. Device Manager shows the Silicon Labs CP210x USB to UART Bridge (Com15) for my port.
The sketch verifies, but does not Upload to the board. Changing the USB port to 4 gets different error.

jfornech, 2018-04-10:
Hello,
I fixed it by pressing "Load" button before switch the ESP32 on. The Rx/Tx leds stop to blink, so the serial port was no more use by the ESP. I tried to launch a connection, and it works .

closed 2018-04-10 by jfornech

tlvlp, 2018-09-19:
Hi,

Although the issue is closed and the workaround works for me as well the problem still persists and I think that it could be related to writing to the Serial when the upload starts.

eg. in the above example "**Invalid head of packet ('S')**" might be coming from:
```
void setup() {
   ...
    Serial.println("Setup done");
}
```

In my case the message is "**Invalid head of packet ('{')**" and I'm using the aREST library that sends a standard json resposne to every serial input and it starts with '{'

Hopefully someone more knowledgeable than me can find a permanent solution.

Speshal, 2018-10-25:
> Hi,
>
> Although the issue is closed and the workaround works for me as well the problem still persists and I think that it could be related to writing to the Serial when the upload starts.
>
> eg. in the [...]

I just got my first ESP32 (ESP32 Dev board), had the same issue, held down the boot button and my sketch uploaded fine.

Huong-nt, 2018-10-26:
I fixed by put ESP32-lyrat board into upload mode:
https://docs.espressif.com/projects/esp-adf/en/latest/get-started/index.html#upload

tlvlp, 2018-10-27:
Yes as I've mentioned this method works for me as well but it makes it impossible to use OTA update with the device as you have to be physically present, pressing the button for the update to be able to run..

sharatkanthi, 2018-11-02:
Press the "Boot" button on your ESP, Start uploading your code from Arduino IDE and keep "Boot" pressed till upload completes.

tlvlp, 2018-11-02:
yes that's what the workaround is mentioned many times before, but in case the OTA (Over The Air) updates you **can not physically access the device**.. that is the whole point of it..

programonauta, 2019-01-30:
Hi,, I had the same problem. I really spent a bunch of time to realize that the problem was I didn't close the serial monitor. I'm using Platform.IO, but I imagine the problem could be similar.

lukile, 2019-04-27:
Hello there! 
I had the same problem and I just found out it was because my upload speed (which is in "Tools") was set on 921600 instead of 115200.
Hope it would help :)

Ghaith111, 2019-06-28:
hello everyone,
im an interne at a lab and i m new to programing. my tutor game me the M5stack to work on with the arduino IDE. i saw in a video that i have to install the m5stack libraries which i did. 
when i go to "examples" to lunch "factory test" to be able to use the other examples on the arduino, i had a problem : 
A fatal error occurred: Invalid head of packet (0xF0)
Can someone pls help ?
thank you !

enayetz, 2020-01-06:
> Hello,
> I fixed it by pressing "Load" button before switch the ESP32 on. The Rx/Tx leds stop to blink, so the serial port was no more use by the ESP. I tried to launch a connection, and it works .

@jfornech 👍 Thank you so much. It works now.

cskang510, 2020-02-19:
I had the same issue with ESP WROOM32.
Thanks for your tip.
Holding BOOT button fixed it.

rajatsaini736, 2020-03-01:
hello every one.. i am in final year of CSE.  I have my major project in IOT based Smart Waste Management System.
i have uno board, esp, breadboard, jump wires....and installed esp libraries in arduino ide.
But when i upload the code while selecting tools->Board->generic ESP8266.. it gives me the error ->esptool.FatalError: Failed to connect to ESP8266: Invalid head of packet (0xF0)

although in my code there are two empty functions setup() and loop()..

Please help me.  thanx in advance

lbernstone, 2020-03-01:
This forum is for esp32.  [esp8266](https://github.com/esp8266/Arduino) is a different piece of hardware.

rajatsaini736, 2020-03-01:
can you plz tell me what should i do..

sharatkanthi, 2020-03-02:
Hold on "Flash" button on ESP8266 while you are uploading your code

rajatsaini736, 2020-03-02:
there are only a reset button on ESP8266 board.... I don't see any flash button on this board

PabloCastellano, 2020-04-09:
My upload speed was 921600. After changing upload speed to 115200 it worked for me.

andreaslink-de, 2020-04-14:
> My upload speed was 921600. After changing upload speed to 115200 it worked for me.

Cool, that solved it for me as well, incredible, I never thought, faster is better :-)! Thanks a lot for this hint!

FasilNegash, 2020-05-21:
> 
> 
> Press the "Boot" button on your ESP, Start uploading your code from Arduino IDE and keep "Boot" pressed till upload completes.

Thank you, this solved my problem

iftoz, 2020-06-12:
hi is there any pdate solutions to fix this issue beside holding the Boot button solutions, ?

madison1776, 2020-07-09:
holding the boot button on ESP32 on uploading code until code loaded solved problem.

utkutpcgl, 2020-08-25:
Holding boot was not sufficient for my case (ESP Lyrat). I had to keep pressing Boot while pressing "RST" button once. I have not seen any other resource that suggested this, I guess I am lucky.

iftoz, 2020-08-25:
I have found the solutions from another blog you can solder some appropriate capacitor arround 10nF or above between RST and GND pin for the detail you can check randomnerdtutorials

feriniguez, 2020-09-03:
> I have found the solutions from another blog you can solder some appropriate capacitor arround 10nF or above between RST and GND pin for the detail you can check randomnerdtutorials

Could you please paste the link?

iftoz, 2020-09-03:
> > I have found the solutions from another blog you can solder some appropriate capacitor arround 10nF or above between RST and GND pin for the detail you can check randomnerdtutorials
> 
> Could you please paste the link?

Here you go
https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/
Also look at the comment section there are some other guys that trying different capacitor value as well

alam121, 2020-10-26:
I had the same problem. For me when I press upload onthe IDE and get the message of connecting, I remove and put back the VCC and it show connecting. It's likely that the same effect would also be if I unplug and plug the rest wire, but I have not tried that.

My connections:

ESP12E Chip                      FTDI
VCC, CH_PD                       3.3V
GRND, GPIO15, GPIO0      GRND
TX                                       RX
RX                                       TX

I remove the GPIO15 and GPIO0, once the upload is done.

VonHirsch, 2021-05-17:
Bad cable here

Sphaaax, 2021-06-20:
Hi everyone,

I am new to esp32 and I spent several hours trying to solve this. I finally did, thanks for your help !
I also used hardware fix on my device see this link : https://randomnerdtutorials.com/solved-failed-to-connect-to-esp32-timed-out-waiting-for-packet-header/?unapproved=632125&moderation-hash=ba430aa0e9cb5de3e2b8ded4f639124d#comment-632125

It seems that every esp32 devices are different so here is my configuration and solution.

I use a NodeMCU Joy-It equipped with the esp32-wroom-32, with esp-idf.
– Bought here : https://www.conrad.fr/p/carte-de-developpement-joy-it-node-mcu-esp32-modul-sbc-nodemcu-esp32-1-pcs-1656367
– Doc here : https://joy-it.net/en/products/SBC-NodeMCU-ESP32

--------------------------------------------------------------------
A) In menuconfig (idf.py menuconfig) —> Serial flasher config :

[ ] Disable download stub
Flash SPI mode (QIO) —>
Flash SPI speed (40 MHz) —>
Flash size (4 MB) —>
[*] Detect flash size when flashing bootloader
Before flashing (Reset to bootloader) —>
After flashing (Reset after flashing) —>
‘idf.py monitor’ baud rate (115200 bps) —>

--------------------------------------------------------------------
B) Add a 47µF between pins “EN” and “GND”.

--------------------------------------------------------------------
C) Before executing “idf.py -p /dev/ttyUSB0 flash”, I have to do this sequence :
1) Press and hold “Boot”
2) Press and release “En”
3) Release “Boot”
After flashing I finally was able to see those logs :
-----
Wrote 163472 bytes (86248 compressed) at 0x00010000 in 2.2 seconds (effective 598.2 kbit/s)…
Hash of data verified.
Compressed 3072 bytes to 103…
Writing at 0x00008000… (100 %)
Wrote 3072 bytes (103 compressed) at 0x00008000 in 0.0 seconds (effective 547.3 kbit/s)…
Hash of data verified.

Leaving…
Hard resetting via RTS pin…

Done
-----

Before all this I obtained following errors :
1) A fatal error occurred: Failed to connect to ESP32: Invalid head of packet (0xZZ)” with 0xZZ random hex value
2) A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header

Hope this will help someone 👍🏽

TLINDEN, 2022-01-01:
I had the same problem with a Wemos lolin d32 pro board. It doesn't have a dedicated EN button. So the trick here is to connect `Port 0` with `GND`, then execute `idf -p /dev/ttyUSB0 flash`. Then disconnect `Port 0` and reset.

snbuch, 2022-02-20:
I had the same issue on a Wemos D1 R32 board. The issue started when I added a Serial.Println into the loop with too small a delay. Finally found IO0 on the board and connected it to GND and then programming went thru. Look for IO0 pin next to GND and VCC. Also avoid adding too many Serial.Println

Transcendence1, 2022-02-21:
I had the same problem with esp32 dev kit. After using another usb to serial converter and connect its outputs to RX0 and TX0 of the board the problem is solved. 
hope it helps

IPlayZed, 2022-03-12:
I use the Devkit-C board and for some reason on my laptop I do not have to press the Boot button. But on my PC I have to, otherwise when using the Arduino IDE, it gives my this error.
I run Arch on both of them.

Fusseldieb, 2022-05-26:
After 2 hours of suffering with my "ESP32Cam 'AI-Thinker'", I finally found it out!

Pressing **the RST button does not suffice**. It just resets it, but does NOT start download mode.

In order to enter download/flash mode, **BRIDGE IO0 to GND**, connect it to your FTDI and try to flash. In some cases that combined with pressing RST makes it flash.

samueldanda, 2022-07-31:
Had the problem and found a solution

Pressing and holding RST/BOOT pins did not work, upload speed was already 115200, and soldering a capacity on the ESP did not click on my head since the board was uploading just fine few seconds before

It turns out the esp was not receiving a stable power suply and changing the power source worked for me
https://docs.espressif.com/projects/esptool/en/latest/esp8266/troubleshooting.html#bootloader-won-t-respond

It is very frustrating when you come accross such errors while fighting a deadline

IPlayZed, 2022-07-31:
The issue is closed but it I think that it should be investigated because depending on the USB hub and cable some devices flash while others don't.
It may be a question of the breakout board as well, but if it is the module's fault then it could be a security vulnerability as well: an unauthorized person might flash a new binary onto the chip if the needed precautions are not taken, thus executing malicious code.

QGB, 2022-11-21:
```
In [7]: !esptool --chip esp32s3 --port {COM} erase_flash
esptool.py v3.2
Serial port COM5
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: Invalid head of packet (0x42): Possible serial noise or corruption.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting

In [8]: !esptool --chip esp32s3 --port {COM} erase_flash
esptool.py v3.2
Serial port COM5
Connecting......................................

A fatal error occurred: Failed to connect to ESP32-S3: No serial data received.
For troubleshooting steps visit: https://github.com/espressif/esptool#troubleshooting

```

QGB, 2022-11-21:
```
ESP-ROM:ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0x1 (POWERON),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x404
load:0x403b6000,len:0x6ec
load:0x403ba000,len:0x26b4
entry 0x403b6138
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x404
load:0x403b6000,len:0x6ec
load:0x403ba000,len:0x26b4
entry 0x403b6138
ESP-ROM:esp32s3-20210327
Build:Mar 27 2021
rst:0xf (BROWNOUT_RST),boot:0x8 (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd0108,len:0x404
load:0x403b6000,len:0x6ec
load:0x403ba000,len:0x26b4
entry 0x403b6138

################################################

arch:ESP32S3,0
company:Ai-Thinker|B&T
wifi_mac:F412FA41649C
ble_mac:F412FA41649E
sdk_version:v4.4-dev-3042-g220590d599-dirty
firmware_version:1.0.1
compile_time:Feb  8 2022 14:37:32

ready

################################################
' not found
' not found
' not found
```

Blueline-lab, 2022-12-23:
Hello Guys, I read your posts because i had the same errors with an ESP8266(ESP01) and an USB/UART converter.

I finnaly find the solution and Upload my files on by using this tuto : https://www.diyhobi.com/flash-program-esp-01-using-usb-serial-adapter/

Have a good day and work.

laoadam, 2023-05-16:
Where can I have an issues number list?

kiranshashiny, 2023-09-15:
> Hello, I fixed it by pressing "Load" button before switch the ESP32 on. The Rx/Tx leds stop to blink, so the serial port was no more use by the ESP. I tried to launch a connection, and it works .

"Load button" - Do you mean the RST button on the ESP32-WebCAM. ?

rasim065, 2024-03-15:
The best solution is to disconnect all power pins of external devices, apply external 5V for ESP32 and 
1) Press and hold “Boot”
2) Press and release “En”
3) Release “Boot”

jorgecgjr, 2024-04-04:
> Press the "Boot" button on your ESP, Start uploading your code from Arduino IDE and keep "Boot" pressed till upload completes.

Love ya budy

HarshaR1642, 2024-06-03:
> My upload speed was 921600. After changing upload speed to 115200 it worked for me.

Thanks a lot. This worked for me.

Tcoton, 2024-06-06:
I have been fighting to get anything uploaded using my iMac M3 and a CallDigit TS4 until I tried to connect the cable directly to an iMac USB port instead of using any of the CallDigit TS4 USB ports. I also set the upload speed to the correct speed 115200, correct frequency for my ESP WROOM 32 and it works. 

Sequence for successful upload: 
1- Click the upload button in Arduino IDE
2- Press and hold boot
3- While keeping a hold on boot, quick press EN to reset
4- Wait for the Connecting...... message in console
5- Release Boot

In case of error, retry from 1.

khangdc2, 2024-12-29:
tool -> upload speed change to 115200