espressif/arduino-esp32 #1255: Arduino WiFi connection example stuck on Connecting....... for capital-letter networks

sdaitzman, 2018-03-25:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32-DEVKITC-V2 (Core Board V2)
Core Installation/update date:			March 23, 2018 (I think)
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
When I run the example code (attached below) the ESP32 is never able to connect to WiFi. It succeeded twice, including once with my phone hotspot, but I have tried two other networks without success. The SSID and password have been changed, as you can see. The two networks that failed both had capital letters at the beginning. Any possible ideas?

### Sketch:
```cpp
#include <WiFi.h>

const char* ssid     = "HIDDENSSIDTHATISMYNETWORK";
const char* password = "PASSWORDFORMYNETWORK";

WiFiServer server(80);

void setup()
{
    Serial.begin(115200);
    pinMode(13, OUTPUT);      // set the LED pin mode

    delay(10);

    // We start by connecting to a WiFi network

    Serial.println();
    Serial.println();
    Serial.print("Connecting to ");
    Serial.println(ssid);

//    WiFi.enableSTA(true);
    WiFi.begin(ssid, password);

    while (WiFi.status() != WL_CONNECTED) {
        delay(500);
        Serial.print(".");
    }

    Serial.println("");
    Serial.println("WiFi connected.");
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP()); 
    
    server.begin();

}

int value = 0;

void loop(){
 WiFiClient client = server.available();   // listen for incoming clients

  if (client) {                             // if you get a client,
    Serial.println("New Client.");           // print a message out the serial port
    String currentLine = "";                // make a String to hold incoming data from the client
    while (client.connected()) {            // loop while the client's connected
      if (client.available()) {             // if there's bytes to read from the client,
        char c = client.read();             // read a byte, then
        Serial.write(c);                    // print it out the serial monitor
        if (c == '\n') {                    // if the byte is a newline character

          // if the current line is blank, you got two newline characters in a row.
          // that's the end of the client HTTP request, so send a response:
          if (currentLine.length() == 0) {
            // HTTP headers always start with a response code (e.g. HTTP/1.1 200 OK)
            // and a content-type so the client knows what's coming, then a blank line:
            client.println("HTTP/1.1 200 OK");
            client.println("Content-type:text/html");
            client.println();

            // the content of the HTTP response follows the header:
            client.print("Click <a href=\"/H\">here</a> to turn the LED on pin 5 on.<br>");
            client.print("Click <a href=\"/L\">here</a> to turn the LED on pin 5 off.<br>");

            // The HTTP response ends with another blank line:
            client.println();
            // break out of the while loop:
            break;
          } else {    // if you got a newline, then clear currentLine:
            currentLine = "";
          }
        } else if (c != '\r') {  // if you got anything else but a carriage return character,
          currentLine += c;      // add it to the end of the currentLine
        }

        // Check to see if the client request was "GET /H" or "GET /L":
        if (currentLine.endsWith("GET /H")) {
          digitalWrite(13, HIGH);               // GET /H turns the LED on
        }
        if (currentLine.endsWith("GET /L")) {
          digitalWrite(13, LOW);                // GET /L turns the LED off
        }
      }
    }
    // close the connection:
    client.stop();
    Serial.println("Client Disconnected.");
  }
}
```

### Debug Messages:
```
Connecting to <MY NETWORK NAME WAS HERE>
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
....[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:298] _eventCallback(): Reason: 201 - NO_AP_FOUND
............................................................................................................
```

stickbreaker, 2018-03-25:
@sdaitzman 
run the Scan example sketch, 
```c++
/*
 *  This sketch demonstrates how to scan WiFi networks.
 *  The API is almost the same as with the WiFi Shield library,
 *  the most obvious difference being the different file you need to include:
 */
#include "WiFi.h"

void setup()
{
    Serial.begin(115200);

    // Set WiFi to station mode and disconnect from an AP if it was previously connected
    WiFi.mode(WIFI_STA);
    WiFi.disconnect();
    delay(100);

    Serial.println("Setup done");
}

void loop()
{
    Serial.println("scan start");

    // WiFi.scanNetworks will return the number of networks found
    int n = WiFi.scanNetworks();
    Serial.println("scan done");
    if (n == 0) {
        Serial.println("no networks found");
    } else {
        Serial.print(n);
        Serial.println(" networks found");
        for (int i = 0; i < n; ++i) {
            // Print SSID and RSSI for each network found
            Serial.print(i + 1);
            Serial.print(": ");
            Serial.print(WiFi.SSID(i));
            Serial.print(" (");
            Serial.print(WiFi.RSSI(i));
            Serial.print(")");
            Serial.println((WiFi.encryptionType(i) == WIFI_AUTH_OPEN)?" ":"*");
            delay(10);
        }
    }
    Serial.println("");

    // Wait a bit before scanning again
    delay(5000);
}
```
See what AP's it can see.

Chuck.

sdaitzman, 2018-03-25:
I think someone else who administers the network changed the WiFi network names around, and several networks that were previously 2.4GHz are now 5GHz. Upon further research, I see that the ESP32 does not work with 5GHz networks by default. Any idea if there is a workaround to use ESP32 with 5GHz networks?

sdaitzman, 2018-03-25:
And thank you so much for the help, @stickbreaker !

closed 2018-03-25 by sdaitzman

KhanKudo, 2019-07-01:
by my knowledge supporting or not supporting 5GHz networks is a hardware thing and cannot be changed. sadly however I too was scratching my head as to why it wasn't working when I came up with the 5GHz idea and this article confirms it. thanks alot.