espressif/arduino-esp32 #1256: How many PWM channels for ESP32 using Arduino?

danlu01, 2018-03-25:
### Hardware:
Board:							 ESP32 Dev board
Core Installation/update date:			Latest of 3/23/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						?115200?

### Description:
In the ESP32 example for LEDCSoftwareFade, there's a comment hinting there are 16 LEDC_Channel starting from 0.  However, any PWM assigned to channel over 8 does not work.  When I search for LEDC_CHANNEL, ledc.h shows only 0-7 channels.  What is the max number of PWM available?  If more than 8, how do enable more than 8?

### Sketch:
     // use first channel of 16 channels (started from zero)
     #define LEDC_CHANNEL_0     0

//Change the code below by your sketch
see ESP32 example LEDCSoftwareFade.ino

lbernstone, 2018-03-25:
There's 16 hardware channels, but only 8 timers.  I think to simplify things the arduino code only exposes 8 channels.  If you want more, you can directly access them (and micromanage them) using the IDF functions.

danlu01, 2018-03-25:
Is there an example/tutorial on how this is done?

lbernstone, 2018-03-26:
https://github.com/espressif/esp-idf/tree/master/examples/peripherals/ledc . If you don't need more than 8 frequencies, you could also modify the arduino library.

danlu01, 2018-03-26:
My mistake...  I got it working now.

closed 2018-03-26 by danlu01