espressif/arduino-esp32 #1257: Wakeup mode

404-baitnotfound, 2018-03-25:
I would need to the chip to wakeup on LOW on three seperat pins. 
Defining three esp_sleep_enable_ext0_wakeup does not work.
And using a Bitmask and esp_sleep_enable_ext1_wakeup only gives me two options:

```
typedef enum {
    ESP_EXT1_WAKEUP_ALL_LOW = 0,    //!< Wake the chip when all selected GPIOs go low
    ESP_EXT1_WAKEUP_ANY_HIGH = 1    //!< Wake the chip when any of the selected GPIOs go high
} esp_sleep_ext1_wakeup_mode_t;
```

I would need "ESP_EXT1_WAKEUP_ANY_LOW".
Is this technical not possible or was this just omitted??

closed 2019-08-15 by stale[bot]

jgcjoec, 2020-12-03:
I was researching this today via Google searches and this unanswered question came up. I've run into this before, and used EXT0 + EXT1 to overcome this limitation, but I too, am curious - was the absence of a "ESP_EXT1_WAKEUP_ANY_LOW" just an omission or is there a technical/hardware challenge preventing its implementation?

RafigRzayev, 2021-01-13:
Having same problem here. But it is interface of esp-idf, I guess should be changed there. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/system/sleep_modes.html

rosmianto, 2023-11-16:
ESP32 doesn't support `ESP_EXT1_WAKEUP_ANY_LOW`. It is supported by other chips though.

https://github.com/espressif/esp-idf/blob/c8243465e45489835d645bf217a6929fd0c01b7f/components/esp_hw_support/include/esp_sleep.h#L37