espressif/arduino-esp32 #1258: ledc interferes with soft ap

shaurya0406, 2018-03-25:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Devkit V1
Core Installation/update date:			23/3/18
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						921600

### Description:
reboot on motor command

### Sketch:
#include <ESPUI.h>
#include <TB6612FnG.h>
#include "WiFi.h"

Tb6612fng motors(27, 14, 12, 13, 26, 25, 33);

char* ssid = "ESPUI";
char* pass = "";
long oldTime = 0;
bool switchi = false;

void slider(Control sender, int type) {
  Serial.println(sender.value);
}

void buttonCallback(Control sender, int type) {
  switch (type) {
  case B_DOWN:
    Serial.println("Button DOWN");
    break;
  case B_UP:
    Serial.println("Button UP");
    break;
  }
}

void buttonExample(Control sender, int type) {
  switch (type) {
  case B_DOWN:
    Serial.println("Status: Start");
    ESPUI.print(0, "Status: Start");
    break;
  case B_UP:
    Serial.println("Status: Stop");
    ESPUI.print(0, "Status: Stop");
    break;
  }
}
void padExample(Control sender, int value) {
  switch (value) {
  case P_LEFT_DOWN:
    Serial.print("left down");
    motors.drive(-1,1);
    break;
  case P_LEFT_UP:
    Serial.print("left up");
    motors.drive(0);
    break;
  case P_RIGHT_DOWN:
    Serial.print("right down");
    motors.drive(1,-1);
    break;
  case P_RIGHT_UP:
    Serial.print("right up");
    motors.drive(0);
    break;
  case P_FOR_DOWN:
    Serial.print("for down");
    motors.drive(1);
    break;
  case P_FOR_UP:
    Serial.print("for up");
    motors.drive(0);
    break;
  case P_BACK_DOWN:
    Serial.print("back down");
    motors.drive(-1);
    break;
  case P_BACK_UP:
    Serial.print("back up");
    motors.drive(0);
    break;
  case P_CENTER_DOWN:
    Serial.print("center down");
    motors.drive(0);
    break;
  case P_CENTER_UP:
    Serial.print("center up");
    motors.drive(0);
    break;
  }
  Serial.print(" ");
  Serial.println(sender.id);
}

void switchExample(Control sender, int value) {
  switch (value) {
  case S_ACTIVE:
    Serial.print("Active:");
    digitalWrite(23,HIGH);
    break;
  case S_INACTIVE:
    Serial.print("Inactive");
    digitalWrite(23,LOW);
    break;
  }
  Serial.print(" ");
  Serial.println(sender.id);
}

void otherSwitchExample(Control sender, int value) {
  switch (value) {
  case S_ACTIVE:
    Serial.print("Active:");
    break;
  case S_INACTIVE:
    Serial.print("Inactive");
    break;
  }
  Serial.print(" ");
  Serial.println(sender.id);
}

void setup(void) {
  Serial.begin(115200);
  pinMode(23,OUTPUT);
  digitalWrite(23,LOW);
  WiFi.mode(WIFI_AP);
WiFi.softAPsetHostname(ssid);
  #if defined(ESP32)
    WiFi.setHostname(ssid);
  #else
    WiFi.hostname(ssid);
  #endif

  WiFi.softAP(ssid);
// WiFi.softAP(ssid,pass);
  Serial.println("");
  Serial.print("IP address: ");
  Serial.println(WiFi.softAPIP());

  // change the beginning to this if you want to join an existing network
  /*
    Serial.begin(115200);
    WiFi.begin(ssid, password);
    Serial.println("");
    // Wait for connection
    while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
    }
    Serial.println("");
    Serial.print("IP address: ");
    Serial.println(WiFi.localIP());
  */

  ESPUI.label("Status:", COLOR_TURQUOISE, "Stop");
  ESPUI.label("Millis:", COLOR_EMERALD, "0");
  ESPUI.button("Push Button", &buttonCallback, COLOR_PETERRIVER);
  ESPUI.button("Other Button", &buttonExample, COLOR_WETASPHALT, "Press");
  ESPUI.pad("Pad with center", true, &padExample, COLOR_SUNFLOWER);
  ESPUI.pad("Pad without center", false, &padExample, COLOR_CARROT);
  ESPUI.switcher("Switch one", false, &switchExample, COLOR_ALIZARIN);
  ESPUI.switcher("Switch two", true, &otherSwitchExample, COLOR_NONE);
  ESPUI.slider("Slider one", &slider, COLOR_ALIZARIN, "30");
  ESPUI.slider("Slider two", &slider, COLOR_NONE, "100");

  ESPUI.begin("SHAURGUI by SHAURYA CHANDRA");
}

void loop(void) {
  if (millis() - oldTime > 5000) {
    ESPUI.print("Millis:", String(millis()));
//    switchi = !switchi;
//    ESPUI.updateSwitcher("Switch one", switchi);
    oldTime = millis();
  }
}
```

### Debug Messages:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
ets_main.c 371 
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

IP address: 192.168.4.1
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 13 - AP_START
Listing directory: /
  FILE: /index.htm  SIZE: 1420
  FILE: /css/style.css  SIZE: 11728
  FILE: /css/normalize.css  SIZE: 1948
  FILE: /js/zepto.min.js  SIZE: 26392
  FILE: /js/controls.js  SIZE: 5925
  FILE: /js/slider.js  SIZE: 1944
UI Initialized
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 15 - AP_STACONNECTED
Connected: 1
JSON Data Sent to Client!
Disconnected!
Connected: 2
JSON Data Sent to Client!
for down/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40085229 on core 1

Backtrace: 0x40087fa0:0x3ffde060 0x4008809f:0x3ffde080 0x40085229:0x3ffde0a0 0x400de177:0x3ffde0e0 0x400ddf82:0x3ffde100 0x400ddfe3:0x3ffde120 0x400de116:0x3ffde150 0x400d1fe1:0x3ffde180 0x400d2086:0x3ffde1a0 0x400d3d1d:0x3ffde1c0 0x40137353:0x3ffde4b0 0x401377d2:0x3ffde4d0 0x400d7ced:0x3ffde520 0x400d7cfd:0x3ffde550 0x400d62f5:0x3ffde570 0x400d6545:0x3ffde5b0

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58

IP address: 192.168.4.1
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 13 - AP_START
Listing directory: /
  FILE: /index.htm  SIZE: 1420
  FILE: /css/style.css  SIZE: 11728
  FILE: /css/normalize.css  SIZE: 1948
  FILE: /js/zepto.min.js  SIZE: 26392
  FILE: /js/controls.js  SIZE: 5925
  FILE: /js/slider.js  SIZE: 1944
UI Initialized
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 15 - AP_STACONNECTED
Connected: 1
JSON Data Sent to Client!
for down/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40085229 on core 1

Backtrace: 0x40087fa0:0x3ffde060 0x4008809f:0x3ffde080 0x40085229:0x3ffde0a0 0x400de177:0x3ffde0e0 0x400ddf82:0x3ffde100 0x400ddfe3:0x3ffde120 0x400de116:0x3ffde150 0x400d1fe1:0x3ffde180 0x400d2086:0x3ffde1a0 0x400d3d1d:0x3ffde1c0 0x40137353:0x3ffde4b0 0x401377d2:0x3ffde4d0 0x400d7ced:0x3ffde520 0x400d7cfd:0x3ffde550 0x400d62f5:0x3ffde570 0x400d6545:0x3ffde5b0

Rebooting...

```

libraries used :

https://github.com/vincasmiliunas/ESP32-Arduino-TB6612FNG

https://github.com/s00500/ESPUI

lbernstone, 2018-03-25:
Sounds like a power issue.  Are you trying to power the motors from the ESP32?  What sort of power supply are you using?

shaurya0406, 2018-03-25:
I have not connected the motor driver and the motors

I was just testing and the setup contains only the esp32

shaurya0406, 2018-03-25:
Esp32 is connected to my laptop

closed 2019-08-15 by stale[bot]