espressif/arduino-esp32 #1259: Hostname and DNS

404-baitnotfound, 2018-03-25:
### Description:
I would like to conect to my esp32 via Browser not by entering  its IP Adresse nur by using the hostname.
My router does the DHCP and DNS and the same ist working with my printer.
I already used WIFI.setHostname(NAME) and my router does see the  esp32 with its correct name. However I can't find it with ping NAME . 
Also I tryed to add MDNS service with MDNS.setHostname  and IT didn't Work. 
Am I doing something wrong or is  this a problem with my router?
I will post  my code when I am back at my PC.

polygontwist, 2018-03-25:
`  WiFi.mode(WIFI_STA);
WiFi.begin(ssid, password);
WiFi.setHostname(HOSTNAME);
while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(5000);
    ESP.restart();
  }
`
it works for me. set the setHostname() after begin().

404-baitnotfound, 2018-03-25:
ok found the error: with using "ping -a <esp_IP>" I got the real hostname under which the esp is prodcasting.

So it was <hostname>.home

Would be nice if this was mentioned in the lib........

closed 2018-03-25 by 404-baitnotfound