espressif/arduino-esp32 #1261: Need IPv6 support

owendelong, 2018-03-26:
This is a feature request.

The current shipping LWIP and ESP-IDF appear to be able to support IPv6, though it apparently implements only SLAAC and static addressing, but not DHCPv6. If we can get that functionality, it would be sufficient for my needs.

Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ESP32 Dev Module? -- Applies to any hardware
Core Installation/update date:			23 Mar 2018
IDE name:							Arduino IDE 1.8.5
Flash Frequency:					        Applies to any frequency
Upload Speed:						Applies to any upload speed

### Description:
Need IPv6 support in Networking stack

https://github.com/Heltec-Aaron-Lee/WiFi_Kit_series contains some ability to set up an IPv6 Link Local address on the WLAN interface, but doesn't get an address from SLAAC or do any DHCPv6 processing. There does not appear to be an interface to assign a static IPv6 address.

It's not clear if there are any abilities to do any form of getaddrinfo(), getnameinfo(), or TCP, UDP, or ICMP IPv6 packet communication using the library.

Unfortunately the documentation (mostly in Chinese) is far from complete over there and the documentation of the IPv6 example is virtually non-existent and appears to be centered on providing some form of IPv6 minimal capability for an ESP32 acting as a WiFi Station and AP simultaneously.

martin-ger, 2018-03-28:
Try to call "netif_create_ip6_linklocal_address(sta-netif, 0);" with the appropriate netif once you are connected to WiFi. 

At least on an ESP8266 with FreeRTOS this made IPv6 autoconfig work...

See also: https://github.com/IPv6-ESP8266/IPv6-ESP8266

owendelong, 2018-03-28:
Thanks... Will let you know how that goes.

owendelong, 2018-03-28:
Hmmm... Perhaps FreeRTOS and Arduino are orthogonal. At least I'm not finding that call in the Arduino libraries anywhere. Is this limited to the ESP-IDF and not the Arduino environment?

martin-ger, 2018-03-28:
It is in lwip/netif.h

owendelong, 2018-03-28:
OK, looking a little further into this, it looks like the Arduino level support is a bit incomplete but I got further than before.

Using the WiFi class in Arduino and using the following code:

```
const char *ssid = "SSID";
const char *password = "network_password";
const char *host = "jimtest.delong.com";

  WiFi.begin(ssid, password);
  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
  Serial.println(" CONNECTED");
  WiFi.enableIpV6();
  
  delay(2000);
  Serial.print("Local IPv4: ");
  Serial.println(WiFi.localIP());
  Serial.print("Local IPv6: ");
  Serial.println(WiFi.localIPv6());
  Serial.print("Connecting to ");
  Serial.print(host);

  WiFiClient client;
  const int httpPort = 80;
  if (!client.connect(host, httpPort)) // v6 experiment
  {
    Serial.println(" Connection Failed");
  }
  else
  {
    Serial.println(" SUCCESS!!");
  }
  Serial.print("Connecting to IPv4 ");
  Serial.print("www.delong.com");
  if (!client.connect("www.delong.com", 80)) // v4 Control
  {
    Serial.println(" Connection Failed");
  }
  else
  {
    Serial.println(" SUCCESS!!");
  }

```

I was able to get the IPv6 Link Local address of the interface (which I got before).
However, using the same suffix from the IPv6 Link Local address, I tried pinging the ESP32 on the local network (2620:0:930::/64 in this case).

So, for example, with link local of fe80::32ae:a4ff:fe55:ca40, on network 2620:0:930::/64 I was able to successfully ping 2620:0:930::32ae:a4ff:fe55:ca40.

Now, the two hosts in the client.connect() calls are as follows:

jimtest.delong.com -- IPv6 only web server at 2620:0:930::dead:beef:cafe
www.delong.com -- Dualstack web server at 192.159.10.7 and 2620:0:930::400:7

The ESP32 does do AAAA and A resolution (in fact, it asks for AAAA first). However, it does not send any SYN packets or any packets at all to 2620:0:930::dead:beef:cafe. About 5 seconds later, it reports "connection failed". It then sends no packets to 2620:0:930::400:7, does do a three-way handshake with 192.159.10.7 and then reports SUCCESS!!

So it appears that all that is missing in terms of getting minimal workable IPv6 TCP as a client is putting the hooks into the Arduino <Interface_Class>.connect() call to:

1. Understand the AF_INET6 returns from getaddrinfo() (if they aren't already there.
2. Set up an AF_INET6 socket based on the above (LWIP socket() call?)
3. Connect the socket to the remote host (LWIP connect() call?)
4. Possibly AF_INET6 exception handling.

Desirable features would include adding hooks to get ALL of the IPv6 addresses instead of just the link_local_ip. (IMHO, we should perhaps make loclaIPv6 a subclass so that localIPv6() returns what we currently get, the link local, but we could also call localIPv6.link_local() and get an array of structures containing link local information (e.g. struct IPv6AddressInformation { uint_128t address; uint_8t length; }), adding the relevant localIPv6.global(), localIPv6.nameservers(), localIPv6.gateway() calls to retrieve similar arrays of information for those categories. Perhaps even a call to localIPv6.addresses() which would yield a larger structure including not only what I listed above, but also char *zone_id, and uint_8t scope to the structure giving more complete information.

I'm just thinking out loud here, so feedback as to whether I'm on the right track or not would be helpful.

owendelong, 2018-04-10:
If I can get guidance from one of the maintainers on the above, I'm willing to work on the code...

1. Is my idea for creating additional localIPv6.x() methods to obtain more complete address information a reasonable approach?
2. Are there preferred or standardized names for such functions?
3. Is my idea to return a pointer to an array of structures containing address information reasonable?

In the meantime, I'll take a look at WiFi.connect() and see if I can get that going for starters.

xaque208, 2019-02-26:
I too would like to see support for IPv6 on the esp32 platform.  Ideally, with SLAAC router announcements, though even static address would be a start.  All the components that I'd like to connect to the ESP boards are running IPv6 only, and so I'd like to be able to connect the ESP to those components using v6.

me-no-dev, 2019-02-27:
Maybe this should be asked in IDF's repo? Arduino here supports IPv6 as much as IDF does

owendelong, 2019-04-08:
I'll note that the ESP8266 Arduino project https://github.com/esp8266/Arduino now has IPv6 support. They call it experimental, but so far, it's working well for me. However, since I need to do some SSL things, it would be nice to be able to use the faster processor. Perhaps the code work done there could be reused here? I confess my skills as a software developer are not up to that task, but hopefully someone who is more versed in this than I can take a look at it.

me-no-dev, 2019-04-12:
So... IPv6 LinkLocal is supported and working. Arduino has API and examples even.

me-no-dev, 2019-04-12:
@owendelong maybe check the wifi events. You should enable IPv6 on interface start, then you will get event when IP is assigned. It does not happen as soon as you call enableIPv6()

owendelong, 2019-04-21:
@me-no-dev So that's great and all, but I need more than link-local support, and, I'd really like to suggest that the developers here take a look at what's been done in the ESP8266 project I referenced.

They've got a pretty nice full code stack including a decent web server, SSL support, and some other capabilities that make IPv6 not only possible, but actually quite easy in the ESP8266 environment. This has me using 8266s in most of my solutions for now, but there are a couple of places where I need a bit stronger processor (e.g. faster SSL response). For now, that's pushing me all the way to a PI 0 W solution, which is overkill both in the size of the device and in the per unit cost.

It would be really cool if this project could make their IPv6-capable libraries compatible with the API of the libraries there so that moving between 8266 and 32 platforms is as simple as a recompile.

me-no-dev, 2019-04-23:
I think you should post this in ESP-IDF as that is where LwIP comes from in Arduino :) If they make it work, it will start working here as well

Gigadoc2, 2019-08-02:
Well, AFAIK ESP-IDF has full IPv6 support?

owendelong, 2019-08-03:
ESP-IDF has full IPv6 support. ESP8266 Arduino has some IPv6 support.

ESP32 Arduino still has zero IPv6 support to the best fo my knowledge.

Please do NOT close this.

@me-no-dev, It's not quite that simple... There's a LOT of binding/glue required.

Gigadoc2, 2019-10-10:
I do not believe that this should be closed.

owendelong, 2019-10-10:
The issue should not be closed. it would be nice to see activity. If I had the skills necessary, I'd happily work on moving it forward.

It would be nice if someone who does have the necessary skills could look at the work that has been done for the 8266 Arduino project and port it over to here.

me-no-dev, 2019-10-11:
it needs to be ported to esp-idf, not Arduino :) once IDF gets it, we will get it too.

me-no-dev, 2019-10-11:
we have no different support than ESP-IDF. under arduino runs FULL esp-idf installation. All libs are there.

closed 2019-10-11 by me-no-dev

leonelsr, 2021-11-28:
So... What is actually still missing?

On ESP8266, IPv6 DHCP "just works", all you gotta do is select the right lwip "flavor" on Arduino IDE (or maybe use a build flag on platformio.

What exactly still needs to be done and/or ported (from where?), so we can have the same functionality on ESP32?

me-no-dev, 2021-11-28:
First it needs to become a configuration value in ESP-IDF (that means the option to be accessible by idf.py menuconfig), then we can enable that for the Arduino builds and bring support to you. Until then, we can not provide this. Manually editing headers is not really an option for us. 8266 Arduino builds it's own LwIP and we use the one in ESP-IDF

rkarlsba, 2023-12-22:
Any news on this?

VojtechBartoska, 2023-12-25:
@rkarlsba Yes, currently proposed here: https://github.com/espressif/arduino-esp32/pull/9016

rkarlsba, 2023-12-25:
Thanks - I'll just follow that :)