espressif/arduino-esp32 #1262: ESP32 does not accept Static IP

williamesp2015, 2018-03-26:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			Latest
IDE name:							Platform.io
Flash Frequency:					       40Mhz
Upload Speed:						115200?

#include <Arduino.h>

#include <WiFi.h>
#include <WiFiMulti.h>
void setup() {
  Serial.begin(115200);
   delay(1000);
  WiFi.mode(WIFI_AP);
  WiFi.softAP("ADSL1", "AbbasMeamarbashi2017***");
  Serial.println("Wait 100 ms for AP_START...");
  delay(300);
 WiFi.onEvent(wifi_event);
 for(int rrr=0;rrr<20;rrr++)
  {
    Serial.println("Set softAPConfig");
    IPAddress Ip(192, 168, 1, 11);
    IPAddress NMask(255, 255, 255, 0);
    WiFi.softAPConfig(Ip, Ip, NMask);
    IPAddress myIP = WiFi.softAPIP();
    Serial.print("AP IP address: ");
    Serial.println(myIP);
    Serial.println("IP address: ");
    Serial.println(WiFi.localIP());
    delay(1000);
}
void loop()
{
}

JacoFourie, 2018-03-27:
I seem to have the same issue. If I use the sample program it will not connect.

JacoFourie, 2018-03-27:
I have tested both methods.

````
 //if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
  //  Serial.println("STA Failed to configure");
 // }

  Serial.print("Connecting to ");
  Serial.println(ssid);

  //WiFi.begin(ssid, password);

  if(!WiFi.config(IPAddress(10, 64, 9, 38), IPAddress(10, 64, 9, 20), IPAddress(255, 255, 255, 0), IPAddress(10, 64, 9, 20))){
        Serial.println("STA Failed to configure");
    }
    if(WiFi.begin(ssid, password) == WL_CONNECT_FAILED){
        Serial.println("STA Failed to start");
    }

  while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }

````

The output I get is this

````
Connecting to Pandora-studio
..............................................................................................................................................................................................................................................

````

If I use DHCP by removing this line

````
WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)
````

Then it connects without any problem.

````
Connecting to Pandora-studio
...
WiFi connected!
IP address: 10.64.9.119
ESP Mac Address: 24:0A:C4:0D:84:40
Subnet Mask: 255.255.255.0
Gateway IP: 10.64.9.20
DNS: 10.64.9.20

````

JacoFourie, 2018-03-27:
OK I found this. https://github.com/espressif/arduino-esp32/issues/1081

TThe error is still there. If I remove 

````
while (WiFi.status() != WL_CONNECTED) {
    delay(500);
    Serial.print(".");
  }
````

and just add a delay of 15 seconds then all works fine.

JacoFourie, 2018-03-27:
OK so this is the fix.

https://github.com/espressif/arduino-esp32/pull/1129/commits/144480637a718844b8f48f4392da8d4f622f2e5e

comment out the if

````
} else if(event->event_id == SYSTEM_EVENT_STA_GOT_IP) {
        //if(WiFiSTAClass::status() == WL_IDLE_STATUS) {
            WiFiSTAClass::_setStatus(WL_CONNECTED);
        //}
    }
````

williamesp2015, 2018-03-28:
Thanks JacoFourie. Could please simplify your correct changes? Thanks

JacoFourie, 2018-03-28:
Go to the file in the libraries folder of the Arduino-Core files

My home path is

````
C:\Users\Jaco\Documents\Arduino\hardware\espressif\esp32\
````

````
C:\Users\Jaco\Documents\Arduino\hardware\espressif\esp32\libraries\WiFi\src\WiFiGeneric.cpp
````

and comment out the if as shown.

````
} else if(event->event_id == SYSTEM_EVENT_STA_GOT_IP) {
        //if(WiFiSTAClass::status() == WL_IDLE_STATUS) {
            WiFiSTAClass::_setStatus(WL_CONNECTED);
        //}
    }
````

Duske, 2018-03-28:
@JacoFourie can confirm that this did the trick for me as well. In my case running Atom with platform.io,
the `WifiGeneric.cpp` file  is located here: `.platformio\packages\framework-arduinoespressif32\libraries\WiFi\src\WiFiGeneric.cpp`

sreenumalae, 2018-05-03:
thank you so much for your answers.
But I'm confused and I'm not able to figure out the sketch to assign a static IP for esp32.
so @JacoFourie can you please provide me the exact sketch?

JacoFourie, 2018-05-03:
@sreenumalae I am posting the part you need for a static IP
````

#include <WiFi.h>

const char* ssid     = "Pandora";
const char* password = "**********";

IPAddress local_IP(10, 64, 9, 38);
IPAddress gateway(10, 64, 9, 20);
IPAddress subnet(255, 255, 0, 0);
IPAddress primaryDNS(10, 64, 9, 20); //optional
IPAddress secondaryDNS(8, 8, 8, 8); //optional

void setup_wifi() {

  

  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);

  if (WiFi.status() == WL_CONNECTED) {
    WiFi.disconnect();
  }

  if (WiFi.status() != WL_CONNECTED) {
    WiFi.persistent(false);
    WiFi.mode(WIFI_OFF);
    WiFi.mode(WIFI_STA);
    
    if (!WiFi.config(local_IP, gateway, subnet, primaryDNS, secondaryDNS)) {
    Serial.println("STA Failed to configure");
      }
    WiFi.begin(ssid, password);
  }

  wificounter = 0;
  while (WiFi.status() != WL_CONNECTED && wificounter < 10) {
    for (int i = 0; i < 500; i++) {
      delay(1);
    }
    Serial.print(".");
    wificounter++;
  }

  if (wificounter >= 10) {
    Serial.println("Restarting ...");
    ESP.restart();
  }

  delay(10);
  // We start by connecting to a WiFi network

  Serial.println("");
  Serial.println("WiFi connected.");
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

}

void setup() {
  
  Serial.begin(115200);
          
  // We start by connecting to a WiFi network
  setup_wifi();

}

closed 2019-08-15 by stale[bot]