espressif/arduino-esp32 #1264: Problem with SPIFFS and FS exists.

rrowe629, 2018-03-26:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							?ADAFRUIT ESP32 Feather?
Core Installation/update date:			?26/mar/2018?
IDE name:							?Arduino IDE?
Flash Frequency:					?80Mhz?
Upload Speed:						?115200?

### Description:
I've been chasing a problem with ESPAsyncWebServer and serving gzipped files. I've narrowed it down to the line 501 of WebResponses.cpp is never true. Here's the line:

     if(!download && !fs.exists(_path) && fs.exists(_path+".gz")){

A bit more research leads me to believe that there is a problem passing SPIFFS to the function. I constructed a sketch to test this theory, which is included below. Sure enough, when passing SPIFFS to the testSPIFFS function, I get a different result. I set Core Debug Level to "Debug", but no debug messages are displayed. I've uploaded a file called "index.html" to SPIFFS, but there is no "index.html.gz" present. Here's the output I get from my sketch:

SPIFFS.exists("/index.html") : 1
SPIFFS.exists("/index.html.gz") : 0
fs.exists("/index.html") : 1
fs.exists("/index.html.gz") : 1

SPIFFS returns the correct result, FS has a false positive on index.html.gz. This is the same behavior I am seeing in WebResponses.cpp - exists always returns true whether the file is there or not.

### Sketch:
#include "SPIFFS.h"

void testSPIFFS(FS &fs, String path) {
  String pathWithGz = path + ".gz";
  Serial.printf("fs.exists(\"%s\") : %d\n", path.c_str(), fs.exists(path));
  Serial.printf("fs.exists(\"%s\") : %d\n", pathWithGz.c_str(), fs.exists(pathWithGz));
}

void setup() {
  Serial.begin(115200);
  SPIFFS.begin();
  String path = "/index.html";
  String pathWithGz = path + ".gz";

  Serial.printf("SPIFFS.exists(\"%s\") : %d\n", path.c_str(), SPIFFS.exists(path));
  Serial.printf("SPIFFS.exists(\"%s\") : %d\n", pathWithGz.c_str(), SPIFFS.exists(pathWithGz));
  testSPIFFS(SPIFFS, path);
}

void loop() {

}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

me21, 2018-08-07:
I confirm, I've hit the issue myself, only with built-in web server.
`fs.exists("/")` returns true, `SPIFFS.exists("/")` returns false inside the constructor of `StaticRequestHandler` class.

alanswx, 2018-11-04:
This just caused me a lot of trouble. I saw your post and temporarily modified ESPAsyncWebServer:

bool Fileexists(FS &fs,const char* path)
{
    File f = fs.open(path, "r");
    return (f == true) && !f.isDirectory();
}

bool Fileexists(FS &fs,const String& path)
{
    return Fileexists(fs,path.c_str());
}

AsyncFileResponse::AsyncFileResponse(FS &fs, const String& path, const String& contentType, bool download, AwsTemplateProcessor callback): AsyncAbstractResponse(callback){
  _code = 200;
  _path = path;

  //if(!download && !fs.exists(_path) && fs.exists(_path+".gz")){
  if(!download && !Fileexists(fs,_path) && Fileexists(fs,_path+".gz")){

And it seemed to fix the problem. Is this a subclass problem?  Where is the right place to fix it? I wonder if this code just broke the other file systems, like SD Cards.

judge2005, 2019-05-05:
I just hit this too. Is there a fix for this anywhere?

judge2005, 2019-05-06:
Submitted this issue to esp-idf: https://github.com/espressif/esp-idf/issues/3427

me21, 2019-08-01:
It can be safely closed, fixed by pull request #2763

closed 2019-10-14 by stale[bot]