espressif/arduino-esp32 #1265: esp_deep_sleep_start() does not work for longer durations

melvinpmathew, 2018-03-27:
### Hardware:
Board:							ESP-WROOM-32
Core Installation/update date:			Xtensa-5.2.0
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Was trying out different sleep times for esp_sleep_enable_timer_wakeup() +esp_deep_sleep_start(). Configuring the sleep time for 30sec,3600sec makes the board go to sleep and wakes up exactly at the configured time. 

The issue is : for long durations of sleep i.e 54000seconds or 15hrs, the board goes to sleep fine, but the chip does not wake up even after 17 hours! Any idea whats going on??

### Code:
Serial.println("System going to sleep for " + t_sec + "seconds");
uint64_t sleeptime = t_sec.toInt() * 1000000;
delay(1000);
esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
esp_sleep_enable_timer_wakeup(sleeptime);
esp_deep_sleep_start();

melvinpmathew, 2018-03-27:
Testing by using..

uint64_t sleeptime = UINT64_C(t_sec.toInt() * 1000000);

to define the constant as a unit64 variable. Might just work!

hpsaturn, 2018-04-01:
Hi, I have a similar issue, but my problem is that the clock is inexact after longer durations, when my ESP is full turn on, the clock works well but when it go to suspend and after long time the clock go to innacurate. (**some minutes forward**). Please, maybe do you know any fix? is obligatory external crystal? or the ESP32 internal RTC is wrong?

---
Board: ESP32 WeMOS OLED (China version)
Framework: PlatformIO
Source code: [Portable EAP Google Authenticator ESP32 hardware](https://github.com/hpsaturn/esp32_potp/tree/av/wifi_suspend_fix)

---

**Some parts of code:**

Suspend main function:
```C
void suspend(){
  Serial.println("Process Suspend..");
  esp_sleep_enable_touchpad_wakeup();
  esp_deep_sleep_start();
}
```
Touch buttons config:
```C
void gotTouch2(){
  touch2detected = true;
}
void setup() {
  Serial.begin(115200);
  Serial.println("== ESP32 Booting ==");
  // Init touch callbacks
  touchAttachInterrupt(T2, gotTouch2, THRESHOLD);
  touchAttachInterrupt(T3, gotTouch3, THRESHOLD);
  Serial.println("== Setup ready ==");
}
```
NTP clock config after wifi is ready:
```C
static void initialize_sntp(void) {
  Serial.println("Initializing SNTP");
  configTime(0,0,"pool.ntp.org");
}

static void obtain_time(void) {
  initialize_sntp();
  // wait for time to be set
  time_t now = 0;
  struct tm timeinfo = {0};
  int retry = 0;
  const int retry_count = 10;
  Serial.print("Waiting for system time to be set.");
  while (timeinfo.tm_year < (2016 - 1900) && ++retry < retry_count) {
    Serial.print(".");
    delay(100);
    time(&now);
    localtime_r(&now, &timeinfo);
  }
  Serial.println("ready.");
}
```

melvinpmathew, 2018-04-02:
To tell you the truth, I could not get the RTC after suspend. The RTC time
is always 0!. So what I have done is stored RTC in the non-volatile storage
before entering suspend mode and when the system restart, extract from the
non-volatile and add the time slept. This is giving me fairly accurate
reading thus far. i do have other systems performing RTC sync too. Maybe
you can share how you are able to get the RTC time after suspend and
restart and we can check it out.

hpsaturn, 2018-04-02:
I don't have any code about it, only I go to suspended via deep_sleep directives, and wakeup it via touch buttons, with this I have a right time but if it got to long sleep (10h for example) the clock be inaccurate, with some positive minutes of difference, the clock be advance but is "coherent". Maybe please do you share you RTC sync code? Thanks

nevercast, 2019-05-15:
On to the original subject, If I sleep for 10 minutes, it is exact to the millisecond.
If I sleep for 1 hour, the chip wakes up instantly.

I am using external crystal and it works very well, except for long durations. I'm hunting integer overflow issues now.

nevercast, 2019-05-15:
@melvinpmathew Thank you, you fixed my issue. ESP32 Deepsleep with external 32kHz clock is now working! This is cause for celebration.

hpsaturn, 2019-05-15:
nice, congrats, but for long duration you fixed the issues? or only with a external 32kHz crystal is enough?

nevercast, 2019-05-16:
If you're using the internal RC oscillator (Not using external crystal) you should know that it has very terrible stability. About 5% drift. So after 10 hours I expect inaccuracy of 30 minutes.

If you want the clock to be accurate for longer durations, you will need to use external 32kHz crystal, external oscillator, or external RTC chip. You cannot sleep for a long time using just the RTC RC Oscillator by itself.

closed 2019-08-14 by stale[bot]