espressif/arduino-esp32 #1270: ESP.restart() kills only loop thread instead of restarting the board?

miknuc, 2018-03-28:
### Hardware:
Board:							ESP32 DEV Board (DoIt)
Core Installation/update date:			Feb 15 2018
IDE name:							eclipse
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
Hi, I am having periodically issues with my wifi net. Sometimes my ESP32 looses it wifi connection:

`E (562420949) wifi: esp_wifi_disconnect 847 wifi not start`

I try a wifi connect in this case, if this doesn't help I intended to restart my ESP32, see loop () below.
My console log shows exactly one line 
`Restart after XXX`
but the system did not restart.
 
Instead I get couple of lines like:
``` 
Task watchdog got triggered. The following tasks did not reset the watchdog in time:^M
 - IDLE (CPU 0)^M
 Tasks currently running:^M
 CPU 0: wifi^M
 CPU 1: IDLE^M
```

Is it possible that ESP.restart() is blocked by something (e.g. broken wifi connection) and that this just stops the loop() thread instead of the hole ESP32?

In some demo sketches I saw a delay(xxx) after the ESP.restart(). I didn't saw any reason for this and didn't include it. Am I missing something?

Best Regards
Michael

### Sketch:
...leaving out some additional code (e.g. MQTT handling stuff)
```

void MQTTLoop(void *pvParameters) {
	  while (1) {
		  if (pruefeAktion(&MQTTSubscriptionCheck,20))  subscribeMQTT();
		  if (WiFi.status() != WL_CONNECTED)
		  {
			  Serial.println("no WiFi! WiFiStatus:"+String(WiFi.status()));
			  WiFi.reconnect();
		  }
		  if (PSClient.connected())
		      {
		      PSClient.loop();
		    }
		    else {
		        debug(4,"loop", "no PSClient.loop()");
		        connectMQTT();
		        }
		  delay(500);
	  }
}

void setup() {
  Serial.begin(115200);
  Serial.println("Booting");
  pinMode(LED1,OUTPUT);
  digitalWrite(LED1,LOW);

    WiFi.begin(ssid, password);

    	for (int i=0;i<100;i++)
    	{
    		if (WiFi.status() != WL_CONNECTED) {
    			delay(500);
    			Serial.print(".");
    		}
    		else
    			break;
        }
    	if (WiFi.status() == WL_CONNECTED)
    	{
    		Serial.println("");
    		Serial.println("WiFi connected");
    		Serial.println("IP address: ");
    		Serial.println(WiFi.localIP());
    	}
    	else
    		Serial.println("keine WiFi Verbindung erhalten.");
    	Serial.println("Ready");

  xTaskCreate( DebugLoop, "DEBUG", 4096,NULL,10,NULL);

  PSClient.setClient(MQTTclient);
  int MQTT_Port = (int) strtol(mqtt_port,NULL,10);
  PSClient.setServer(mqtt_server,MQTT_Port);
  PSClient.setCallback(MQTTcallback);
  if (connectMQTT())
  {
    Serial.println("MQTT connect erfolgreich");
    }
  else
  {
    Serial.println("MQTT connect fehlgeschlagen");
  }

  xTaskCreate( MQTTLoop, "MQTT", 4096,NULL,10,NULL);
  xTaskCreate( LEDLoop, "LED", 4096,NULL,10,NULL);
  xTaskCreate( MonitorLoop, "Monitor", 4096,NULL,10,NULL);

  //Subscriptions initial erstellen
  subscribeMQTT();
  debug(1,"setup","started.");

}

void loop() {

	uint32_t Uptime = millis();
	if (WiFi.status() != WL_CONNECTED)
		WiFi.reconnect();
	if (WiFi.status() != WL_CONNECTED && (Uptime > 1000* 60*5))
	{
		debug(1,"loop","Restart after "+ String(Uptime));
		ESP.restart();
	}
	ArduinoOTA.handle();
  delay (200);

}

```

lbernstone, 2018-03-28:
I think you are not doing your xtasks properly and are getting endless loops that the watchdog kills.  Have a look at https://desire.giesecke.tk/index.php/2018/02/07/using-the-multitasking-capabilities-of-the-esp32-freertos/ (@beegee-tokyo)

miknuc, 2018-03-29:
These watchdog message appear the first time after calling ESP. restart().

The other threads are still looping as expected afterwards, as I can see by their debug output. The only blocked thread I can see is the loop thread and I assume that is got blocked in ESP. restart() and is therefore not reaching the delay() statement. What if my assumption is correct? Would ESP. reset() be a better option?

lbernstone, 2018-03-29:
There is no ESP.reset in ESP32.  I think it is just an alias for restart.  I use the watchdog timer to do a hard restart like this:
void hard_restart() {
  esp_task_wdt_init(1,true);
  esp_task_wdt_add(NULL);
  while(true);
}
It sets the wdt to 1 second, adds the current process and then starts an infinite loop.

miknuc, 2018-03-29:
Is my understanding of this example correct:
you initialize watchdog timer to call the panic handler (which reboots the ESP), if the timer does not get a reset every second and than you start a CPU consuming loop that will trigger the timer quickly.

Interesting idea to commit suicide...
I will give it a try these days, thank you very much.

aj-ptw, 2018-04-06:
@lbernstone what headers are you including to use the 
```
/Users/aj/Documents/Arduino/libraries/OpenBCI_Air_Shield_Library/examples/DefaultAirShield/DefaultAirShield.ino: In function 'void hard_restart()':
/Users/aj/Documents/Arduino/libraries/OpenBCI_Air_Shield_Library/examples/DefaultAirShield/DefaultAirShield.ino:458:27: error: 'esp_task_wdt_init' was not declared in this scope
   esp_task_wdt_init(1,true);
                           ^
/Users/aj/Documents/Arduino/libraries/OpenBCI_Air_Shield_Library/examples/DefaultAirShield/DefaultAirShield.ino:459:24: error: 'esp_task_wdt_add' was not declared in this scope
   esp_task_wdt_add(NULL);
```

miknuc, 2018-04-06:
These includes should help:
```
#include <esp_int_wdt.h>
#include <esp_task_wdt.h>
```

neoaurion, 2018-10-06:
> There is no ESP.reset in ESP32. I think it is just an alias for restart. I use the watchdog timer to do a hard restart like this:
>
> ```
> void hard_restart() {
>   esp_task_wdt_init(1,true); [...]

I tried and it works like a charm ! Thank you !

luongvanchung93, 2018-12-15:
I encounter this error please help me >>>stack>>>

ctx: cont 
sp: 3ffffc60 end: 3fffffc0 offset: 01b0
3ffffe10:  00000000 00000000 00000000 00000000  
3ffffe20:  00000000 3ffffebc 00000000 3ffffeb8  
3ffffe30:  3ffffeb4 00000000 00000000 00000000  
3ffffe40:  00000000 3fffff90 3fffff90 00000000  
3ffffe50:  3ffef568 00000001 3ffef568 40212330  
3ffffe60:  3ffffea4 00000000 3ffef568 40207f91  
3ffffe70:  00000000 00000000 00000000 4020a300  
3ffffe80:  40000000 00000000 00000000 40211b2c  
3ffffe90:  00000000 00000000 00000000 00000000  
3ffffea0:  3fffff90 3fffff90 3ffef568 40211b6d  
3ffffeb0:  3ffef568 4021226c 3ffef568 402122e3  
3ffffec0:  3ffee970 00000000 00000038 4021246d  
3ffffed0:  3ffee970 00000000 3fffff30 402047d2  
3ffffee0:  3fffff90 40212478 00000000 40100898  
3ffffef0:  3fff0204 00000000 40202584 402025d4  
3fffff00:  3ffee6a4 00000000 3fffff40 402025a1  
3fffff10:  3fffdad0 40212478 3fffff40 3ffee8f0  
3fffff20:  3fffff90 00000000 3ffee6a4 402048c9  
3fffff30:  3ffe8632 00000001 00000000 402025a1  
3fffff40:  3fff016c 00000000 40202584 402025d4  
3fffff50:  3fffdad0 3ffee6a4 3fffff90 40204932  
3fffff60:  3fff019c 00000000 40202584 402025d4  
3fffff70:  402089e8 00000000 3ffee8bc 402026bc  
3fffff80:  3ffef51c 00ffffff 40202584 402025d4  
3fffff90:  3ffef504 0000000f 00000001 40208065  
3fffffa0:  3fffdad0 00000000 3ffee8bc 402080f0  
3fffffb0:  feefeffe feefeffe 3ffe8514 40100ae5  
<<<stack<<<

last failed alloc call: 4020ADF1(1052)

 ets Jan  8 2013,rst cause:2, boot mode:(1,6)

 ets Jan  8 2013,rst cause:4, boot mode:(1,6)

wdt reset

luongvanchung93, 2018-12-16:
Thank you very much!

closed 2019-08-15 by stale[bot]