espressif/arduino-esp32 #1271: ESP32 with usb host shield

jef737, 2018-03-28:
Hardware:
Board:							ESP32 DevkitV1
Core Installation/update date:		        03/02/2018
IDE name:							Arduino IDE
Flash Frequency:					40Mhz
Upload Speed:						115200

Hi,
I would like to use a Sparkfun usb host shield with my ESP32 in order to get data from usb peripheral and send it to a server by wifi . I'm using the usbhostshield library and first of all i have tried the sketch USB_desc and board_qc, but unfortunatly both doesn't work. i use the ESP32 VSPI pin for the SPI interface. As it works fine with an Arduino Uno, although having spent a lot of time on the net i didn't find any thing to solve my problem. Then if someone could help me it would be greatfull.

lbernstone, 2018-03-28:
You need to take this up with the library maintainers. https://github.com/felis/USB_Host_Shield_2.0

jef737, 2018-03-29:
Ok thanks 
i'm going to do it

closed 2018-03-29 by jef737

FLYINGSC0T, 2019-09-06:
Hi,
I would like to use an ESP32 with a Mini USB Host Shield and be able to run the pl2303_tinygps.ino shetch in the examples folder. The code compiles but when I run the program on the ESP32 the line if(Pl.isReady()) never returns true. My Mini USB Host Shield does have a MAX3421E onboard but I don't think it has a PL2303.  This is my host shield: https://www.aliexpress.com/item/32733047611.html?spm=a2g0s.9042311.0.0.467f4c4dNWMEPK 
Any help would be appreciated in getting a USB GPS to work with a Mini USB Host Shield running on an ESP32 using TinyGPSPlus library.

jef737, 2019-09-07:
Hi,
Sorry to say that but i've never could do it working with arduino hardware. I've changed for raspberry PI zero with integrated usb host and linux OS, now it works fine!

FLYINGSC0T, 2019-09-07:
Hi jef737,
Thanks for your reply. For my project it's not an option to change to a Raspi. Someone, surely, must have come up with a solution. My search continues....

masterglob, 2020-09-20:
Hi flyingscot,

I am having exacly the same issue. did you manage to make it work?

I tried using USB_Host_Shield_2.0  library and the following PINOUT:
MOSI / GPIO23
MISO / GPIO19
CLOCK/ GPIO18
CS / GPIO5
INt / GPIO 17

I alos tried the other SPI configuration with no more success (pins 12/13/14/15)

Running the "USB_desc" example prints out "OSC did not start."
Regards,

lbernstone, 2020-09-20:
This still needs to be handled at the library repo.
Espressif now makes a chip that natively speaks USB. https://www.mouser.com/ProductDetail/Gravitech/GRDE-3930?qs=vmHwEFxEFR8PVsJiGhs5dA%3D%3D

masterglob, 2020-09-20:
Thanks... I'm sad I just provisionned some ESP32 & USB shields...

Enyway, even the ESP8266 demo (https://wikifab.org/wiki/Brancher_des_p%C3%A9riph%C3%A9riques_USB_sur_un_ESP8266) deos not work ... So I guess I got a defective part of SB shield (SKetch stuck as soon as I plug th I2C "SS" signal...)

mkc2019, 2020-10-23:
Guys I seem to manage to get it to work!
This is the pin connection is used:
GPIO5 : SS, GPIO17 : INT, GPIO18 : SCK, GPIO19 : MISO, GPIO23 : MOSI

Aside from that, I also:
bridge the RST pin of USB shield to VCC (i.e. 3.3V)
Cut off the track of VBus to 3.3V, then solder a wire connecting the VBus to the Raw pin (with USB connector at the top and board facing upwards, Raw pin is the one at the bottom left, just below GND). Then I power the raw pin with 5V.

I still need to do more testing with this setup...but at least with this configuration I got the usb_desc script running an return useful info. (Super excited because I worked on this for a whole day...)

masterglob, 2020-10-25:
Hi mkc!

I'm very thankfull to you...

I tried so many different things and have not been able to make it... What seems to be missing is that RST has to be driven to 5V.

mkc2019, 2020-10-27:
@masterglob you are welcome! Are you using USB host shield mini? Because mini runs on 3.3V.
Did it work out in the end for you?

masterglob, 2020-10-27:
Yes more precisely, this is the USB Mini v2.0 DuinoFun (3,3V). I followed instructions for 5V USB power as shown in the link above (Initally for ESP8266, but the principle is the same. For the record, I added the routing informations on the page for ESP32)

I managed to connect a USB MIDI device (IK Multimedia IRIG 37 pro), and was able to read all events correctly. I was planning to make a standalone PIANO/SYNTH using an ESP32, an USB host shield and an external DAC (PCM 5102) that can accept serial MIDI or USB connection. But I fear that RAM amount will be a critical issue for my initial design...

roelandg, 2021-02-15:
Hi @masterglob ,

Thanks for your instructions here and on the link you shared. I have a ESP-WROOM-32 and the USB Mini v2.0 DuinoFun.

I have connected it according to your suggestions, **without** the 5V cut solution.

ESP32 <--> Mini

D5/GPIO15 <--> SS
TX2/GPIO17 <--> INT
D18/GPIO18 <--> SCK
D19/GPIO19 <--> MISO
D23GPIO23 <--> MOSI
GND <--> GND
VIN <--> 5V

My plugged USB device gets 5V from the USB host, which I checked with a multimeter, also the soldered led on the USB device lights on. I didn't change the UsbCore.h settings, as it seems ok:
`
#elif defined(ESP32)
typedef MAX3421e<P5, P17> MAX3421E; // ESP32 boards
`

However it results in: 

`
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
Start
OSC did not start.
`

Also tried cutting the PCB to perform the 5V trick from the tutorial, but exactly the same result.

Any tips or idea's? 

Thanks in advance, very much appriciated.

mkc2019, 2021-02-16:
@roelandg i don't think D5 is GPIO15 iirc, perhaps you can check the documentation on pin layout?

Otherwise, use a multimeter to check whether all the pins are connected correctly. In my case, i need an additional 3.3V connection to the RST pin of Arduino USB Host Shield mini. 

Whether to use 3.3V or 5V for VIN depends on your shield. IIRC Arduino USB host can run on either but the Mini version only runs on 3.3V.

Btw i cut off thr VBUS track because according to USB specification VBUS requires 5V but my mini USB host only supplies 3.3V, so i need an external input for VBUS. This is only required of your USB device gets power from VBUS (like keyboard or mouse), otherwise it is not necessary.

roelandg, 2021-02-17:
@mkc2019 Thanks, will continue tonight, after new ESP32 has arrived. Fried the other one 😅

masterglob, 2021-02-19:
@roelandg
> Any tips or idea's?

Yes, As I see in your description, you missed the information in my prevous post : 
> What seems to be missing is that RST has to be driven to 5V.

mrweaver, 2021-03-08:
@roelandg any luck with this? I've had the same issues, I've cut the VBUS trace and am attempting to connect to a keyboard as a test, but get:

```
Start
OSC did not start.
```

I took a look at the [MAX3421E datasheet](https://datasheets.maximintegrated.com/en/ds/MAX3421E.pdf), and it seems that the Reset pin shouldn't be subject to 5V.

I've tried using the HSPI and VSPI ports of the ESP32 without success. Perhaps I made a mistake somewhere though.

chegewara, 2021-03-08:
This issue makes me thinking it would be good idea to write a library/project which would use esp32 S2 exactly the same way (USB over SPI) to add USB to old esp32. Sadly we dont have host USB on S2 yet.

roelandg, 2021-03-08:
@mrweaver Nice to see that you have made it work! I gave it up, so I made it easy for myself and bought a raspberry nano with usb already attached. But with your comment I can give it a try again.

mrweaver, 2021-03-09:
I must admit, I've had similar issues with the ESP32 in another project (RGB LED screens). Drove me nuts!

It was nice to finally get it connected. I plan on using one of those Chinese remote "air mice" as an input to the system - I started trying to hack into the nRF24 cloned hardware, eventually gave up when the other seemed like I lacked the skill and I realised this was a better solution. I also bought an ESP32-S2 module, but alas, would have to wait for some cluey devs to release the usb host drivers.

Good luck - I'm keen to hear how you go with it!

hyshiah, 2021-03-29:
in avrpins.h @ about line 1461:
#elif defined(ESP8266) || defined(ESP32)

#define MAKE_PIN(className, pin) \
class className { \
public: \
  static int8_t iopin; \
  static void Set() { \
    digitalWrite(pin, HIGH);\
  } \
  static void Clear() { \
    digitalWrite(pin, LOW); \
  } \
  static void SetDirRead() { \
    pinMode(pin, INPUT); \
  } \
  static void SetDirWrite() { \
    pinMode(pin, OUTPUT); \
  } \
  static uint8_t IsSet() { \
    return digitalRead(pin); \
  } \
  static int8_t pinNum() { \
    return pin; \
  } \
};

in usbhost.h @ about line 49:
#elif defined(SPI_HAS_TRANSACTION)
        static void init() {
                #if defined( ESP32 )
                log_d("defined SPI_HAS_TRANSCATION AND ESP32...\n");
                USB_SPI.begin(SPI_CLK::pinNum(), SPI_MISO::pinNum(), \
                SPI_MOSI::pinNum(), SPI_SS::pinNum());
                log_d("SPI_CLK....%d\n", SPI_CLK::pinNum());
                log_d("SPI_MOSI...%d\n", SPI_MOSI::pinNum());
                log_d("SPI_MISO...%d\n", SPI_MISO::pinNum());
                log_d("SPI_SS.....%d\n", SPI_SS::pinNum());
                #else
                USB_SPI.begin();
                #endif
                SPI_SS::SetDirWrite();
                SPI_SS::Set();
        }
#elif defined(STM32F4)

will work for esp32 :-)

philipn7, 2021-04-12:
Finally got the ESP32 dev board working with a USB Host Shield Mini (3.3V). Credit goes to @kn1cht in [this thread](https://github.com/felis/USB_Host_Shield_2.0/issues/324).

The labels on most versions of the USB Host Shield (UHS) boards are wrong! The **CLK** and **MOSI** connections should be swapped.

| _UHS_ | _ESP32 Dev Board_ |
| --- | --- |
| SS | GPIO5 (D5) |
| **CLK** | **MOSI (D23)** |
| MISO | MISO (D19) |
| **MOSI** | **SCK (D18)** |
| INT | GPIO17 (TX2) |
| MAX_RST | 3.3V (or solder a wire between MAX_RST and VCC on the UHS) |
| GND | GND |
| VCC | 3.3V |
| VBUS | 5V (Cut the trace between VBUS and the 2k2 resistor) |

I have two boards from different chinese vendors. One of them is marked 'Aug,16,2014' while the other does not have a date. They both have the exact same circuit layout and both had wrong labels. 

The library worked for me unmodified. Run the 'board_qc' example code and the serial monitor should show a revision number, SPI long test passed and GPIO test fail (this is normal). The 'SPI long test' should pass even without a USB device attached or even cutting the VBUS-to-2k2 trace.

In order for USB devices to work, the UHS needed to be modified by cutting the trace between VBUS and the 2k2 resistor as shown in [this ESP8266 tutorial](https://www.hackster.io/139994/plug-any-usb-device-on-an-esp8266-e0ca8a). This allows you to connect a 5V supply to VBUS and power your USB device. Otherwise, it's getting 3.3V which is not enough for most USB devices.

If you're new to the ESP32 then you should also know that different ESP32 board designs may have different pinouts from the 'dev' board. Refer to pinout diagram for your board and match up the GPIO and SPI labels.

Kakomis, 2021-11-19:
Hi @philipn7 ! my setup consist in ESP-WROOM-32  + USB Mini v2.0 DuinoFun + Sparkfun Scanner. If a valid code is scanned, the ESP32 will turn a relay on for 2 sec. I based on the acm_terminal example, but after a while communication stops and Acm.isReady() condition is false for ever, unless it is reset. Do you know what could be the reason? Relay is the common 5v relay module, which is on GPIO 26 with its own power supply. If you could help me or someone would be great.

fuadlotfisoltan, 2021-11-19:
Yes,I can halpe you.give me your number

dedesite, 2021-11-24:
Hi @philipn7,

I try to get an UHS Mini v2.0 bought [here](https://opencircuit.shop/Product/Mini-USB-Host-Shield) working with an [Inkplate 6 ](https://inkplate.readthedocs.io/en/latest/features.html) (ESP32 based e-ink board).
My shield doesn't have a date or DuinoFun marked on the board, just "UHS mini v2.0".
Do you think the MOSI and CLK are also mixed up ?

Also, as the Inkplate 6 is 3.3V only, I guess I should use a sort of voltage level converter to plug the VBUS, right ?

In the tutorial, they say that the INT pin wasn't necessary, is it also what you've seen ?

I'm really new to electronics and I don't understand why the RST pin should be plug into the 3.3V, how could the board receive the RESET signal if it is plugged to 3.3V ?

Thanks in advance for your help,
Andréas