espressif/arduino-esp32 #1273: Crash when using Preferences variable in setup() function

vupqtran, 2018-03-29:
Please fill the info fields, it helps to get you faster support ;)

If you have a Guru Meditation Error, please decode it:
https://github.com/me-no-dev/EspExceptionDecoder

----------------------------- Remove above -----------------------------

### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Describe your problem here
I can not use if statetement with variable get value from Preference in setup() function. It run if i replace variable with 1 or 0.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true);

  // Get all saved data
  preferences.begin("preferences", false); 
  bluetooth_status =  preferences.getBool("bluetooth_status", false); //NVS key bluetooth status
  
  preferences.end();
  
  DEBUG_PRINT("bluetooth_status: ");
  DEBUG_PRINTLN(bluetooth_status);
  
  if(bluetoot_status)
  {

  }
}
void loop() {
}
```

### Debug Messages:
```
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_
load:0x3fff0010,len:4
load:0x3fff0014,len:732drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x40078000,len:0
load:0x40078000,len:11572
entry 0x40078a14
[D][ESP32RGBmatrixPanel.cpp:262] ESP32RGBmatrixPanel(): Init ESP32RGBmatrixPanel!

[D][ESP32RGBmatrixPanel.cpp:262] ESP32RGBmatrixPanel(): Init ESP32RGBmatrixPanel!

[D][ESP32RGBmatrixPanel.cpp:262] ESP32RGBmatrixPanel(): Init ESP32RGBmatrixPanel!

[D][ESP32RGBmatrixPanel.cpp:262] ESP32RGBmatrixPanel(): Init ESP32RGBmatrixPanel!

assertion "res == pdTRUE" failed: file "/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./ipc.c", line 85, function: esp_ipc_init
abort() was called at PC 0x4014a6e7 on core 0

Backtrace: 0x40089620:0x3ffe3b50 0x4008971f:0x3ffe3b70 0x4014a6e7:0x3ffe3b90 0x4015fe81:0x3ffe3bc0 0x40083009:0x3ffe3c00 0x400831f0:0x3ffe3c30 0x4007886a:0x3ffe3c50 0x400788d1:0x3ffe3c80 0x40078a0e:0x3ffe3cb0 0x40078b33:0x3ffe3e70 0x40007c31:0x3ffe3eb0 0x4000073d:0x3ffe3f20

Rebooting...
```

vupqtran, 2018-04-02:
@beegee-tokyo thanks for your reply,
I try your method but it is not success.
I use EspExceptionDecoder and get below information:

`Decoding 16 results
0x4014a437: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x40089620: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x4008971f: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 553
0x4014a437: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8)
0x4015fbc1: esp_ipc_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./ipc.c line 85 (discriminator 1)
0x40083009: start_cpu0_default at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 317
0x400831f0: call_start_cpu0 at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./cpu_start.c line 207
`

beegee-tokyo, 2018-04-02:
Looking into you exception result I do not see anything pointing to preferences. I don't think your problem is the preferences library.
What I see again (like in your first comment) is
```
0x4014a437: __assert_func at /Users/ivan/e/newlib_xtensa-2.2.0-bin/newlib_xtensa-2.2.0/xtensa-esp32-elf/newlib/libc/stdlib/../../../.././newlib/libc/stdlib/assert.c line 63 (discriminator 8) 
0x4015fbc1: esp_ipc_init at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./ipc.c line 85 (discriminator 1) 
```
Which points again to an assert error in esp_ipc_init.

I saw as well in your first post
`[D][ESP32RGBmatrixPanel.cpp:262] ESP32RGBmatrixPanel(): Init ESP32RGBmatrixPanel!`    
What else are you doing in `setup()`? Try to comment out step by step library inits and calls until you know what causes your crash.

vupqtran, 2018-04-02:
@beegee-tokyo 
Thanks, i comment out step by step and find out when i call bluetooth init, it will crash my app.

`Serial.println("** Stopping WiFi");
    WiFi.mode(WIFI_OFF);
    Serial.println(WiFi.status());
    // Get free RAM
    DEBUG_PRINTLN(">>>>>>>>>>>>>> RAM 1 left: ");
    free_RAM = ESP.getFreeHeap();//system_get_free_heap_size();
    DEBUG_PRINTLN(free_RAM);
    resetResolution(is64x32);
    // Show Logo
    matrix[0].setProperties(makeZoneInforOneItem(is64x32, 0, 0, width, height, (text_type)TEXT_RAW, BLUE, true, false, (string)"", 0, true));
    matrix[0].setActive(false);
  
    matrix[0].drawRGBBitmap((width-50)/2, (height-11)/2, logo, 50, 11);

    
    i2c_master_init();
    
    int ret_conf;
    
    ret_conf = wm8731_config(I2C_MASTER_NUM, wm8731_cmd_entries, sizeof(wm8731_cmd_entries) / sizeof(wm8731_cmd_entry_t), 1);
  
    log_e("*******************\n");
    log_e("WM8731 initialization\n");
    log_e("*******************\n");
    
    if (ret_conf == ESP_OK) {
        log_e("Codec initialization success!\n");
        //disableDisplay();
        matrix[0].setProperties(makeZoneInforOneItem(is64x32, 0, 4, width, height, (text_type)TEXT_RAW, WHITE, true, true, (string)"Ready to   receive   stream...", 0, true));
        delay(SHOW_MESSAGE_DELAY_MS);
    } else {
        log_e("No ack, codec not connected...skip...\n");
        //disableDisplay();
        matrix[0].setProperties(makeZoneInforOneItem(is64x32, 0, 4, width, height, (text_type)TEXT_RAW, WHITE, true, true, (string)"No ack, codec not connected...skip...", 0, true));
        delay(SHOW_MESSAGE_DELAY_MS);
    }
  
    vTaskDelay(500 / portTICK_RATE_MS);
    
    i2s_driver_install((i2s_port_t)0, &i2s_config, 0, NULL);
    i2s_set_pin((i2s_port_t)0, &pin_config);

    
    esp_err_t ret = nvs_flash_init();
    if (ret == ESP_ERR_NVS_NO_FREE_PAGES) {
        ESP_ERROR_CHECK(nvs_flash_erase());
        ret = nvs_flash_init();
    }
    
    esp_bt_controller_config_t bt_cfg = BT_CONTROLLER_INIT_CONFIG_DEFAULT();
    if (esp_bt_controller_init(&bt_cfg) != ESP_OK) {
        log_e("initialize controller failed\n");
        return;
    }
  
    if (esp_bt_controller_enable(ESP_BT_MODE_BTDM) != ESP_OK) {
        log_e("enable controller failed\n");
        return;
    }    
    /*
    if (esp_bluedroid_init() != ESP_OK) {
        log_e("initialize bluedroid failed\n");
        return;
    }
  
    if (esp_bluedroid_enable() != ESP_OK) {
        log_e("enable bluedroid failed\n");
        return;
    }
  
    // create application task 
    bt_app_task_start_up();
  
    // Bluetooth device name, connection mode and profile set up 
    bt_app_work_dispatch(bt_av_hdl_stack_evt, 0, NULL, 0, NULL);
      // Get free RAM
    DEBUG_PRINTLN(">>>>>>>>>>>>>> RAM 2 left: ");
    free_RAM = ESP.getFreeHeap();//system_get_free_heap_size();
    DEBUG_PRINTLN(free_RAM);
    */`

closed 2019-08-15 by stale[bot]

DerrickLau, 2020-04-18:
I get the exact same problem.