espressif/arduino-esp32 #1274: ESP32 Dev Kit WROOM32 doit.am boards don't want to boot without pressing EN button

valentinAlkan, 2018-03-30:
I'm having a very weird problem that I know is not related to the code I'm running. I have a bunch of ESP32 Dev kits I'm using for a project, they're made by doit.am, and about half of them refuse to boot into run mode unless I press the EN/FLASH button. They do this regardless of the code, I've checked with several different example sketches and they all exhibit the same issue. I need a fix for this as these will be deployed to a customer very soon. I never noticed this issue because my test modules weren't doing it, but I don't have enough time to get different modules. Furthermore the doit.am boards are a different pinout than pretty much every other ESP32 module, so I'd have to redesign my carrier boards and that's not possible. So if anyone can help me I would greatly appreciate it. 

I'll run through what I have seen so far.

* Plug in usb cable
* Put ESP32 into programming mode by pressing BOOT and EN at the same time
* Load any sketch or example code
* ESP32 sits there and does nothing until you press EN/RESET. 
* Once reset manually, ESP32 works normally. 
* When next powered up, boards sit there and do nothing until you press EN/RESET
* This mode is not programming mode as I still have to press BOOT and EN at the same time to load sketches - otherwise esptool.py times out.

I have noticed a great deal of variation in how the boards act. All were bought at the same time and are presumably the same 'batch'. I have 25 of them, and about half of them have one form or another of this issue. Some of the boards are able to be reset by the USB serial line, others are not and must be put into programming mode manually. Others still work normally as one would expect.

I had them running on a benchtop power supply with an amperage reading, so I could tell something was up when they weren't pulling enough milliamps. When I press EN, the current consumption goes up to normal levels and the chips come alive and do their thing flawlessly. Also, rapidly power cycling them (and I mean very rapidly - less than 100 ms) will break them of their spell when in this low-power state and cause them to boot normally.

On the boards that will automatically load the program via the USB (that is to say, without having to manually put the boards in program mode) , this 'mode' almost appears to be programming mode, but on some of the other boards that require to be manually put into programming mode, I noticed that even when in this post-boot mode, they wouldn't accept a sketch unless put into programming mode manually again. 

So I'm stumped. I technically have a workaround - to rapidly powercycle all boards, but that is a very last resort to have to tell the customer to do this. 

I suspected the strapping pins were at play here, but I double and triple checked them and tied GPIO0 and GPIO2 to 3.3V and it has no effect whatsoever on this issue. 

If anyone has any ideas, I would appreciate them as I'm kind of at my wits end.

kurtow, 2018-03-30:
I have the same problem,

I bought three 'ESP32 Dev Kit WROOM32' boards from AZ-delivery.
Two boards are running normally.  The last board which I have bought three weeks ago
shows the same behaviour.

valentinAlkan, 2018-03-31:
I think the pullup on GPIO0 is the culprit. It looks like GPIO0 is floating when it should be pulled high unless the EN button is pressed. @kurtow, can you try hooking GPIO2 (blue LED pin) to 3.3V and see if you are still getting this issue? That should prevent it from entering boot mode even if GPIO0 is floating. I'm running a test where I hook GPIO2 to 3.3V and cycling the power on the ESP32 and wait for a response from some code that it's running. If after 30 seconds I get no response, I start the test again. I'm automating it, but if you don't mind just doing it a few times manually and seeing if you get any failures, I would appreciate it. I'm seeing about a 50% failure rate if you don't strap GPIO2 to 3.3V, but if you do strap it to 3.3V, it seems to work perfectly.

valentinAlkan, 2018-03-31:
Also strap GPIO12 to ground. I still got the occasional failure but after strapping GPIO2 high and GPIO12 low, that seems to have taken care of it. I definitely think its weak pullup/down resistors.

valentinAlkan, 2018-04-01:
Been looking into it some more, turns out it may have been related to a power supply that wasn't providing a good 5V. Will do some more testing and then close if that's what it was.

helmut64, 2018-04-01:
I have a similar problem, no reset on power-on.  Battery and extern power is fine, I reviewed this on the scope. Interesting is when I hold the reset key during power-on and release it will will not reset. If I press reset after power-on it will do a proper reset.

On my own designed board I have no pullup in IO0 I because the docs says there is an internal pullup. I will try an extern pullup.

valentinAlkan, 2018-04-02:
I made an automated test setup whereby I can cut the 5V to the ESP32 by hooking the power through a relay. I have a raspberry pi controlling the relay, and a python script running on the pi that turns the relay on (powering the ESP32), and waits for 60 seconds for a ping from the ESP32. If the pi gets the ping from the ESP32, it logs it as a successful boot, and then cuts the power for 2 seconds, starts it up again and repeats. If it gets no ping in 60 seconds, it counts it as a failed boot.

All the ESP32 does is connect to a wifi network and send a packet to a specific IP (the pi). 

I've tested with a couple different power supply setups. On my original 5v supply (https://www.cui.com/product/resource/p78a-0500.pdf), I got about a 40% boot rate with GPIO2 floating. With GPIO2 strapped down, I consistently got a 98% boot rate. When I bypassed by 5V regulator and used a benchtop power supply, I got 100% successful boots over a few thousand cycles, with GPIO2 strapped low and also with it left floating. 

I then started running the ESP32 off the raspberry pi's regulated 5v supply and also got a 100% boot rate over 3600 cycles and counting, with GPIO2 floating. 

I am sufficiently satisfied that the issue was caused by a noisy power supply, however it seems that these particular dev boards are quite sensitive to power fluctuations.

closed 2018-04-02 by valentinAlkan

dtboy1995, 2018-06-07:
[esp32 devkitC V4] What do i need to do if i want the power to start automatically, instead of pressing EN

dtboy1995, 2018-06-07:
The usb power went off and the program did not start.
But sometimes the program will run after the EN button is pressed
Who knows why, ask for help
Thanks

helmut64, 2018-06-07:
Messure via a scope the level and timing of VDD, EN and IO0 during power on. This may explains it.

dtboy1995, 2018-06-07:
@helmut64 Thank you for your reply. 
Is there any way to start the program without pressing the en button?

helmut64, 2018-06-07:
@dtboy1995 do you I recommended, show me the scope picture of the startup and I may can tell you what is wrong.

ukpiglet, 2018-11-21:
Old post I know, but when getting the same issue it was resolved by increasing the input voltage by 0.3v

helmut64, 2018-11-22:
@ukpiglet I fixed the problem by adding a 10k pullup to IO0 (PROG) on my own board, the internal MCU pullup does not work reliable on reset. On EN I have also a 10k Pullup and a 100nF EN to GND. With this combination the power-on reset timing works for me.

epietari, 2019-01-02:
Testing with oscilloscope shows that GPIO0 goes down (at least sometimes) after EN goes up while it should happen in the other order. The correct thing would be to change the original booting software.
But adding a capacitor 1uF ceramic between EN and GND works also.

gustavosinbandera1, 2019-03-06:
normally this type of board let me flashing the first time when I plug in the board into USB port  with out press program mode button, but is the only time where I can do it, I found out the very easy solution to fix this problem , the way is reset the USB port before flashing device. Put the next code into a file called usbrest.c

1. compile the program 
    $ cc usbreset.c -o usbreset
2. Get the bus and Device ID of the USB device you want to reset
    $ lsusb
    Bus 002 Device 003: ID 0fe9:9010 DVIC0
3.Make our compiled program executable
  chmod +x usbreset
4. execute the program with sudo privilege
   sudo ./usbreset /dev/bus/usb/002/003

in this point you we have our USB port reset, after that you could execute the next command in order to flashing  esp device
sudo ./usbreset/dev/bus/usb/002/003 && make flash

#include <stdio.h>
#include <unistd.h>
#include <fcntl.h>
#include <errno.h>
#include <sys/ioctl.h>

#include <linux/usbdevice_fs.h>

int main(int argc, char **argv)
{
    const char *filename;
    int fd;
    int rc;

    if (argc != 2) {
        fprintf(stderr, "Usage: usbreset device-filename\n");
        return 1;
    }
    filename = argv[1];

    fd = open(filename, O_WRONLY);
    if (fd < 0) {
        perror("Error opening output file");
        return 1;
    }

    printf("Resetting USB device %s\n", filename);
    rc = ioctl(fd, USBDEVFS_RESET, 0);
    if (rc < 0) {
        perror("Error in ioctl");
        return 1;
    }
    printf("Reset successful\n");

    close(fd);
    return 0;
}

dj44, 2020-06-12:
a 10µF capacitor with (-) at GND and (+) at EN did the job for me ... : 100% autoboot after  coldstart. :)

LouDnl, 2020-12-30:
> a 10µF capacitor with (-) at GND and (+) at EN did the job for me ... : 100% autoboot after coldstart. :)

Tried all the other options and this was the one that fixed it for me.

eos1d3, 2022-01-17:
> a 10µF capacitor with (-) at GND and (+) at EN did the job for me ... : 100% autoboot after coldstart. :)

10µF may be a bit crazy, I tried with 0.1µF and works very well.

bantugam, 2022-03-19:
Hi,
I had the power-up issue with my AZ-Delivery "ESP32-D1-R32" as well. With PC USB power everything was working fine but when I used a normal USB power supply (without PC) the board did not come [up.] After searching, reading and testing a lot, I found the following very simple solution: at least for this particular board the GPIO0 has to be connected to 3.3V. Since I did that it comes up every time.
[image]

TNeutron, 2022-12-05:
> Also strap GPIO12 to ground. I still got the occasional failure but after strapping GPIO2 high and GPIO12 low, that seems to have taken care of it. I definitely think its weak pullup/down resistors.

> I think the pullup on GPIO0 is the culprit. It looks like GPIO0 is floating when it should be pulled high unless the EN button is pressed. @kurtow, can you try hooking GPIO2 (blue LED pin) to 3.3V [...]

So, I was having a very tough time looking for a booting issue for a project. The Espressif documentation does mention IO2 and IO12 but they don't emphasize it enough which is why I used IO2 elsewhere and that was causing booting issues. I hereby verify and highly recommend what @valentinAlkan said. I have found it working and it just saved my ass. 

Thank you so much.

inventivemejo, 2024-12-12:
If anybody is facing this issue with a good quality dev board, as a first step I would suggest disconnecting any components connected to GPIO pins 0,2 and 15.