espressif/arduino-esp32 #1276: How can i clear esp32 ble rx and tx buffer?

aliMesutInce, 2018-03-30:
### Hardware:
Board:							ESP32 WROOM
IDE name:					        Arduino IDE 1.85 
Flash Frequency:					40Mhz
Upload Speed:					115200

I am working esp32 ble comminication. I creat the ble comminication one master and one slave ESP32 WROOM device. But i have a problem. Master is sending message until slave send "ok" message. So esp32 rx and tx buffer fill in more same data package so another loop tour that issue creat a problem. How can i clear esp32 ble rx and tx buffer?

Thanks for your interest.

malaimoo, 2018-06-06:
@aliMesutInce Hello ,Have you solved this problem? Did you find some method to clear the ble send buffer?

closed 2019-08-15 by stale[bot]

baptistepattyn, 2020-04-27:
Has anyone been able to solve this error?

xelons, 2020-09-05:
Interested as well, is there a way to clear the rx tx buffer? especially when sending out notifications, I always receive the last notification value instead of the fresh new one.