espressif/arduino-esp32 #1277: Not an issue, just trying to help with a Wiki

beegee-tokyo, 2018-03-31:
As this repo doesn't have a Wiki attached, I started to go through the issues regularly and collected solved issues in my [ESP32 Arduino Wiki](https://desire.giesecke.tk/index.php/2018/01/30/esp32-wiki-entries/).
Hope @me-no-dev will add the link to the Wiki somewhere here in this repo. 
Until then I will update this comment with the list of the current entries of the Wiki. Hope it helps you to find out if someone else has already solved your problem.    
### Tips and tricks
[Change global variables from ISR](http://desire.giesecke.tk/index.php/2018/01/30/change-global-variables-from-isr/)
[Suppress debug output during boot](http://desire.giesecke.tk/index.php/2018/01/30/suppress-debug-output-during-boot/)
[Get the last reset reason](http://desire.giesecke.tk/index.php/2018/01/30/get-the-last-reset-reason/)
[Sleep mode examples](http://desire.giesecke.tk/index.php/2018/01/30/sleep-mode-examples/)
[Multithreading support with xTaskCreate](http://desire.giesecke.tk/index.php/2018/01/30/multithreading-support-with-xtaskcreate/)
[Missing WebServer](http://desire.giesecke.tk/index.php/2018/01/30/missing-webserver/)
[Change partition size](http://desire.giesecke.tk/index.php/2018/01/30/change-partition-size/)
[Expansion of NVS partition](https://desire.giesecke.tk/index.php/2018/03/30/expansion-of-nvs-partition-requires-adapt-default-app-partition-address-in-esptool/)
[Simultaneously use of two I2C buses - TwoWire interfaces](http://desire.giesecke.tk/index.php/2018/01/30/simultaneously-use-of-two-i2c-buses-twowire-interfaces/)
[Websocket](http://desire.giesecke.tk/index.php/2018/01/30/websocket-client-and-server/)
[OneWire alternative library](http://desire.giesecke.tk/index.php/2018/01/31/onewire-alternative-library/)
[Switch off Bluetooth and WiFi](http://desire.giesecke.tk/index.php/2018/02/05/switch-off-bluetooth-and-wifi/)
[Get stored WiFi AP and STA info](http://desire.giesecke.tk/index.php/2018/02/14/get-stored-wifi-ap-and-sta-info/)
[Deep-Sleep needs WiFi and Bluetooth switched OFF for low current consumption](http://desire.giesecke.tk/index.php/2018/02/15/deep-sleep-needs-wifi-switched-off-for-low-current-consumption/)
[Ping library](http://desire.giesecke.tk/index.php/2018/02/18/esp32-ping-library/)
[SSDP - Simple Service Discovery Protocol](http://desire.giesecke.tk/index.php/2018/02/18/esp32-ssdp-simple-service-discovery-protocol/)
[Initialize Bluetooth Serial and BLE server](http://desire.giesecke.tk/index.php/2018/02/27/initialize-bluetooth-serial-and-ble-server/)
[TTGOV2 SD card support](http://desire.giesecke.tk/index.php/2018/03/10/ttgov2-sd-card-support/)
### Arduino IDE specific
[Alternative to install arduino-esp32 for Arduino IDE (Windows)](http://desire.giesecke.tk/index.php/2018/02/27/alternative-to-install-arduino-esp32-for-arduino-ide-windows/)
[ESP32 framework installation on Windows XP](http://desire.giesecke.tk/index.php/2018/03/10/esp32-framework-installation-on-windows-xp/)
### PlatformIO specific
[Enable log levels](http://desire.giesecke.tk/index.php/2018/01/30/enable-log-levels-in-platformio/)
### Common problems
[ESP32 not showing in serial port (MAC Desktop/Laptop problem)](http://desire.giesecke.tk/index.php/2018/01/30/esp32-not-showing-in-serial-port-mac-desktop-laptop-problem/)
[analogRead() triggers unrelated interrupt](http://desire.giesecke.tk/index.php/2018/01/30/analogread-triggers-unrelated-interrupt/)
[/avr-pgmspace.h: No such file or directory](http://desire.giesecke.tk/index.php/2018/01/30/pgmspace-h-no-such-file-or-directory/)
[ADC2 channel cannot be used when WiFi is in use](http://desire.giesecke.tk/index.php/2018/01/30/adc2-channel-cannot-be-used-when-wifi-is-in-use/)
[ESPNow not working after connection to WiFi AP](http://desire.giesecke.tk/index.php/2018/01/30/espnow-and-connection-to-wifi-ap/)
[Missing DNS settings when using static IP](http://desire.giesecke.tk/index.php/2018/01/03/missing-dns-settings-when-using-static-ip/)
[WiFi connection problem - NVS corrupted](http://desire.giesecke.tk/index.php/2018/02/26/wifi-connection-problem-nvs-corrupted/)
[WiFi Auto Reconnect problems](http://desire.giesecke.tk/index.php/2018/03/02/wifi-auto-reconnect-problems/)
[Problem with Adafruit FONA library](http://desire.giesecke.tk/index.php/2018/03/02/problem-with-adafruit-fona-library/)
[WiFi AP and STA active at the same time](http://desire.giesecke.tk/index.php/2018/03/14/wifi-ap-and-sta-active-at-the-same-time/)
[Important when using the Preferences library - key name length limited to 15 characters!](https://desire.giesecke.tk/index.php/2018/03/31/important-when-using-the-preferences-library/)

palomarcoder, 2018-04-08:
This is helpful, thanks for your contribution.  It would be nice if you'd include a link to the associated github issue so people can look for updates.

beegee-tokyo, 2018-04-20:
Updated the structure, hope it makes it easier to find stuff:

### Network tips and problems
[Missing WebServer](http://desire.giesecke.tk/index.php/2018/01/30/missing-webserver/)
[Websocket](http://desire.giesecke.tk/index.php/2018/01/30/websocket-client-and-server/)
[Switch off Bluetooth and WiFi](http://desire.giesecke.tk/index.php/2018/02/05/switch-off-bluetooth-and-wifi/)
[Get stored WiFi AP and STA info](http://desire.giesecke.tk/index.php/2018/02/14/get-stored-wifi-ap-and-sta-info/)
[Ping library](http://desire.giesecke.tk/index.php/2018/02/18/esp32-ping-library/)
[SSDP - Simple Service Discovery Protocol](http://desire.giesecke.tk/index.php/2018/02/18/esp32-ssdp-simple-service-discovery-protocol/)
[ESPNow not working after connection to WiFi AP](http://desire.giesecke.tk/index.php/2018/01/30/espnow-and-connection-to-wifi-ap/)
[Missing DNS settings when using static IP](http://desire.giesecke.tk/index.php/2018/01/03/missing-dns-settings-when-using-static-ip/)
[WiFi connection problem - NVS corrupted](http://desire.giesecke.tk/index.php/2018/02/26/wifi-connection-problem-nvs-corrupted/)
[WiFi Auto Reconnect problems](http://desire.giesecke.tk/index.php/2018/03/02/wifi-auto-reconnect-problems/)
[WiFi AP and STA active at the same time](http://desire.giesecke.tk/index.php/2018/03/14/wifi-ap-and-sta-active-at-the-same-time/)

### I2C, SPI tips and problems
[How to use stickbreaker’s I2C improved code](https://desire.giesecke.tk/index.php/2018/04/20/how-to-use-stickbreakers-i2c-improved-code/)
[Simultaneously use of two I2C buses - TwoWire interfaces](http://desire.giesecke.tk/index.php/2018/01/30/simultaneously-use-of-two-i2c-buses-twowire-interfaces/)
[OneWire alternative library](http://desire.giesecke.tk/index.php/2018/01/31/onewire-alternative-library/)

### Bluetooth and BLE tips and problems
[Switch off Bluetooth and WiFi](http://desire.giesecke.tk/index.php/2018/02/05/switch-off-bluetooth-and-wifi/)
[Initialize Bluetooth Serial and BLE server](http://desire.giesecke.tk/index.php/2018/02/27/initialize-bluetooth-serial-and-ble-server/)

### Programming tips and problems
[Change global variables from ISR](http://desire.giesecke.tk/index.php/2018/01/30/change-global-variables-from-isr/)
[Sleep mode examples](http://desire.giesecke.tk/index.php/2018/01/30/sleep-mode-examples/)
[Deep-Sleep needs WiFi and Bluetooth switched OFF for low current consumption](http://desire.giesecke.tk/index.php/2018/02/15/deep-sleep-needs-wifi-switched-off-for-low-current-consumption/)
[Multithreading support with xTaskCreate](http://desire.giesecke.tk/index.php/2018/01/30/multithreading-support-with-xtaskcreate/)
[analogRead() triggers unrelated interrupt](http://desire.giesecke.tk/index.php/2018/01/30/analogread-triggers-unrelated-interrupt/)
[/avr-pgmspace.h: No such file or directory](http://desire.giesecke.tk/index.php/2018/01/30/pgmspace-h-no-such-file-or-directory/)
[ADC2 channel cannot be used when WiFi is in use](http://desire.giesecke.tk/index.php/2018/01/30/adc2-channel-cannot-be-used-when-wifi-is-in-use/)
[Important when using the Preferences library - key name length limited to 15 characters!](https://desire.giesecke.tk/index.php/2018/03/31/important-when-using-the-preferences-library/)
[Important when using the Preferences library - tag length limited to 1984 characters!](https://desire.giesecke.tk/index.php/2018/04/20/preferences-library-max-tag-length/)

### Debugging tips and problems
[Suppress debug output during boot](http://desire.giesecke.tk/index.php/2018/01/30/suppress-debug-output-during-boot/)
[Get the last reset reason](http://desire.giesecke.tk/index.php/2018/01/30/get-the-last-reset-reason/)

### Partition tips and problems
[Change partition size](http://desire.giesecke.tk/index.php/2018/01/30/change-partition-size/)
[Expansion of NVS partition](https://desire.giesecke.tk/index.php/2018/03/30/expansion-of-nvs-partition-requires-adapt-default-app-partition-address-in-esptool/)

### ESP32 module specific stuff
[TTGOV2 SD card support](http://desire.giesecke.tk/index.php/2018/03/10/ttgov2-sd-card-support/)

### Arduino IDE specific
[Alternative to install arduino-esp32 for Arduino IDE (Windows)](http://desire.giesecke.tk/index.php/2018/02/27/alternative-to-install-arduino-esp32-for-arduino-ide-windows/)
[ESP32 framework installation on Windows XP](http://desire.giesecke.tk/index.php/2018/03/10/esp32-framework-installation-on-windows-xp/)
[get.exe/get.py fails to download required files](https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/)
[Change partition size (Arduino IDE)](https://desire.giesecke.tk/index.php/2018/04/20/change-partition-size-arduino-ide/)

### PlatformIO specific
[Enable log levels](http://desire.giesecke.tk/index.php/2018/01/30/enable-log-levels-in-platformio/)
[Change partition size (PlatformIO)](https://desire.giesecke.tk/index.php/2018/04/20/change-partition-size-platformio/)

### Common problems
[ESP32 not showing in serial port (MAC Desktop/Laptop problem)](http://desire.giesecke.tk/index.php/2018/01/30/esp32-not-showing-in-serial-port-mac-desktop-laptop-problem/)
[Problem with Adafruit FONA library](http://desire.giesecke.tk/index.php/2018/03/02/problem-with-adafruit-fona-library/)

closed 2019-08-15 by stale[bot]