espressif/arduino-esp32 #1280: get.exe download stuck while installing esp32 library

salmanabbasi112, 2018-03-31:
[image]

The download is not completing. Tried to run as administrator as well. Restarted computer and run get.exe. But still not working. Please help fast.

lbernstone, 2018-04-01:
If you look down in the comment of this [article](https://desire.giesecke.tk/index.php/2018/02/27/alternative-to-install-arduino-esp32-for-arduino-ide-windows/), @beegee-tokyo lists the urls to download the files.  Put them in the tools/dist folder and see if get.exe completes.

Onno-Dirkzwager, 2018-04-01:
salmanabbasi112 
in my case I had to disable the virus scanner temporary before clicking get.exe

beegee-tokyo, 2018-04-02:
Improved the instructions if [get.exe/get.py fails to download required files](https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/) in the [Wiki](https://desire.giesecke.tk/index.php/2018/01/30/esp32-wiki-entries/)

closed 2019-08-15 by stale[bot]

syndicated-8, 2022-03-19:
I'm having the same issue. The get.exe stays stuck downloading. Article link to download files separately doesn't work.