espressif/arduino-esp32 #1283: Tuning latency of WiFiUdp

mitsuhito, 2018-04-02:
### Hardware:
Board:							ESP32 DevkitC
Core Installation/update date:			9/Mar/2017
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Hi, I'm making a kind of instruments hardware. The main function of esp32 is watching piezo sensor signal and send udp (OSC) packet if it's triggered. It's almost OK but I measured the latency by using [Is It Snappy?](http://isitsnappy.com/), it took about 60ms~ at least. Does anyone know about low latency way to send or measuring the actual sending time inside arduino-esp32 code?

### Sketch:
```cpp
// Arduino IDE
// dependency version: 1.9.0-beta
#include "Arduino.h"

// arduino-esp32
// dependency version: https://github.com/espressif/arduino-esp32/commit/3a4ec66d41615cbb1c3e830cb6e761cdc4cca9d3
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include <driver/adc.h>
#include <WiFi.h>
#include <WiFiUdp.h>
#include <ESPmDNS.h>
#include <ArduinoOTA.h>

// CNMAT OSC
// dependency version: https://github.com/CNMAT/OSC/releases/tag/1.3.5
#include <OSCMessage.h>
#include <OSCBundle.h>

char ssid[] = "wlan_ssid_here";
char pass[] = "wlan_pass_here";

WiFiUDP udp;
unsigned long time_stamp = 0;
uint8_t led_pin = 25;
uint8_t piezo_pin = 39;
const IPAddress dst_ip(192, 168, 20, 21);
const unsigned int dst_port = 9999;

void setup()
{
  Serial.begin(115200);

  pinMode(piezo_pin, INPUT);
  analogSetPinAttenuation(piezo_pin, ADC_0db); // because LM358N outputs 0v to Vcc-1.5v
  pinMode(led_pin, OUTPUT);

  // delete cache
  WiFi.disconnect();

  Serial.println();
  Serial.print("Connecting to ");
  Serial.println(ssid);
  WiFi.begin(ssid, pass);
  delay(1000);
  Serial.println("IP address: ");
  Serial.println(WiFi.localIP());

  Serial.flush();
}

void loop()
{
  int raw = analogRead(piezo_pin);
  if (raw > 4000)
  {
    if (micros() - time_stamp > 10000)
    {
      digitalWrite(led_pin, HIGH);
      if (WiFi.isConnected())
      {
        OSCMessage msg("/test");
        msg.add(raw);
        udp.beginPacket(dst_ip, dst_port);
        msg.send(udp);
        udp.endPacket();
        msg.empty();
      }
      digitalWrite(led_pin, LOW);
    }
  }
}

/*
// MAX/MSP patch

<pre><code>
----------begin_max5_patcher----------
412.3ocuTsraCBCD7L7UXsmoowDRDza86npJx.tINhXPFSJsQ4eu1qg7npHE
PQ8xJ40i2clYW3nuGjV1xqAxKj2HddG887vT1Ddcm8f8r1rBVMBCjM6S4JHv
cUYitfq0eUwc0.DRMDPfTlbCPduCk4MBoAGVAZWxJlNaqPtYshmocudUzrkA
DJMd17.xx41XnIdtNeTJ00huwdQCMW6RKxQlUlt6oX3JnR1dDJ7pRvJfKjww
ZjMgmYixfVyUq4RVZA9t41qN46aCA2q8v+zvi9do4snz.kokbxyZdsdPqy5a
C3YgC6YQnMEuvFM923srnQaYSvVRaz5R4fReZqKNoSiv0kjj9345bkHW72Rg
93lvM4UFlwEG3FRjjL7TdhxzEcpbjSX5XmvWaKXefBg72+l.YpM+sdUcYiJq
Wrc6WjKrM27EfPxzBy1vELw2fYqHOmiW2KlbQs8ix7g+p7QxlE+argdGrIZh
rwM3XUUG3p5tRhDwrNuqTYOtJ.OJjti3NHn3GD83oXS.lxrSpMKjMJ2tS6pH
v21nS9+.k22b1B
-----------end_max5_patcher-----------
</code></pre>
*/
```

lbernstone, 2018-04-03:
Why don't you put the lights on and off just around the udp piece?  Creating and destroying the OSC object probably takes more time than the udp send. I don't have a fancy iPhone to test it.

lbernstone, 2018-04-03:
I just tested with a micros timer around the udp begin and end, and I get an average of 214 microseconds for the udp send.  Don't try to build and destroy the OSCmessage every loop.  Trust that empty works.  If you are really trying to optimize for speed on this thing, you will need an external comparator, as the esp32 does not have one, and polling is a very slow (and for the esp32, IMHO inaccurate) way to try to catch changes in an analog signal.

mitsuhito, 2018-04-15:
@lbernstone thanks for your suggestion. 
> I measured the latency by using Is It Snappy?, it took about 60ms~ at least.
 
I just figured out why it took so long. The latency depends on display refresh rate and software rendering speed. So your test is right. I'll close this issue.

closed 2018-04-15 by mitsuhito