espressif/arduino-esp32 #1284: esp32 always reseting after uploading the code

MaDeireiroGame, 2018-04-03:
esptool.py v2.1
Connecting........_
Chip is ESP32D0WDQ6 (revision 1)
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 4MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 2849.4 kbit/s)...
Hash of data verified.
Compressed 14128 bytes to 9196...

Writing at 0x00001000... (100 %)
Wrote 14128 bytes (9196 compressed) at 0x00001000 in 0.8 seconds (effective 135.7 kbit/s)...
Hash of data verified.
Compressed 164128 bytes to 84591...

Writing at 0x00010000... (16 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (83 %)
Writing at 0x00024000... (100 %)
Wrote 164128 bytes (84591 compressed) at 0x00010000 in 7.5 seconds (effective 174.0 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 122...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (122 compressed) at 0x00008000 in 0.0 seconds (effective 819.2 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting...

lbernstone, 2018-04-03:
Yes, it has to reset to run your new code. This is not an error.

MaDeireiroGame, 2018-04-03:
But nothing happens um Trying to run a wifi scan but only give me radon character's

lbernstone, 2018-04-03:
Are you sure you have the correct baud set in your serial monitor?

MaDeireiroGame, 2018-04-03:
I will check today when I arrive home

MaDeireiroGame, 2018-04-03:
Im really sorry for my inexperience, yes the problem was the baud thanks a lot :+1:

closed 2018-04-03 by MaDeireiroGame