espressif/arduino-esp32 #1285: Permission denied creating temp file

iotlearner0level, 2018-04-03:
somehow, esp32 installation doesn't work for me on windows 10/arduino 1.8.5. esp8266 works fine. i think it has something to do with environment variables. while executing tools>get.exe as normal user, i get error can't open temporary directory. As admin user works ok.

inside arduino, blinky doesn't work. it fails when compiler creating first output.

debug log as below:

C:\Program Files (x86)\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ABC\AppData\Local\Arduino15\packages -hardware C:\Users\ABC\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ABC\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ABC\Documents\Arduino\libraries -fqbn=espressif:esp32:lolin32:FlashFreq=80,UploadSpeed=921600 -ide-version=10805 -build-path C:\Users\ABC\AppData\Local\Temp\arduino_build_230425 -warnings=none -build-cache C:\Users\ABC\AppData\Local\Temp\arduino_cache_134231 -prefs=build.warn_data_percentage=75 -verbose C:\Users\ABC\AppData\Local\Temp\untitled931719074.tmp\sketch_apr03a\sketch_apr03a.ino
C:\Program Files (x86)\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files (x86)\Arduino\hardware -hardware C:\Users\ABC\AppData\Local\Arduino15\packages -hardware C:\Users\ABC\Documents\Arduino\hardware -tools C:\Program Files (x86)\Arduino\tools-builder -tools C:\Program Files (x86)\Arduino\hardware\tools\avr -tools C:\Users\ABC\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files (x86)\Arduino\libraries -libraries C:\Users\ABC\Documents\Arduino\libraries -fqbn=espressif:esp32:lolin32:FlashFreq=80,UploadSpeed=921600 -ide-version=10805 -build-path C:\Users\ABC\AppData\Local\Temp\arduino_build_230425 -warnings=none -build-cache C:\Users\ABC\AppData\Local\Temp\arduino_cache_134231 -prefs=build.warn_data_percentage=75 -verbose C:\Users\ABC\AppData\Local\Temp\untitled931719074.tmp\sketch_apr03a\sketch_apr03a.ino
Using board 'lolin32' from platform in folder: C:\Users\ABC\Documents\Arduino\hardware\espressif\esp32
Using core 'esp32' from platform in folder: C:\Users\ABC\Documents\Arduino\hardware\espressif\esp32
Detecting libraries used...
"C:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_flash" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fexceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_LOLIN32 -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="LOLIN32" -DARDUINO_VARIANT="lolin32"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32\variants\lolin32" "C:\Users\ABC\AppData\Local\Temp\arduino_build_230425\sketch\sketch_apr03a.ino.cpp" -o "nul"
Generating function prototypes...
"C:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++" -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/nvs_f [... cut ...]
Error compiling for board WEMOS LOLIN32.

### Hardware:
Board:							?ESP32 lolin?
Core Installation/update date:			?11/jul/2017?
IDE name:							?arduino 1.8.5?
Flash Frequency:					?40Mhz?
Upload Speed:						?115200?

### Description:
Describe your problem here

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>

void setup() {
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

beegee-tokyo, 2018-04-03:
You can try to download the files manually as normaler user and then run get.exe as shown [here](https://desire.giesecke.tk/index.php/2018/04/02/get-exe-get-py-fails-to-download-required-files/)

iotlearner0level, 2018-04-03:
sorry maybe i wasn't clear enough. get.exe is only a part of the problem. since the get.exe step is required only onece, i executed it as administrator and we are good to go (files are downloaded)

unfortunately, the tool chain from within arduino can't access temporary folder, not sure why.

from windows shell, if i execute the compiler command as normal user, it fails:
 ./xtensa-esp32-elf-g++ -DESP_PLATFORM -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h" -DHAVE_CONFIG_H "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/config"
"-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bluedroid" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_trace" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/app_update" "-IC:\
Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bootloader_support" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/bt" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/driver" "-IC:\Users\AB
C\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp32" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/esp_adc_cal" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ethernet" "-IC:\Users\ABC\Document
s\Arduino\hardware\espressif\esp32/tools/sdk/include/fatfs" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/freertos" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/heap" "-IC:\Users\ABC\Documents\Arduino\hardwar
e\espressif\esp32/tools/sdk/include/jsmn" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/log" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mdns" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools
/sdk/include/mbedtls" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/mbedtls_port" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/incl
ude/nvs_flash" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/openssl" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spi_flash" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/sdmmc
" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/spiffs" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/tcpip_adapter" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/ulp" "-IC:\User
s\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/vfs" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wear_levelling" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/xtensa-debug-module" "-IC:\Us
ers\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/coap" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/console" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/expat" "-IC:\Users\ABC\Documents\
Arduino\hardware\espressif\esp32/tools/sdk/include/json" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/lwip" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/newlib" "-IC:\Users\ABC\Documents\Arduino\hardware\esp
ressif\esp32/tools/sdk/include/nghttp" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/soc" "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32/tools/sdk/include/wpa_supplicant" -std=gnu++11 -fno-exceptions -Os -g3 -Wpointer-arith -fex
ceptions -fstack-protector -ffunction-sections -fdata-sections -fstrict-volatile-bitfields -mlongcalls -nostdlib -w -Wno-error=unused-function -Wno-error=unused-but-set-variable -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wno-unused-parameter -Wno-sign
-compare -fno-rtti  -c -w -x c++ -E -CC -DF_CPU=240000000L -DARDUINO=10805 -DARDUINO_LOLIN32 -DARDUINO_ARCH_ESP32 -DARDUINO_BOARD="LOLIN32" -DARDUINO_VARIANT="lolin32"  -DESP32 -DCORE_DEBUG_LEVEL=0 "-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32\cores\esp32" "
-IC:\Users\ABC\Documents\Arduino\hardware\espressif\esp32\variants\lolin32" "C:\Users\ABC\AppData\Local\Temp\arduino_build_287216\sketch\sketch_apr03a.ino.cpp" -o "C:\Users\ABC\AppData\Local\Temp\arduino_build_287216\preproc\ctags_target_for_gcc_minus_e.cpp"
cc1plus.exe: fatal error: opening output file C:\Users\ABC\AppData\Local\Temp\arduino_build_287216\preproc\ctags_target_for_gcc_minus_e.cpp: Permission denied

same error from within arduino logs.

however, if i launch a cmd.exe with Administrator rights, i works ok.

to test the hypothesis, i launched Arduino with Administrator rights, it works without any problem. But i don't think, running arduino ide with administrator rights is a wise idea...

i've this problem with esp32 on arduino, rest every thing seems to be working properly as a regular user...

but it is very frustrating.

beegee-tokyo, 2018-04-03:
I am so glad that I still stick to Windows 7, much less problems. And if I need a new computer most likely I will switch complete to Linux.

Did you try this: [How to Configure User Permissions for Files and Folder in Windows 10](https://www.howto-connect.com/windows-10-configure-user-permissions-for-files-folders/) for the **_C:\Users\ABC\AppData\Local\Temp\_** folder?

iotlearner0level, 2018-04-03:
thanks for posting that link but did not help. i've infact added EVERYONE read/write permissions. doesn't help. Strange that such a problem occurred to me only with the esp32 toolchain, that why i posted in this issues tracker...not sure about the origin though. 

and by the way, except for the "automatic biannual upgrade" part, i find windows 10 to be much bettter than windows 7 ;-) just my opinion. linux is just too much "work" for even the most trivial tasks......(sorry do not mean to offend anyone)

beegee-tokyo, 2018-04-03:
About Linux/Windows no offense taken, everybody has his preferences.

About you problem, I have no further ideas.

AriaMoKr, 2018-07-15:
I had a similar permission problem in Windows 10 with running get.exe "Access is denied.", to fix it go to the Arduino directory in C:\Users\USERNAME\Documents, right click and click properties, go to the Security tab, click Advanced, make sure that your user account has Allow and Full control and applies to This folder, subfolders and files. SYSTEM and Administrators will probably show up but can be ignored (those accounts will likely have the same permissions). Then check "Replace all child object permission...." and click OK and confirm.
My guess is that some of the files or folders when checked out (probably under hardware) ended up with incorrect permissions.

closed 2019-08-15 by stale[bot]