espressif/arduino-esp32 #1286: CPU Freq Setting

Natalie88, 2018-04-03:
### Hardware:
Board:	ESP32 Generic						?ESP32 Dev Module?
Core Installation/update date:			25/03/2018
IDE name:							?Arduino IDE? 
Flash Frequency:					?80Mhz?
Upload Speed:						?115200?

### Description:
I am calling rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M); in setup() just one time to adjust the cpu freq to 80Mhz . after deep sleep the frequency is back to 240? I thought the freq change will survive the deep sleep? 
if I call the function each time the ESP wake up it hangs after several sleep periods, I think due to WDT reset as  rtc_clk_cpu_freq_set consume a huge amount of time. 
I have tried to yield() and put at delay(300) after the function call, it hang less but however it still hang sometimes.

### Sketch:
```cpp

//Change the code below by your sketch
#include <Arduino.h>
RTC_DATA_ATTR  bool freqIsChanged= false;
void setup() {
  if(! freqIsChanged){
    rtc_clk_cpu_freq_set(RTC_CPU_FREQ_80M);
    yield();
    RTCdata.freqIsChanged = true;  
  }
  Serial.begin(115200);
  Serial.println();
  Serial.println("Booted");
  Serial.flush();
  Serial.println(String(rtc_clk_cpu_freq_get()));
  Serial.flush();
// do stuff 

// Go to deep sleep
}

void loop() {
}
```

### Debug Messages:
```
Enable Core debug level: Debug on tools menu of Arduino IDE, then put the serial output here 
```

lbernstone, 2018-04-03:
If you want the speed fixed at 80MHz, I recommend you follow the instructions at https://github.com/espressif/arduino-esp32/issues/1142 or just download the file from https://github.com/lbernstone/arduino-esp32-80MHz. Changing the cpu speed is going to be rough on the arduino timers.

Natalie88, 2018-04-09:
@lbernstone  Thank you for your reply
I changed the mentioned file but I am still getting 3 when I call this function 
rtc_clk_cpu_freq_get(), this means that the Freq is still 240Mhz

lbernstone, 2018-04-15:
I just updated to the latest code and tested it.  I get 1 for the frequency.

beats0126, 2018-11-29:
@lbernstone  the 80mhz file cannot use for v1.0.1-rc1
can u update it?

atanisoft, 2018-11-29:
@beats0126 direct link to the instructions: https://github.com/espressif/arduino-esp32/issues/1142#issuecomment-367867856

closed 2019-08-15 by stale[bot]