espressif/arduino-esp32 #1287: Arduino.exe crashes after installing the ESP32 core files.

pratikguru, 2018-04-03:
I have installed the arduino app in drive E. The ESP32 core files as suggested by espressif has been placed in the suggested path and cloned. 

Once the get.exe is compiled it installs the necessary files but fails to run arduino.exe, the app crashes on start.

Running on Windows 10
Arduino ver 1.8.5

per1234, 2018-04-03:
If you move the ESP32 core folder somewhere else does the problem go away?

If so, please move it back to the original location and then run the program arduino_debug.exe (which you will find in your Arduino IDE installation folder alongside arduino.exe) from the command line and then paste the output here. Hopefully that will provide some information on the cause of the IDE not starting.

pratikguru, 2018-04-03:
Hey,

The problem has been solved. I did try to put the installation in another directory and it still failed.'

**What I tried?**
I added the files to:
C://Documents/Arduino/Hardware/Espresif/esp32/
And it worked. I believe adding the core files to the installation directory's hardware folder has issues with the permission. 

**Note:** I manually added the espresif/esp32 folder in the Documents/Arduino folder, but this can also be done through the git bash and git gui process as described in the original installation manual.

Thanks!

closed 2019-08-15 by stale[bot]