espressif/arduino-esp32 #1290: Firebeetle ESP32 A fatal error occurred: Failed to connect to ESP32

pacav69, 2018-04-04:
Unable to get firebeetle esp32 into program mode, tried different cables still not working.
Reinstalled CP210x Silicon Labs driver with version 4 as version 5 does not work on a MAC with these boards.

 appears in device list as:

```cpp

/dev/cu.usbmodem1421

```

### Hardware:
Board:							Firebeetle ESP32
Core Installation/update date:			Feb 17  2017 get.py
IDE name:							Arduino IDE
Flash Frequency:					      80Mhz
Upload Speed:						115200

### Description:
Describe your problem here

### Sketch:
```cpp
blink.ino

//Change the code below by your sketch
/*
  Blink

  Turns an LED on for one second, then off for one second, repeatedly.

  Most Arduinos have an on-board LED you can control. On the UNO, MEGA and ZERO
  it is attached to digital pin 13, on MKR1000 on pin 6. LED_BUILTIN is set to
  the correct LED pin independent of which board is used.
  If you want to know what pin the on-board LED is connected to on your Arduino
  model, check the Technical Specs of your board at:
  https://www.arduino.cc/en/Main/Products

  modified 8 May 2014
  by Scott Fitzgerald
  modified 2 Sep 2016
  by Arturo Guadalupi
  modified 8 Sep 2016
  by Colby Newman

  This example code is in the public domain.

  http://www.arduino.cc/en/Tutorial/Blink
*/

// the setup function runs once when you press reset or power the board
void setup() {
  // initialize digital pin LED_BUILTIN as an output.
  pinMode(LED_BUILTIN, OUTPUT);
}

// the loop function runs over and over again forever
void loop() {
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(800);                       // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(100);                       // wait for a second
}
```

### Debug Messages:
```
A fatal error occurred: Failed to connect to ESP32: Timed out waiting for packet header
*** [upload] Error 2```

pacav69, 2018-04-14:
Found that this seems to be related to Mac and maybe linix not being able to communicate to the device. 
I got access to a windows 7 computer, using the
[install instructions for windows](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md)installed the latest Arduino IDE 1.8.5, installed git, installed ESP32 environment everything seemed fine, when plugged in the Firebeetle ESP32 board, it  attempted to install driver but failed so i downloaded the firebeetle32.inf file, opened the device manager and updated the driver by going to the firebeetle32.inf it then proceeded to install the driver. then used blink sketch to test  the connection and it uploaded the sketch but no blink of led. Press reset, nothing happened. By unplugging and replugging the USB cable it began to blink. Checking the serial output there was nothing displayed, after pressing reset still nothing. I uploaded the sketch again everything was ok, checked serial monitor nothing being displayed, pressing reset however, displayed reset message and showed 'ready for upload'.  

Still a mystery why it works on Win7 and not a MAC my only conclusion is that there is a driver issue.

doronby, 2018-12-04:
I am having the same issue.

It works one out of 6-10 times and the rest of the time it fails with this message:
esptool.py v2.3.1
Connecting........_____....._____....._____....._____....._____....._____....._____....._____....._____....._____

A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('\x08')
A fatal error occurred: Failed to connect to ESP32: Invalid head of packet ('\x08')

when it finally works it looks like this:

esptool.py v2.3.1
Connecting........___
Chip is ESP32D0WDQ6 (revision (unknown 0xa))
Features: WiFi, BT, Dual Core, VRef calibration in efuse
Uploading stub...
Running stub...
Stub running...
Configuring flash size...
Auto-detected Flash size: 8MB
Compressed 8192 bytes to 47...

Writing at 0x0000e000... (100 %)
Wrote 8192 bytes (47 compressed) at 0x0000e000 in 0.0 seconds (effective 5891.0 kbit/s)...
Hash of data verified.
Flash params set to 0x023f
Compressed 13696 bytes to 8946...

Writing at 0x00001000... (100 %)
Wrote 13696 bytes (8946 compressed) at 0x00001000 in 0.8 seconds (effective 136.6 kbit/s)...
Hash of data verified.
Compressed 174992 bytes to 88349...

Writing at 0x00010000... (16 %)
Writing at 0x00014000... (33 %)
Writing at 0x00018000... (50 %)
Writing at 0x0001c000... (66 %)
Writing at 0x00020000... (83 %)
Writing at 0x00024000... (100 %)
Wrote 174992 bytes (88349 compressed) at 0x00010000 in 7.9 seconds (effective 177.5 kbit/s)...
Hash of data verified.
Compressed 3072 bytes to 144...

Writing at 0x00008000... (100 %)
Wrote 3072 bytes (144 compressed) at 0x00008000 in 0.0 seconds (effective 1073.1 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

lbernstone, 2018-12-04:
@doronby : Please don't hijack issues.  Your intermittent connection problems are probably due to a bad cable, or modes that your flash can't handle.  Try another cable, reduce the upload speed, switch to DIO mode.

doronby, 2018-12-04:
I have done all of the above. 
All speeds available, 3 different cables, 2 different OSX machines.
All my other ESP32 variants work 100% of the time with the same cable and same machine. 
(This is the only ESP32 with an ARM USB chip (STM32F042F6P6) and I suspect the problem is there)

doronby, 2018-12-05:
I installed the latest Arduino and ESP32 framework as well as the board INF file on an old windows 7 PC I had laying around and the same board and cable **work 100% of the time**. This problem is IMHO an **OSX** and the  STM32F042F6P6.

pacav69, 2018-12-05:
@doronby if you read my post https://github.com/espressif/arduino-esp32/issues/1290#issuecomment-381297868 you will see that it is that DFRobot have not got a driver for the MAC OSX and will only work on a Windows 7 or later operatings system.

doronby, 2018-12-05:
The manufacturer emailed me their claims that they have no issue with their MAC but I do on both of our MACs.
The MAC recognized the serial dev port OK but is very fussy (success is 1 in 5 to 1 in 10 attempts) with programming the board.  
Serial monitor always works flawlessly on the same port at any speed... 
I suspect its the firmware of the STM32F042F6P6 which is used as the USB to Serial chip on this board.

g7uvw, 2019-06-20:
Did anyone get to the bottom of this?
I'm seeing the same issue on a mac with the Firebeetle esp32 V2.0
When it works, it seems to work well. When it fails it fails repeatedly. Adding a button to Boot0 did not help (was suggested elsewhere). You just have to keep trying and eventually it'll work.

pacav69, 2019-08-28:
i have abandoned using the firebeetle 32 as it is unreliable and have had no response from DFROBOT in resolving this issue. I now use nodemcu or other clones that don't use the STM32F042F6P6 chip.

ASTEG012, 2019-09-03:
I have the same problem with the Beetle ESP32 - which does not have a Reset and Enable button and sometimes uploads work and other times under the exact same circumstances it does not work for days

chanshukwong, 2019-09-10:
I just buy a Fire Beetle ESP32, when it starts to upload (after compile), I keep on pressing the boot button without releasing, it start to upload. Or otherwise, it will timeout. A stupid way to upload!

pacav69, 2019-11-09:
there is still no action or response by DFROBOT

closed 2020-01-24 by stale[bot]