espressif/arduino-esp32 #1292: is I2C_MUTEX_LOCK really necessesary?

terryjmyers, 2018-04-05:
The short issue: I'm having an issue implementing a project that fully utilized FreeRTOS instead of the standard Arduino "loop()" function because of the call to I2C_MUTEX_LOCK() in esp32-hal-i2c.c, and I think the solution is that it should be removed.

The details:
While I'm using this Arduino toolchain, I'm utilizing FreeRTOS the way it was designed by creating tasks and simply calling vTaskDelete(NULL) in loop().  Two of my tasks utilize the I2C bus (a DAC and an ADC).  I created my own mutex in order to arbitrate the I2C bus between these two asyncronous tasks.  However, I also need to use taskENTER_CRITICAL() as I don't want any FreeRTOS task to interupt until I've written to the I2C bus(Wire.Write()) AND ALSO read the data returned (Wire.RequestFrom()) as my ADC requires a register pointer change.  However, in the taskENTER_CRITICAL() documentation (https://www.freertos.org/taskENTER_CRITICAL_taskEXIT_CRITICAL.html) it says NOT to call any FreeRTOS API functions inside taskENTER_CRITICAL(), which esp32-hal-i2c.c will do when it arbitrates its own mutex.
I can't think of a reason why this toolchain would want to use a mutex considering most people who use this toolchain are not going to use FreeRTOS async tasks.  Therefore I think any FreeRTOS api calls such as this could be removed.  Anyone who wants to properly use FreeRTOS should be responsible for properly arbitrating the I2C bus between async tasks that want to use it.
esp32-hal-i2c.c does not properly call taskENTER_CRITICAL to disable FreeRTOS.  It shouldn't because the appropriate place would be in the users main code as a wrapper AROUND the outside of their I2C calls.  
For example a typical and simplified I2C read would be:
Wire.beginTransmission(_i2cAddress);
Wire.write(DATA)
Wire.endTransmission();
//Another async task could just in a perform its own call to Wire.requestFrom(_Another i2cAddress, (uint8_t)2); which may screw something up maybe?
value = Wire.requestFrom(_i2cAddress, (uint8_t)2);

However if I'm using other tasks that want to do the same thing and could interrupt at anytime I need to do my own semaphore arbitration and properly utilize taskENTER_CRITICAL like this:
xSemaphoreTake(handle);
taskENTER_CRITICAL()
Wire.beginTransmission(_i2cAddress);
Wire.write(DATA)
Wire.endTransmission();
value = Wire.requestFrom(_i2cAddress, (uint8_t)2);
taskEXIT_CRITICAL();
xSemaphoreGive(handle);

But I shouldn't do this because the FreeRTOS documentation says not to use any FreeRTOS api calls inside a taskENTER_CRITICAL() which  esp32-hal-i2c.c is going to do.  

So basically I don't think the I2C_MUTEX_LOCK() buys anybody anything when using loop() in a "linear" or single threaded application, and it messes people up who want to do things asynchronously.  Am I wrong?  What am I missing

stickbreaker, 2018-04-05:
@terryjmeyers `taskENTER_CRITICAL()` works by disabling interrupts. Why do you think you need a critical section around your `Wire()` calls? 
I2C is a slow bus. 100KHz .. 400KHz, disabling interrupts for hundreds of thousands of clock cycles Dangerous.

Chuck

versamodule, 2018-04-12:
@stickbreaker 

> taskENTER_CRITICAL() works by disabling interrupts. Why do you think you need a critical section around your Wire() calls?
> I2C is a slow bus. 100KHz .. 400KHz, disabling interrupts for hundreds of thousands of clock cycles Dangerous.

I2C is timing sensitive in looking for the ack signal. An IRQ can happen any time causing it to miss the ack signal.

stickbreaker, 2018-04-12:
@versamodule the i2c hardware peripheral handles ACK/NAK detection directly. The Mutex is used to serialize access to the hardware. The OP is thinking about the necessity of multi-thread/multi-processor support when he plans on only using single thread coding.  He is being short sighted. We should build the environment to use all of the capabilities of the hardware.
Chuck.

tferrin, 2018-04-14:
> I can't think of a reason why this toolchain would want to use a mutex considering most people who use this toolchain are not going to use FreeRTOS async tasks.

Definitely not true. Lots of us here are using FreeRTOS tasks.

stickbreaker, 2019-05-19:
Mutex necessary to serialize access,

closed 2019-05-19 by stickbreaker