espressif/arduino-esp32 #1293: SDFS File::flush() appears to do nothing

everslick, 2018-04-05:
I have a long running process that regularly appends data to a file on SD card and calls flush() on it afterwards. Still when I open the very same file and read its content it has no data appended nor its file size updated. When I close the file between writes, the data is there though. The exact same code works as expected when the file lifes on SPIFFS. So I assume that SD flush() is either not implemented or is buggy.

everslick, 2018-04-05:
Or do I just see this behaviour mentioned here: https://github.com/espressif/arduino-esp32/pull/514#issuecomment-315703988 ?

everslick, 2018-04-06:
I see there have been reports about this issue, putting them here for reference: #1173 and #1169

psykokwak-com, 2018-05-02:
I had the same issue.
I fixed it by adding  `fsync(fileno(_f));`  after `fflush(_f);` in the `void VFSFileImpl::flush()` in file vfs_api.cpp

ericlangel, 2019-03-01:
Same Problem with Flash (FFAT) File write.

This works...but its to slow
loop(){
File Data = FFat.open("/Data.txt", FILE_APPEND);
Data.write(Buffer,18);
Data.close();
}

And this doesn't work... nothing is written
loop(){
File Data = FFat.open("/Data.txt", FILE_APPEND);
while(1){
Data.write(Buffer,18);
Data.flush
}
}

I have to write 18 Bytes every 1ms.
How can i get it working?
FFat.open() needs 8ms.

psykokwak-com, 2019-03-01:
Just use the POSIX implementation : fopen, fwrite, fsync, fclose.

rrroonn, 2019-03-19:
`fsync(fileno(myFileDesc));
` does indeed work around the problem.

Bascy, 2019-04-14:
I'm getting a `'fileno' was not declared in this scope` when compiling
Where should this method be living?

everslick, 2019-04-14:
@Bascy : the workaround has already been merged 5 days ago. I think it will land in the stable release sooner or later. fileno() is an internal function, you cannot use it in a sketch.

closed 2019-04-14 by everslick

Bascy, 2019-04-14:
I'm getting the compile error when compiling FS library through my script, not using `fileno()`  myself

Bascy, 2019-04-14:
Solved, was caused by a stray compiler option

ArosPrince, 2020-08-19:
For me. the error is caused by -std=c++17. But I wouldn't call that *a stray compiler option*. There are libraries that require this option.

davefiddes, 2020-08-31:
@ArosPrince If you are looking to use a newer standard from the one defined in the platform.txt you need to use the GNU extension variant. So the ESP32 Arduino project currently uses -std=gnu++11 if you wanted to use C++17 you change this to -std=gnu++17. Using a newer C++ standard should probably be raised as a separate issue.

ArosPrince, 2020-10-05:
@davefiddes OK, thanks. Good to know.