espressif/arduino-esp32 #1295: irremote  ledcWrite xQueueGenericReceive crash, minimum application

Vorms, 2018-04-06:
Exception decoder
Decoding stack results
0x400870c0: invoke_abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 139
0x400871bf: abort at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/esp32/./panic.c line 148
0x40084429: xQueueGenericReceive at /Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c line 1441
0x400d0b7b: ledcWrite at D:\Users\Thierry\Documents\Arduino\hardware\espressif\esp32\cores\esp32\esp32-hal-ledc.c line 164
0x400d0955: IRsend::mark(unsigned int) at D:\Users\Thierry\Documents\Arduino\libraries\ESP32-IRremote\irSend.cpp line 25
0x400d0ad5: IRsend::sendSony(unsigned long, int) at D:\Users\Thierry\Documents\Arduino\libraries\ESP32-IRremote\ir_Sony.cpp line 28
0x400d08c4: loop() at D:\Users\Thierry\Documents\Arduino\idBox/idBox.ino line 11
0x400e2f5d: loopTask(void*) at D:\Users\Thierry\Documents\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 1

----------------------------- Remove above -----------------------------

### Hardware:
Board:	ESP32 Lolin					
Core Installation/update date:			6 avril 2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
The ESP32 crash 

code:
#include <IRremote.h>
int EMET_PIN = 2;

IRsend irsend(EMET_PIN);

void setup() {
  
}

void loop() {
   irsend.sendSony(0x8305E8, 24);
  delay(10000);
}

trace
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13076
entry 0x40078a58
/Users/ficeto/Desktop/ESP32/ESP32/esp-idf-public/components/freertos/./queue.c:1441 (xQueueGenericReceive)- assert failed!
abort() was called at PC 0x40084429 on core 1

Backtrace: 0x400870c0:0x3ffca290 0x400871bf:0x3ffca2b0 0x40084429:0x3ffca2d0 0x400d0b7b:0x3ffca310 0x400d0955:0x3ffca330 0x400d0ad5:0x3ffca350 0x400d08c4:0x3ffca370 0x400e2f5d:0x3ffca390

Rebooting..

Vorms, 2018-04-06:
Sorry it's my fault !
I forget a call to enableIROut()

stickbreaker, 2018-04-06:
Glad you found it.  Now you can close this issue.

closed 2018-04-06 by Vorms