espressif/arduino-esp32 #1301: user_interface.h and other missing?

twinotter, 2018-04-09:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			8/apr/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						921600

### Description:
My compiles aren't finding user_interface.h and other files.

I used the Windows install from the main page and have successfully compiled and run code on the ESP32 dev kit C. I'm trying to get the WifiManager package that is being updated by others running. I'm getting a "fatal error: user_interface.h: No such file or directory"

This file is provided in the 8266 library at esp8266\2.4.1\tools\sdk\include

And it seems that no one else compiling the package is have the user_interface.h missing problem. 

Is there a component of my installation that didn't work? Should this be in a complete esp32 install?

Thanks!
Rudy

lbernstone, 2018-04-09:
tzapu seems to have abandoned the project, so it does not get updates needed for ESP32 compatibility. Try https://github.com/alanswx/ESPAsyncWiFiManager

liebman, 2018-04-09:
@lbernstone the development branch is extremely active and includes ESP32 support.....

lbernstone, 2018-04-09:
There you go.  Sorry to have maligned tzapu.

twinotter, 2018-04-10:
I figured out my problem. User error and a lack of understanding of git. I needed to switch to the development branch. I thought I had by selecting the pulldown on the webpage and then copying the clone link into git gui. I still haven't figured it out with the gui, but I figured out the command line flags.

`git clone https://github.com/tzapu/WiFiManager.git --branch development --single-branch WiFiManager`

closing this issue...

closed 2018-04-10 by twinotter