espressif/arduino-esp32 #1304: BluetoothSerial shows up as audio device on macOS, immediately disconnects

gohai, 2018-04-10:
### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation/update date:			April 10 2018
IDE name:							Arduino 1.8.5
Flash Frequency:					80MHz
Upload Speed:						921600

### Description:
Running the SerialToSerialBT example shows up as an "audio" device on macOS 10.13.4. The device disconnects on its own soon after pairing. Thus, it cannot be used currently. (See screenshot for details.)

[image]

beegee-tokyo, 2018-04-11:
ESP32 bluetooth is showing as speakers on Android as well, but Bluetooth Serial is working fine as shown in my little [tutorial](https://desire.giesecke.tk/index.php/2018/04/10/esp32-wifi-setup-over-bluetooth-serial/). Cannot not test on MAC or iOS though.

gohai, 2018-04-12:
@copercini This define appears to set the default via `p_bta_dm_cfg`, but shouldn't it be possible to re-configure this e.g. through [BTM_SecAddDevice](https://github.com/espressif/esp-idf/blob/f620a3c4f35cb715e5b6e37e8556ec2278628660/components/bt/bluedroid/stack/btm/btm_dev.c#L41) or [BTM_SetDeviceClass](https://github.com/espressif/esp-idf/blob/f620a3c4f35cb715e5b6e37e8556ec2278628660/components/bt/bluedroid/stack/btm/btm_devctl.c#L552)?

gohai, 2018-04-12:
I looked at the output of macOS' _Console_ application while I attempted to connect to the device: the process `bluetoothaudiod` attempts a couple of times to creates an audio device, realizes that `No protocols known for device or services are nil`, and finally gives up, at which point `bluetoothd` removes the device. So this confirms the suspicion that the device class plays a role here.

lbernstone, 2018-06-17:
You can update the code yourself if you want to test.  https://github.com/espressif/arduino-esp32/issues/1142

gohai, 2018-06-17:
@lbernstone Thank you for the pointer. I just tried this, following your instructions - would you mind helping me a bit further?

First, I found the [idf-update](https://github.com/espressif/arduino-esp32/commits/idf-update) branch in this repository. Unfortunately it looks like its IDF is a few versions behind https://github.com/espressif/esp-idf/commit/789edc0b52b3a21a7ddd7a47f5d0d7b203eeac4d still.

Next I tried compiling `blink` with the following combinations:

- https://github.com/espressif/esp-idf/commit/789edc0b52b3a21a7ddd7a47f5d0d7b203eeac4d (first revision I can use) and https://github.com/espressif/arduino-esp32/commit/0cd15b445a66e4835e1bb96509798fcfc370dc75 (idf-update head): fails with `'sdmmc_host_io_int_enable' was not declared in this scope`
- https://github.com/espressif/esp-idf/commit/789edc0b52b3a21a7ddd7a47f5d0d7b203eeac4d (first revision I can use) and https://github.com/espressif/arduino-esp32/commit/89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a (master): succeeded
- https://github.com/espressif/esp-idf/commit/9a55b42f0841b3d38a61089b1dda4bf28135decd (master) and https://github.com/espressif/arduino-esp32/commit/0cd15b445a66e4835e1bb96509798fcfc370dc75 (idf-update head): succeeded
- https://github.com/espressif/esp-idf/commit/9a55b42f0841b3d38a61089b1dda4bf28135decd (master) and https://github.com/espressif/arduino-esp32/commit/89859f7f4c07ce3e5d487a4b689dbda78f5d5c3a (master): fails with `gatt_api.h: No such file or directory`

How do I go from here? Which of the above would you use? Should copying `libesp32.a` (and the specific header file that I need) be enough? Would you suggest I then use this on top of the arduino-esp32 `master` or `idf-update` branch?

Thanks in advance.

copercini, 2018-06-17:
Same issue solved: https://www.esp32.com/viewtopic.php?f=13&t=6099

Now the only problem is with the outdated IDF version of this core

gohai, 2018-06-19:
Here is a recompiled IDF with the default COD set to the changed values, if anyone else needs this as a workaround till this repo updates IDF again: https://github.com/gohai/arduino-esp32/commit/2c772325ffe94173f1674583c6e9cafb07826df0

copercini, 2018-07-08:
Fixed in https://github.com/espressif/arduino-esp32/commit/901a341949f23afe54bc6fc5c6617974ff9b36a9

Thanks @gohai !

closed 2018-07-08 by copercini