espressif/arduino-esp32 #1306: WiFi Mode() not working

sticilface, 2018-04-10:
### Hardware:
Board:							ESP32 Dev Module?
Core Installation/update date:			@ 29b3a81
IDE name:							Platform.io
Flash Frequency:					        40Mhz
Upload Speed:						115200

### Description:
WiFi.mode()  not working.  Set it to 0 (WIFI_OFF) and it stays at 3. 

### Sketch:
```cpp

#include <Arduino.h>
#include <WiFi.h>

void setup() {
  Serial.begin(115200);
  Serial.setDebugOutput(true); 
  Serial.println();

  Serial.println("----------   WiFi Info --------");
  WiFi.printDiag(Serial); 
  Serial.println();
  
  WiFi.mode(WIFI_MODE_NULL);

  Serial.println("----------   WiFi Info --------");
  WiFi.printDiag(Serial); 
  Serial.println();
}

void loop() {
}
```

### Debug Messages:
```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90

----------   WiFi Info --------
E (61) wifi: esp_wifi_get_mode 799 wifi is not init
E (61) wifi: esp_wifi_get_channel 1214 wifi is not init
E (62) wifi: esp_wifi_get_auto_connect 1391 wifi is not init
Mode: NULL
Channel: 253
Auto connect: 63
E (71) wifi: esp_wifi_get_config 1057 wifi is not init
SSID (68): Ù..@ƒ..@d..@¸..@Ä..@ÿ..@t..@ò..@D..@Ã..@ò..@∞..@§é
Ä..˝?¥<¸?\.@?L..@
Passphrase (36): D..@Ã..@ò..@∞..@§é
Ä..˝?¥<¸?\.@?L..@
BSSID set: 48

----------   WiFi Info --------
Mode: AP
Channel: 0
Auto connect: 1
SSID (0): 
Passphrase (0): 
BSSID set: 1
```

so it is clear that WiFi.mode(WIFI_MODE_NULL) is not setting the mode to 0, but 3 which is AP. 

Any ideas?

sticilface, 2018-04-15:
A little more digging on this issue.   All is not entirely happy with the WiFi modes. 

This sketch 
```cpp
void setup()
{
    Serial.begin(115200);
    // delete old config
    WiFi.disconnect(true);
    delay(1000);
    WiFi.begin(ssid, password);
    WiFi.mode(WIFI_MODE_NULL); 
}
```

That at no point calls any AP functions leads to this debug output
```
entry 0x40078a90
[D][WiFiGeneric.cpp:303] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:303] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:303] _eventCallback(): Event: 3 - STA_STOP
[D][WiFiGeneric.cpp:303] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:303] _eventCallback(): Event: 14 - AP_STOP
[D][WiFiGeneric.cpp:303] _eventCallback(): Event: 14 - AP_STOP
```

so it is curious that there are any AP_START event call backs given that there are no calls to start an AP. 

I've also discovered that the underlying `bool WiFiGenericClass::mode(wifi_mode_t m)` function 
https://github.com/espressif/arduino-esp32/blob/1da3138ed6bf4675836f6ebaa15bb5db280a6e36/libraries/WiFi/src/WiFiGeneric.cpp#L370 calls `espWiFiStop()` when `WIFI_MODE_NULL` is used.  This should then call `wifiLowLevelDeinit()` but this function does nothing and has the comment that it is not working.
https://github.com/espressif/arduino-esp32/blob/1da3138ed6bf4675836f6ebaa15bb5db280a6e36/libraries/WiFi/src/WiFiGeneric.cpp#L129

It would appear that `WIFI_MODE_NULL` is not working as intended.  as `    err = esp_wifi_set_mode(m)` is called but the mode remains at 2. 

further more..  if the only line in the sketch is `WiFi.getMode()` this calls `wifiLowLevelInit()` first thing which should initialise all the wifi stack as follows:
```cpp
static bool wifiLowLevelInit(){
    static bool lowLevelInitDone = false;
    if(!lowLevelInitDone){
        tcpipInit();
        wifi_init_config_t cfg = WIFI_INIT_CONFIG_DEFAULT();
        esp_err_t err = esp_wifi_init(&cfg);
        if(err){
            log_e("esp_wifi_init %d", err);
            return false;
        }
        esp_wifi_set_storage(WIFI_STORAGE_FLASH);
        esp_wifi_set_mode(WIFI_MODE_NULL);
        lowLevelInitDone = true;
    }
    return true;
}
```
  which should set the mode to `WIFI_MODE_NULL` but yet the result of a single `WiFi.getMode()` call is 2!  which is AP mode.  

I am beginning to think there is a bug in the API!

any thoughts?

sticilface, 2018-04-16:
workaround:  

`WIFI_MODE_NULL` does not set the mode to null.  I've edited the `mode` function so that it returns `WIFI_MODE_NULL` if `_esp_wifi_started` is `false`.  I've then removed the `wifiLowLevelInit()` from `getMode()` and changed the order of `mode()`.  

The end result is that is works as expected despite the SDK reporting mode 2 (AP) even when only the low level init had been done and it was not actually in AP mode. At least this way the WiFi class reports the right mode.  

example sketch below. Along with 2 debug outputs.   Sorry for using my Tasker lib but it simplifies scheduling of things... 

```cpp
#include <WiFi.h>
#include <Tasker.h> //https://github.com/sticilface/Tasker

const char* ssid     = "abc";
const char* password = "def";

Task tasker;
const char * mysystem_event_names[] = { "WIFI_READY", "SCAN_DONE", "STA_START", "STA_STOP", "STA_CONNECTED", "STA_DISCONNECTED", "STA_AUTHMODE_CHANGE", "STA_GOT_IP", "STA_LOST_IP", "STA_WPS_ER_SUCCESS", "STA_WPS_ER_FAILED", "STA_WPS_ER_TIMEOUT", "STA_WPS_ER_PIN", "AP_START", "AP_STOP", "AP_STACONNECTED", "AP_STADISCONNECTED", "AP_PROBEREQRECVED", "GOT_IP6", "ETH_START", "ETH_STOP", "ETH_CONNECTED", "ETH_DISCONNECTED", "ETH_GOT_IP", "MAX"};

void WiFiEvent(WiFiEvent_t event)
{
  Serial.printf("[WiFi-event] event: %d :%s\n", event,mysystem_event_names[event] );
  Serial.printf("WiFi Mode = %d\n", WiFi.getMode());
}

void setup()
{
  Serial.begin(115200);

  delay(1000);

  WiFi.onEvent(WiFiEvent);

  Serial.println("\n\t\t\tStarting STA , mode should be 1");
  WiFi.enableSTA(true);
  WiFi.begin(ssid, password);

  tasker.add([](Task & t) {
    Serial.println("\n\t\t\tStopping WiFi mode should be 0");
    WiFi.mode(WIFI_MODE_NULL);
  }).setTimeout(10000);

  tasker.add([](Task & t) {
    Serial.println("\n\t\t\tStarting STA_AP, mode should be 3");
    WiFi.mode(WIFI_MODE_APSTA);
    WiFi.reconnect();
    WiFi.softAP("test_AP");
  }).setTimeout(20000);

  tasker.add([](Task & t) {
    Serial.println("\n\t\t\tContiniue AP only, mode should be 2");
    WiFi.enableSTA(false);
  }).setTimeout(25000);

  tasker.add([](Task & t) {
    Serial.println("\n\t\t\tStop all, mode should be 0");
    WiFi.enableAP(false);
  }).setTimeout(30000);

}

void loop()
{
  tasker.run();
}
```

NORMAL result from git head
```
...Starting STA , mode should be 1
[WiFi-event] event: 13 :AP_START
WiFi Mode = 3
[WiFi-event] event: 13 :AP_START
WiFi Mode = 3
[WiFi-event] event: 4 :STA_CONNECTED
WiFi Mode = 3
[WiFi-event] event: 7 :STA_GOT_IP
WiFi Mode = 3

...Stopping WiFi mode should be 0
[WiFi-event] event: 3 :STA_STOP
WiFi Mode = 3
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 3
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 3

...Starting STA_AP, mode should be 3
E (22160) wifi: esp_wifi_disconnect 877 wifi not start
[WiFi-event] event: 13 :AP_START
WiFi Mode = 3
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 3
[WiFi-event] event: 13 :AP_START
WiFi Mode = 3
[WiFi-event] event: 13 :AP_START
WiFi Mode = 3

...Continiue AP only, mode should be 2
[WiFi-event] event: 3 :STA_STOP
WiFi Mode = 2

...Stop all, mode should be 0
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 2
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 2
```

with above mentioned changes

```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13256
entry 0x40078a90

...Starting STA , mode should be 1
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 1
[WiFi-event] event: 2 :STA_START
WiFi Mode = 1
[WiFi-event] event: 2 :STA_START
WiFi Mode = 1
[WiFi-event] event: 4 :STA_CONNECTED
WiFi Mode = 1
[WiFi-event] event: 7 :STA_GOT_IP
WiFi Mode = 1

...Stopping WiFi mode should be 0
[WiFi-event] event: 3 :STA_STOP
WiFi Mode = 0
[WiFi-event] event: 3 :STA_STOP
WiFi Mode = 0

...Starting STA_AP, mode should be 3
[WiFi-event] event: 13 :AP_START
WiFi Mode = 3
[WiFi-event] event: 13 :AP_START
WiFi Mode = 3
[WiFi-event] event: 4 :STA_CONNECTED
WiFi Mode = 3
[WiFi-event] event: 7 :STA_GOT_IP
WiFi Mode = 3

...Continiue AP only, mode should be 2
[WiFi-event] event: 3 :STA_STOP
WiFi Mode = 2
[WiFi-event] event: 3 :STA_STOP
WiFi Mode = 2

...Stop all, mode should be 0
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 2
[WiFi-event] event: 14 :AP_STOP
WiFi Mode = 0
[WiFi-event] event: 8 :STA_LOST_IP
WiFi Mode = 0
```

so in conclusion, it looks like my changes work appropriately.  I'd appreciate some other testing of this please.  so far no one has identified it as an issue...

tablatronix, 2018-04-23:
I do not see this behavior at all, with either the 2 sketches provided.

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:812
load:0x40078000,len:0
load:0x40078000,len:11584
entry 0x40078a60
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 2 - STA_START
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 5 - STA_DISCONNECTED
[W][WiFiGeneric.cpp:298] _eventCallback(): Reason: 201 - NO_AP_FOUND
```

```
----------   WiFi Info --------
Mode: NULL
Channel: 0
Auto connect: 1
SSID (6): <
BSSID set: 0

----------   WiFi Info --------
Mode: STA
Channel: 1
Auto connect: 1
SSID (6): 
BSSID set: 0

----------   WiFi Info --------
Mode: NULL
Channel: 1
Auto connect: 1
SSID (6): 
BSSID set: 0
```

tablatronix, 2018-04-23:
oh btw that garbage when calling printdiag before init is a pita #1088

tablatronix, 2018-04-23:
> which should set the mode to WIFI_MODE_NULL but yet the result of a single WiFi.getMode() call is 2! which is AP mode.

Actually it appears to be the persistent mode saved in flash, if you set mode and reset, and wifiinit, and getmode it will be what you last had it as.

tablatronix, 2018-04-23:
I did one time catch this
notice the 4 ap_start events, and the wifi init failed, not really sure what happeend there, and I cannot reproduce it, maybe some race condition
```
[D][WiFiGeneric.cpp:297] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:297] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:297] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:297] _eventCallback(): Event: 13 - AP_START

E (1022) wifi: esp_wifi_get_mode 799 wifi is not init
0
mode:
0
E (1023) wifi: esp_wifi_start 810 wifi is not init
E (1023) wifi: esp_wifi_connect 868 wifi is not init
----------   WiFi Info --------
E (1029) wifi: esp_wifi_get_mode 799 wifi is not init
E (1034) wifi: esp_wifi_get_channel 1214 wifi is not init
E (1039) wifi: esp_wifi_get_auto_connect 1391 wifi is not init
Mode: NULL
Channel: 0
Auto connect: 0
E (1048) wifi: esp_wifi_get_config 1057 wifi is not init
SSID (5):  ��?
Passphrase (4): ��?�
BSSID set: 31
```

sticilface, 2018-04-23:
There should be no resetting at all to get it to work.  
the issue i've identified is that getMode() does not report the right value back to the WiFi lib when wifi is not initialised.  Fixing that then allows the WiFi lib to make the required changes as it has 
https://github.com/espressif/arduino-esp32/blob/1424b6d1a4435830b908d2d45c74b1af06831ca7/libraries/WiFi/src/WiFiGeneric.cpp#L376
as checks to see if it needs to do anything. 

making getMode() a passive function that does not by itself initialise the WiFi stack seems more sensible to me, than initialising it and having it in the wrong mode. 

I have not yet experimented with saving the wifi settings to flash.  so i cannot comment on those.

tablatronix, 2018-04-23:
Yeah I was just noting the mode is not random its the esp stored mode that was last set.

tablatronix, 2018-04-24:
So it appears that  WIFI_MODE_NULL is not an sdk mode , its a pseudo mode, this is what was confusing me.

I also found another issue with this, see here
https://github.com/espressif/arduino-esp32/commit/21ff3d0e32eed03e7b332b51bbcf7820fccf9f72

closed 2019-08-15 by stale[bot]

daald, 2019-12-28:
In my project, I successfully fixed the reconnect issue with the following hack:

```
diff --git a/libraries/WiFi/src/WiFiGeneric.cpp b/libraries/WiFi/src/WiFiGeneric.cpp
index e562921..aab5805 100644
--- a/libraries/WiFi/src/WiFiGeneric.cpp
+++ b/libraries/WiFi/src/WiFiGeneric.cpp
@@ -483,8 +483,10 @@ void WiFiGenericClass::enableLongRange(bool enable)
 bool WiFiGenericClass::mode(wifi_mode_t m)
 {
     wifi_mode_t cm = getMode();
+    log_d("mode() cm=%d, m=%d", cm, m);
     if(cm == m) {
-        return true;
+        log_d("HACK: skip return true");
+        //return true;
     }
     if(!cm && m){
         if(!wifiLowLevelInit(_persistent)){
```

But I think the real issue is how getMode is implemented

philbowles, 2020-02-03:
Not sure what version was in use when this problem was noted, but it's happening again in 1.0.4:

```cpp
#include<WiFi.h>
void setup() {
  Serial.begin(115200);
  WiFi.begin();
  WiFi.disconnect(true,true);
  Serial.printf("STA mode? %s AP mode? %s [%d]\n",WiFi.getMode() & WIFI_STA ? "true":"false",WiFi.getMode() & WIFI_AP ? "true":"false",WiFi.getMode());
  WiFi.printDiag(Serial);
  WiFi.mode(WIFI_OFF);
  Serial.printf("STA mode? %s AP mode? %s [%d]\n",WiFi.getMode() & WIFI_STA ? "true":"false",WiFi.getMode() & WIFI_AP ? "true":"false",WiFi.getMode());
  WiFi.printDiag(Serial);
}

void loop() {
}
```

```cpp
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 0 - WIFI_READY
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 2 - STA_START
[E][WiFiSTA.cpp:219] begin(): connect failed!
STA mode? false AP mode? false [0]
[D][WiFiGeneric.cpp:337] _eventCallback(): Event: 3 - STA_STOP
Mode: STA
Channel: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
STA mode? false AP mode? false [0]
Mode: STA
Channel: 1
SSID (0): 
Passphrase (0): 
BSSID set: 0
```
getMode() returns 0, yet printDiag reports STA - this is just plain WRONG! Do getMode and printDiag retrieve the REAL value of mode from different places? What is going on here and why has it not been fixed?

atanisoft, 2020-02-03:
WiFi.getMode() and WiFi.printDiag() both are working as designed and expected on 1.0.4. Your code is incorrectly testing for bit mask rather than equality as it should be.