espressif/arduino-esp32 #1308: How to make ESP32 connect to previous WiFi

tobeedelafuente, 2018-04-11:
### Hardware:
Board:							ADAFRUIT HUZZAH32
Core Installation/update date:			07/apr/2018
IDE name:							Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:						115200

### Description:
Just tested out ESP32. So far most of the code I use for ESP8266 works fine with ESP32. One issue I encountered however is the usage of this code snippet when connecting to the last router I connected to. 

### Sketch:
```cpp
...
String ssid = "";
String pass = "";
WiFi.begin(ssid.c_str(), pass.c_str());
...
```

Initially the ssid and pass are set to match the WiFi creds of my router. Afterwards I upload the sketch above. On the ESP8266, I am able to connect to the router. But on the ESP32, I am not able to connect. Does the ESP32 have a mechanism that would enable me to connect to the last router I connected to?

lbernstone, 2018-04-11:
That should not work in esp8266 either.  The correct method is to call WiFi.begin().  No parameters means it will use the last connection info.  Blank strings are not the same.  I usually do:
```
#define MYSSID "wifissid"
#define PASSWD "passwd"
#ifdef MYSSID
  WiFi.begin(MYSSID,PASSWD);
#else    
  WiFi.begin();
#endif
```

tobeedelafuente, 2018-04-26:
Thank you for pointing out the WiFi.begin() call when SSID is blank. I am able to connect on the previous WiFi I connected to. Cheers guys!

closed 2018-04-26 by tobeedelafuente