espressif/arduino-esp32 #1309: no begin

sticilface, 2018-04-11:
https://github.com/espressif/arduino-esp32/blob/69f72eca849a44688e2da38028eb9aa707e91f08/libraries/FS/src/FS.h#L88

In ESP8266 FS implementation there is a begin function, but not here..

https://github.com/esp8266/Arduino/blob/4305275f8d28c8c449d39d41290675ccceb01a94/cores/esp8266/FS.h#L107

any reason why?

everslick, 2018-04-12:
https://github.com/espressif/arduino-esp32/blob/69f72eca849a44688e2da38028eb9aa707e91f08/libraries/SPIFFS/src/SPIFFS.h#L26

On ESP32 the FS is an more abstract interface, each FS implementation has its own begin().

sticilface, 2018-04-15:
I had worked that out, but it makes code incompatible with the ESP8266.

everslick, 2018-04-16:
FS code is incompatible in many places. I had to wrap the whole FS API into my own abstraction layer to make FS access transparent on both platforms.

closed 2019-08-15 by stale[bot]