espressif/arduino-esp32 #1312: SD_MMC.cpp build fails with latest ESP IDF

hamishcunningham, 2018-04-12:
hi, I just pulled the latest from this repo and the latest from ESP IDF and now my build fails as follows:

```
CXX build/arduino/libraries/SD_MMC/src//SD_MMC.o
In file included from /home/hamish/esp/esp-idf/components/fatfs/src/esp_vfs_fat.h:20:0,
                 from /home/hamish/more-fish/idf/snurgle/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:21:
/home/hamish/more-fish/idf/snurgle/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp: In member function 'bool fs::SDMMCFS::begin(const char*, bool)':
/home/hamish/esp/esp-idf/components/driver/include/driver/sdmmc_host.h:49:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
/home/hamish/more-fish/idf/snurgle/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:45:25: note: in expansion of macro 'SDMMC_HOST_DEFAULT'
     sdmmc_host_t host = SDMMC_HOST_DEFAULT();
                         ^
/home/hamish/esp/esp-idf/components/driver/include/driver/sdmmc_host.h:49:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
/home/hamish/more-fish/idf/snurgle/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:45:25: note: in expansion of macro 'SDMMC_HOST_DEFAULT'
     sdmmc_host_t host = SDMMC_HOST_DEFAULT();
                         ^
/home/hamish/esp/esp-idf/components/driver/include/driver/sdmmc_host.h:49:1: sorry, unimplemented: non-trivial designated initializers not supported
 }
 ^
/home/hamish/more-fish/idf/snurgle/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:45:25: note: in expansion of macro 'SDMMC_HOST_DEFAULT'
     sdmmc_host_t host = SDMMC_HOST_DEFAULT();
                         ^
/home/hamish/more-fish/idf/snurgle/components/arduino/libraries/SD_MMC/src//SD_MMC.cpp:57:5: warning: missing initializer for member 'esp_vfs_fat_mount_config_t::allocation_unit_size' [-Wmissing-field-initializers]
     };
     ^
/home/hamish/esp/esp-idf/make/component_wrapper.mk:273: recipe for target 'libraries/SD_MMC/src//SD_MMC.o' failed
make[1]: *** [libraries/SD_MMC/src//SD_MMC.o] Error 1
/home/hamish/esp/esp-idf/make/project.mk:457: recipe for target 'component-arduino-build' failed
make: *** [component-arduino-build] Error 2
```

Any ideas? More details below
Thanks!
Hamish

### Hardware:
Board:							Adafruit ESP32 Feather
Core Installation/update date:			12th April
IDE name:						IDF component
Flash Frequency:					40Mhz
Upload Speed:					921600

### Description:

IDF-based compile breaks with latest commits.

### Sketch:
```
// main.cpp

// IDF includes...
#include <stdio.h>
#include "freertos/FreeRTOS.h"
#include "freertos/task.h"
#include "esp_system.h"
#include "esp_spi_flash.h"
#include "Arduino.h"

static const char *TAG = "MAIN";

/////////////////////////////////////////////////////////////////////////////
// example code from the adafruit library touchpaint_featherwing example
// (with minimal changes)

/***************************************************
  This is our library for the Adafruit HX8357D Featherwing
  ----> http://www.adafruit.com/products/2050
[... 170 lines cut ...]
  hx8357_setup();
  while(1) {
    micros(); //update overflow
    hx8357_loop();
    vTaskDelay(
      10 / portTICK_PERIOD_MS // https://github.com/espressif/esp-idf/issues/1646
    ); // & https://github.com/espressif/arduino-esp32/issues/595 but doesn't fix
  }
}

// main
extern "C" void app_main()
{
  const BaseType_t TASK_CORE = 1;
  xTaskCreatePinnedToCore(
    touchpaint_featherwing, "touchpaint_task",
    8192, NULL, 1000, NULL, TASK_CORE
  );
}
```

### Debug Messages:
```
see make output above 
```

Curclamas, 2018-04-13:
I can confirm this, I think @me-no-dev already implemented a fix in his branch: 6e99108fb45e61201084526f4de84b4e721a0590 , if you want to try: I think this should fix it.

hamishcunningham, 2018-04-16:
Thanks @Curclamas that worked for me :)

Curclamas, 2018-04-18:
@hamishcunningham I think this issue should stay open as long as the fixes are not yet upstream in master.

closed 2018-04-24 by igrr