espressif/arduino-esp32 #1313: Timer timerAlarmEnable() bug

PRosenb, 2018-04-14:
When you enable a timer alarm, disable it and then enable it again, the function timerAlarmEnable() only works, if you place a yield() call in front of it (or call it twice).
This behaviour can be reproduced on a ESP32 with using the following sketch. Comment the line with yield() to see the wrong behaviour.

### Sketch:
```cpp
#include <esp32-hal-timer.h>

hw_timer_t *timer;

void IRAM_ATTR isrTimerExpired() {
  Serial.println("timer expired");
}

void startTimer() {
  Serial.println("startTimer()");
  if (timer == NULL) {
    timer = timerBegin(0, 80, true);
    timerAttachInterrupt(timer, &isrTimerExpired, true);
  }
  const unsigned long durationMs = 1000;
  timerAlarmWrite(timer, durationMs * 1000, false);
  
  // if the following yield() is removed, the timer will not
  // be enabled the second time
  yield();
  timerAlarmEnable(timer);
  Serial.print("isEnabled: "); Serial.println(timerAlarmEnabled(timer));
}

void stopTimer() {
  Serial.println("stopTimer()");
  if (timer != NULL) {
    timerAlarmDisable(timer);
    timerDetachInterrupt(timer);
    timerEnd(timer);
    timer = NULL;
  }
}

void setup() {
  Serial.begin(115200);
  startTimer();
  stopTimer();
  startTimer();
}

void loop() {
  // not used
  delay(1000);
}
```

stickbreaker, 2018-04-14:
@PRosenb is the `yield()` function causing the code to be shuffled between the cores?  I know that there are multiple warnings about attaching an interrupt in one core and not being able to detach it in the other core?

Do you need to initially NULL the `timer` variable?
```c++
hw_timer_t *timer=NULL;
```

Chuck.

PRosenb, 2018-04-14:
Hi @stickbreaker , thanks for your quick reply. Setting it to NULL makes no difference, it is initialized to NULL already. I updated the question to make it visible.
Good point about the possibility that it is executed on the other core. As I understood is all Arduino code supposed to be executed on the same core, right?
https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/main.cpp

cyberman54, 2018-09-21:
I can confirm this effect.
Maybe - not sure - this happenes only on ESP32 boards with CPU revision 0. Same code running on a board with CPU revision 1 did not let me reproduce this issue.
I did not use yield(), but a ESP_LOGI() in front of the timerAlarmEnable. This suppresses the effect, too.
Turns out, the timer is sometimes enabled, sometimes not.

This is causing side effects, hard to find...

No idea what's going wrong here.

Delsoo, 2018-11-21:
This problem happened to me. I developed a program with a Timer module in CPU revision 1 and it works well. I sent my code to my friend that has CPU revision 0, the timer module runs once and stops. When we add yield() just mentioned before, the timer worked fine!

Thanks for helping!

fabiuz7, 2019-04-12:
Yeah it seems related... Also I've just checked my board 

> Chip is ESP32D0WDQ6 (revision 1)

Actually I don't know if I have a rev 0 to try to fix the bug...

PRosenb, 2019-08-01:
I got a module with CPU revision 1 in the meantime and can confirm that it works as expected on revision 1.

closed 2019-08-01 by PRosenb

TLS1000, 2019-09-02:
hi, with the esp32 rev0, using the dimmable light library; the proposed solution of adding a 'yield()' call causes a crash:

> Guru Meditation Error: Core  1 panic'ed (InstrFetchProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x00000000  PS      : 0x00050c33  A0      : 0x00000000  A1      : 0x3ffbe720  
A2      : 0x3ffbe760  A3      : 0x000017bf  A4      : 0x3ffc00d8  A5      : 0xc00a1000  
A6      : 0x3ffb1e10  A7      : 0x00000001  A8      : 0x3ffb8498  A9      : 0x80081178  
A10     : 0x3ffbe780  A11     : 0x000026dd  A12     : 0x3ffc0110  A13     : 0x000000c8  
A14     : 0x000000c8  A15     : 0x0000251b  SAR     : 0x0000001c  EXCCAUSE: 0x00000014  
EXCVADDR: 0x00000000  LBEG    : 0x0000001a  LEND    : 0x3ffc02e8  LCOUNT  : 0x3ffc0108  

Backtrace: 0x00000000:0x3ffbe720 0x7ffffffd:0x3ffbe740 0x40080edd:0x3ffbe760 0x40081123:0x3ffbe780 0x40081175:0x3ffbe7a0 0x400819dd:0x3ffbe7c0 0x4000bfed:0x3ffb1ed0 0x40088e35:0x3ffb1ee0 0x40083eea:0x3ffb1f00 0x400d180e:0x3ffb1f20 0x400d1825:0x3ffb1f50 0x400d10d6:0x3ffb1f70 0x400d0d1f:0x3ffb1f90 0x400d2439:0x3ffb1fb0 0x4008a3e1:0x3ffb1fd0

> 

Calling the timer enable twice doesn't cause a crash and the timer does seem to enable, but dimmable light library doesn't execute correctly.

TLS1000, 2019-09-06:
I worked out a solution where the timer of a REV0 board can be used in a successive manner:
https://github.com/fabiuz7/Dimmable-Light-Arduino/issues/8

fabiuz7, 2019-09-06:
Thank for the feedback, I can image blood and sweat for timer (it isn’t a trivial peripheral). I will look at your solution as soon as possible. I don’t have a rev0, so once the library gets updated, I would be glad to hear a feedback from you.

fabiuz7/Dimmable-Light-Arduino#8<https://github.com/fabiuz7/Dimmable-Light-Arduino/issues/8>

buelowp, 2024-04-10:
Holy cats, I found this, and fixed my problem. I have a REV 0 CPU, and I had the worst time. Thanks for this. It's still helping people.