espressif/arduino-esp32 #1314: Cannot use UART0/UART1/UART2 concurrently on ESP32 using Arduino IDE

zamorae, 2018-04-14:
### Hardware:
Board:							ESP-WROOM-32
Core Installation/update date:	       11/jul/2017
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:				       115200

### Description:
Hi everyone,

I am working on a project in which I need to have an ESP32 connected via serial to another ESP32 and at the same time to a Nextion HMI. I also need to communicate from the central ESP32 to these 2 devices concurrently ( having the serial communications in the "central" ESP32 open to both devices and issuing commands simultaneously)
I have been making some testing in preparation to the interfacing to these devices by replacing the Nextion with an ESP826MOD:

Central ESP32: ESP-WROOM-32 devkit, connected to a USB port of a PC
Client Device A: ESP-WROOM-32 devkit connected to USB port of another PC
Client Device B: ESP8266MOD devkit connected to a USB port of the same PC as Central ESP32

I load a simple sketch (Based on serial passthrough example by Erik Nyquist) on the central ESP32 to open UART1 and UART2. The idea is to open the Serial Monitor connected via the USB port to each device and issue simple strings, which I expect reflected on any of the other Serial Monitors from A and B.
 For devices A and B,  I setup UART1 for the ESP8266MOD (Device B - TX/RX pins) and UART2 on the 2nd ESP32 (Device A - pins 16 & 17).  I then proceed with the following sketches (Based on serial passthrough example by Erik Nyquist):

### Sketch:
```cpp

//Central ESP32

#include <HardwareSerial.h>

HardwareSerial Serial1(1);
HardwareSerial Serial2(2);

void setup() {
  // initialize serial:
  Serial.begin(115200);
  Serial1.begin(115200,SERIAL_8N1, 4, 2);    //Baud rate, parity mode, RX, TX
  Serial2.begin(115200,SERIAL_8N1, 16, 17);
}

void loop() {
  
   if(Serial.available())                   //if user typed something through the USB...
    {    
      Serial1.write(Serial.read());         //Push it through port 1 to Device B
      Serial2.write(Serial.read());         //Push it through port 2 to Device A
     
      Serial.flush();                       //erase any data on serial port after operation 
    }
 
   if (Serial1.available())                 //If data is received through Serial 1 from Device B...
    {    
       Serial.write(Serial1.read());        //Show data on Serial Monitor
       Serial.flush();                      //Flush port
    }

    if (Serial2.available())                //If data is received through Serial 2
    {    
       Serial.write(Serial2.read());        //Show data on Serial Monitor 1
       Serial.flush();                      //Flush port
    }
}
```
```cpp

//Client A (ESP32)
HardwareSerial Serial2(2);             //Needed since this is an ESP32

void setup() {
  Serial.begin(115200);
  Serial2.begin(115200, SERIAL_8N1, 16, 17);
}

void loop() {  
  if (Serial.available())           // If user typed something through the USB...
  {      
    Serial2.write(Serial.read());   // Push it through Port 2 to Central ESP32
  }
  if (Serial2.available())          // If data is received through Port 2 from Central ESP32
  {     
    Serial.write(Serial2.read()); // Show data on Serial Monitor 2 
  }
}
```
```cpp

//Client B (ESP8266MOD)

void setup() {
  Serial.begin(115200);
  Serial1.begin(115200, SERIAL_8N1);                 //Does not need HardwareSerial initiaization or pin def since
}                                                    //this Serial is from the awesome ESP8266

void loop() {
  
    if (Serial.available())              //if user typed something through the USB...
    {     
      Serial1.write(Serial.read());      //Push it through port 1 to Central ESP32
    }

    if (Serial1.available())             //If data is received through Serial 1 from Central ESP32...
    {    
      Serial.write(Serial1.read());    //Show data on Serial Monitor 3
    }
}

```

In summary, I can type any text from Serial monitor of Devices A and B and I will see it on Serial Monitor of Central ESP32, but if I type something on Serial Mon of Central ESP32 I never see anything on Device B and only garbled text on Device A. 

I have tried several configurations to no avail. My question is: does anybody knows if an updated library or firmware is available for the ESP32, or some sort of workaround to overcome this issue?
Can somebody point me out in the right direction to be able to use the 3 UARTs concurrently? it is very urgent/critical for me so any idea is welcome.

Thanks in advance
-EZ

negativekelvin, 2018-04-14:
https://github.com/espressif/arduino-esp32/issues/1189

lbernstone, 2018-04-14:
That issue explains it fairly well.  tldr is, use Serial0 and 1 for your duplex devices and serial2 for your debug messages.

AbdelrhmanElsawy, 2019-05-30:
I need link for that library you use .
can i use it to receive sensors readings just on central node ? 
( three sensors send data by uart )

lbernstone, 2019-05-30:
[HardwareSerial](https://github.com/espressif/arduino-esp32/blob/master/cores/esp32/HardwareSerial.h) is part of arduino-esp32 core.

AbdelrhmanElsawy, 2019-05-30:
when i include it to my program it is not be declared !!

lbernstone, 2019-05-30:
Then you have a problem with your installation.  It is in core, you don't even need to include it.

iPAS, 2019-06-27:
Should it be ?

>   if(Serial.available())                   //if user typed something through the USB...
    {    
     char c = Serial.read();

      Serial1.write(c);         //Push it through port 1 to Device B
      Serial2.write(c);         //Push it through port 2 to Device A
     
      Serial.flush();                       //erase any data on serial port after operation 
    }

odair-rocha, 2019-08-20:
https://quadmeup.com/arduino-esp32-and-3-hardware-serial-ports/

include <HardwareSerial.h>

HardwareSerial MySerial(1);

void setup() {
    MySerial.begin(9600, SERIAL_8N1, 16, 17);
}

void loop() {
    while (MySerial.available() > 0) {
        uint8_t byteFromSerial = MySerial.read();
        // Do something
    }
    
    //Write something like that
    MySerial.write(rand(0, 255));
}

closed 2019-11-02 by stale[bot]

KrisUAV, 2020-07-27:
I cannot even view anything on my serial monitor.
I was wondering if anyone could help me out.... I am so lost....

Here is the code below.

#include "HardwareSerial.h"
#include "PMS.h"
HardwareSerial pmsSerial (2);
const int SerialDataBits = 9600;

PMS pms(Serial);
PMS::DATA data;
char col;
unsigned int CR1 = 0,CR2 = 0;
unsigned char buffer_RTT[40]={};   //Serial buffer; Received Data

void setup()
#define pmsSerial Serial
{
  Serial.begin(9600);
  pmsSerial.begin (9600, SERIAL_8N1, 16,17);
  Serial2.begin(9600,SERIAL_8N1,16,17);
 
}

void loop(){
    for(int i=0;i<40;i++)
    {
      col =Serial2.read();
      buffer_RTT[i]=(char)col;
      delay(2);
    }
    
    Serial2.flush();
    CR1 =(buffer_RTT[38]<<8) + buffer_RTT[39];
    CR2 = 0;
    for(int i=0;i<38;i++)
      CR2 += buffer_RTT[i];
    if(CR1 == CR2)               
    
{
  while (pmsSerial.available()>0){
    uint8_t byteFromSerial = pmsSerial.read();
  }
  if (pms.read(data))
{ 
    Serial.print("PM 1.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_1_0);

    Serial.print("PM 2.5 (ug/m3): ");
    Serial.println(data.PM_AE_UG_2_5);

    Serial.print("PM 10.0 (ug/m3): ");
    Serial.println(data.PM_AE_UG_10_0);

    Serial.println();
    if (Serial2.available())             
    {    
       Serial.write(Serial2.read());
  
  }
}
}
}

Eiyvor, 2020-07-28:
@KrisUAV, I'm also using this PMS library for pms9003m and having the same issue as yours, it works perfectly alone but when i combine the code with the code that connects to mosquitto broker via sim800, I not getting any reading from that sensor, but the sim800 is working without any issue. I'm using the Ttgo tcall sim800l for my proto.

Eiyvor, 2020-07-28:
In a video I saw that we can configure any pin as uart RX and TX, all you need to do is go to the hardware serial.cpp and change pins. But unfortunately that also didn't work. Are all these uart pins multiplexed to a single uart buffer?!

Hemant7777, 2021-04-14:
is there any way to use esp 32 GPIO pins as RX and TX in esp32?
I have 3 sensors that have 3 RX and 3 TX pins so can I connect them with esp32.