espressif/arduino-esp32 #1315: ESP32 does not accept OTA

williamesp2015, 2018-04-15:
PlatformIO 3.5.3a7
Platformio.ini
[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_port = 192.168.1.6

13:50:30 [INFO]: Upload size: 599168
Sending invitation to 192.168.1.6
13:50:30 [INFO]: Waiting for device...
13:50:40 [ERROR]: No response from device
*** [upload] Error 1

I have tested with:
[env:esp32dev]
platform = https://github.com/platformio/platform-espressif32.git#feature/stage
board = esp32dev
framework = arduino
upload_port = 192.168.1.6
Result: Same

everslick, 2018-04-15:
**upload_port = 192.168.1.6**
 
???

beegee-tokyo, 2018-04-15:
Can be many problems.
IP address correct? Can you ping the IP address?
Is the ESP32 really connected? Can you ping the IP address?
Is the ESP32 on the same local network?  Can you ping the IP address?

Are you calling ArduinoOTA.handle() frequently in your loop()? Do you have many other tasks running in your app that keeps the ESP32 too busy to response to the OTA request?

williamesp2015, 2018-04-15:
I am using the BasicOTA example and after finding local IP I try to reprogram it but I was not successfull.
Sending invitation to 192.168.1.5 . ( I got it after first time finding my Esp module IP)
I also checked pinging my router and it was ok.

16:20:39 [INFO]: Waiting for device...
16:20:49 [ERROR]: No response from device

#include <WiFi.h>
#include <ESPmDNS.h>
#include <WiFiUdp.h>
#include <ArduinoOTA.h>
const char* ssid = "MySSID";
const char* password = "Pass";

void setup() {
  Serial.begin(115200);
  delay(1000);
  Serial.println("Booting");
  WiFi.mode(WIFI_STA);
  WiFi.begin(ssid, password);
  while (WiFi.waitForConnectResult() != WL_CONNECTED) {
    Serial.println("Connection Failed! Rebooting...");
    delay(1000);
    ESP.restart();
  }
  Serial.print("IP address: ");
     Serial.println(WiFi.localIP());
  // Port defaults to 3232
  // ArduinoOTA.setPort(3232);

  // Hostname defaults to esp3232-[MAC]
  // ArduinoOTA.setHostname("myesp32");

  // No authentication by default
  // ArduinoOTA.setPassword("admin");

  // Password can be set with it's md5 value as well
  // MD5(admin) = 21232f297a57a5a743894a0e4a801fc3
  // ArduinoOTA.setPasswordHash("21232f297a57a5a743894a0e4a801fc3");

  ArduinoOTA
    .onStart([]() {
      String type;
      if (ArduinoOTA.getCommand() == U_FLASH)
        type = "sketch";
      else // U_SPIFFS
        type = "filesystem";

      // NOTE: if updating SPIFFS this would be the place to unmount SPIFFS using SPIFFS.end()
      Serial.println("Start updating " + type);
    })
    .onEnd([]() {
      Serial.println("\nEnd");
    })
    .onProgress([](unsigned int progress, unsigned int total) {
      Serial.printf("Progress: %u%%\r", (progress / (total / 100)));
    })
    .onError([](ota_error_t error) {
      Serial.printf("Error[%u]: ", error);
      if (error == OTA_AUTH_ERROR) Serial.println("Auth Failed");
      else if (error == OTA_BEGIN_ERROR) Serial.println("Begin Failed");
      else if (error == OTA_CONNECT_ERROR) Serial.println("Connect Failed");
      else if (error == OTA_RECEIVE_ERROR) Serial.println("Receive Failed");
      else if (error == OTA_END_ERROR) Serial.println("End Failed");
    });
  ArduinoOTA.begin();
  Serial.println("Ready");
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());
}
void loop() {
  ArduinoOTA.handle();
}

everslick, 2018-04-15:
try using the espota.py (or espota.exe) tool directly, with something like:

espota.py  -r -i 192.168.1.5 -f YOUR-FIRMWARE.bin -d -p 3232 -a YOUR-PASSWD

it will show you more debug output

williamesp2015, 2018-04-15:
18:10:05 [INFO]: Starting on 0.0.0.0:25932
18:10:05 [INFO]: Upload size: 599360
Sending invitation to 192.168.1.5
18:10:05 [INFO]: Waiting for device...
18:10:15 [ERROR]: No response from device

everslick, 2018-04-15:
what is the IP address of the maschine you work on? is it in the same
subnet? does it also have an IP from the same router? can you ping the
ESP32 (192.168.1.5) ?

everslick, 2018-04-22:
Last idea: A personal firewall on your windows machine is blocking incoming traffic on the listening port (in this case it is 25932).

csowada, 2018-04-23:
Same here, example is not working. Tried two different esp32 boards. Latest Platformio with stage platform.

williamesp2015, 2018-04-24:
In my case  [ERROR]: No response from device solved after unblock python in my antivirus.

marcelstoer, 2018-05-04:
@williamesp2015 close this issue if it's solved?

closed 2018-05-05 by williamesp2015

tomtobback, 2018-12-05:
Just in case it helps anybody; on my chromebook Linux (crouton chroot) i had to add a host port in /etc/rc.local to allow incoming traffic.
However, using the Arduino IDE, espota.py assigns a random host port. From the command line, i can run espota.py, specifying the host port with -P so that works. To make it work in the Arduino IDE, i edited espota.py to use a fixed port (line 266), the same as i specified in /etc/rc.local - also works well.

riban-bw, 2023-04-19:
@tomtobback said: "...add a host port in /etc/rc.local to allow incoming traffic" but did not describe how to add the port to this file. I can't find a description of how to do this on the Internet. Are you able to describe how you edited rc.local to enable this?