espressif/arduino-esp32 #1318: ISR additionally called after wakeup from light sleep

PRosenb, 2018-04-16:
When configuring wake up from external interrupt and timer, the interrupt service routine of the external interrupt is executed when the CPU wakes up from the timer interrupt.

### Sketch:
```cpp
#define INTERRUPT_PIN 4
volatile unsigned long isrCount = 0;

void isrInterruptPin() {
  noInterrupts();
  isrCount++;
  interrupts();
}

void setup() {
  Serial.begin(115200);

  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_4, LOW);

  pinMode(INTERRUPT_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), isrInterruptPin, FALLING);
}

void loop() {
  noInterrupts();
  unsigned long isrCountLocal = isrCount;
  interrupts();

  Serial.print("isrCount: ");
  Serial.println(isrCountLocal);

  unsigned long durationMs = 5000;
  esp_sleep_enable_timer_wakeup(durationMs * 1000L);
  esp_light_sleep_start();
}
```
Without touching the external interrupt, the output is as follows.
```
isrCount: 0
isrCount: 1
...
```

stickbreaker, 2018-04-16:
@PRosenb out of curiosity, in you ISR code, do a `digitalRead()` of the interrupt pin, I think you are seeing the misconfiguration of `attachInterrupt()`  in my experience, `attachInterrupt()` ignores the `FALLING` directive.  It is implemented a `CHANGE`. This means the ISR is called each time the pin level changes from high to low and back to high.
```c++
#define INTERRUPT_PIN 4
volatile unsigned long isrCount = 0;
volatile unsigned long isrCountHigh = 0;

void isrInterruptPin() {
 // noInterrupts();  this is a null statement on the ESP32
if (digitalRead(INTERRUPT_PIN)) {
  isrCountHigh++;
  }
else { 
  isrCount++;
  }
 // interrupts(); null statement
}
```

Display the counts in the loop.

Chuck.

stickbreaker, 2018-04-17:
I ran into a problem with my i2c interrupt ISR. I was receiving two interrupts per cycle. Once when the event happened, once after the event was cleared. My driver uses the 13 possible interrupt events to control how it responds. I was receiving an interrupt with a status of zero, (no event).  I traced it down to a miss configured interrupt allocation call.  I looked through the `attachInterrupt()` and saw what looked like the same problem. 
Have you connected a pullup resistor to your external int pin. It looks to me that the internal pullup is being disabled during sleep.
Chuck.

PRosenb, 2018-04-17:
I see, good point.
According to the [Sleep Modes](http://esp-idf.readthedocs.io/en/latest/api-reference/system/sleep_modes.html) documentation the pull up resistor should still be active with my configuration.

I attached a 10k resistor from pin 4 to VCC but the behaviour was unchanged.

stickbreaker, 2018-04-17:
So, the pin is reporting a LOW when it is pulled to vcc with a 10k? I would connect a 'scope to that pin and see what is really happening.

PRosenb, 2018-04-18:
You are right, that's actually weird.. Unfortunately, I don't have a 'scope available. Do you have one there?

I added
```c++
  Serial.print("pin: ");
  Serial.print(digitalRead(INTERRUPT_PIN));
```
in `loop()` and then the output is
```
pin: 1, isrCountHigh: 0, isrCountLow: 0
pin: 0, isrCountHigh: 0, isrCountLow: 1
pin: 0, isrCountHigh: 0, isrCountLow: 2
...
```

stickbreaker, 2018-04-18:
@PRosenb I'm confused:
attach a copy of your sketch, I'll measure it 

I think you are describing this [gpio wakeup](http://esp-idf.readthedocs.io/en/latest/api-reference/system/sleep_modes.html#external-wakeup-ext0)

Chuck.

PRosenb, 2018-04-19:
Yes, I was referring to [gpio wakeup](http://esp-idf.readthedocs.io/en/latest/api-reference/system/sleep_modes.html#external-wakeup-ext0) regarding the pull-up configuration. I added the method call `gpio_pullup_en()` as described there but the behaviour does not change. The pin is still not usable as input as long as you do not call `pinMode()` again.

Below is the script. Many thanks for measuring with the 'scope.
```C++
#define INTERRUPT_PIN 4
volatile unsigned long isrCountHigh = 0;
volatile unsigned long isrCountLow = 0;

void isrInterruptPin() {
  if (digitalRead(INTERRUPT_PIN) == HIGH) {
    isrCountHigh++;
  } else {
    isrCountLow++;
  }
}

void setup() {
  Serial.begin(115200);

  esp_sleep_pd_config(ESP_PD_DOMAIN_RTC_PERIPH, ESP_PD_OPTION_ON);
  esp_sleep_enable_ext0_wakeup(GPIO_NUM_4, LOW);
  gpio_pullup_en(GPIO_NUM_4);

  pinMode(INTERRUPT_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), isrInterruptPin, FALLING);
}

void loop() {
  Serial.print("pin: ");
  Serial.print(digitalRead(INTERRUPT_PIN));
  Serial.print(", isrCountHigh: ");
  Serial.print(isrCountHigh);
  Serial.print(", isrCountLow: ");
  Serial.println(isrCountLow);

  unsigned long durationMs = 5000;
  esp_sleep_enable_timer_wakeup(durationMs * 1000L);
  esp_light_sleep_start();

  pinMode(INTERRUPT_PIN, INPUT_PULLUP);
  attachInterrupt(digitalPinToInterrupt(INTERRUPT_PIN), isrInterruptPin, FALLING);
}
```

PRosenb, 2019-08-04:
This issue is still present in the current software version and on hardware revision 1.

closed 2019-10-17 by stale[bot]