espressif/arduino-esp32 #1324: EDB Not Work on ESP32

laercionit, 2018-04-17:
PC: 0x400ecb42: EDB::limit() at D:\Save\Arduino\Arduino\Sketchbook\libraries\EDB\EDB.cpp line 177
EXCVADDR: 0x00000000

Decoding stack results
0x400ecb42: EDB::limit() at D:\Save\Arduino\Arduino\Sketchbook\libraries\EDB\EDB.cpp line 177
0x400d0c5a: recordLimit() at C:\Users\LAERCI~2\AppData\Local\Temp\arduino_modified_sketch_665957/EDB_SDCARD_Optimized.ino line 119
0x400d10bf: setup() at C:\Users\LAERCI~2\AppData\Local\Temp\arduino_modified_sketch_665957/EDB_SDCARD_Optimized.ino line 82
0x400ec367: loopTask(void*) at D:\Save\Arduino\Arduino\hardware\espressif\esp32\cores\esp32\main.cpp line 15

----------------------------- Remove above -----------------------------

### Hardware:
Board:							        ESP32
Core Installation/update date:			Last  update Master 2018/04/16
IDE name:							Arduino IDE
Flash Frequency:				        	80Mhz
Upload Speed:						115200

Description:
The library EDB not work....

### Sketch:
#include "Arduino.h"
#include <EDB.h>
#include <SPI.h>
#include <SD.h>

#define TABLE_SIZE 8192
#define RECORDS_TO_CREATE 10

char* db_name = "/db/edb_test.db";
File dbFile;

struct LogEvent {
    int id;
    int temperature;
}
logEvent;

inline void writer (unsigned long address, const byte* data, unsigned int recsize) {
    dbFile.seek(address);
    dbFile.write(data,recsize);
    dbFile.flush();
}

inline void reader (unsigned long address, byte* data, unsigned int recsize) {
    dbFile.seek(address);
    dbFile.read(data,recsize);
}

EDB db(&writer, &reader);

void setup()
{
    Serial.begin(115200);
    Serial.println(" Extended Database Library + External SD CARD storage demo");
    Serial.println();
    if (!SD.begin()) {
        Serial.println("No SD-card.");
        return;
    }

    // Check dir for db files
    if (!SD.exists("/db")) {
        Serial.println("Dir for Db files does not exist, creating...");
        SD.mkdir("/db");
    }

    if (SD.exists(db_name)) {

        dbFile = SD.open(db_name, FILE_WRITE);

        // Sometimes it wont open at first attempt, espessialy after cold start
        // Let's try one more time
        if (!dbFile) {
            dbFile = SD.open(db_name, FILE_WRITE);
        }

        if (dbFile) {
            Serial.print("Openning current table... ");
            EDB_Status result = db.open(0);
            if (result == EDB_OK) {
                Serial.println("DONE");
            } else {
                Serial.println("ERROR");
                Serial.println("Did not find database in the file " + String(db_name));
                Serial.print("Creating new table... ");
                db.create(0, TABLE_SIZE, (unsigned int)sizeof(logEvent));
                Serial.println("DONE");
                return;
            }
        } else {
            Serial.println("Could not open file " + String(db_name));
            return;
        }
    } else {
        Serial.print("Creating table... ");
        // create table at with starting address 0
        dbFile = SD.open(db_name, FILE_WRITE);
        db.create(0, TABLE_SIZE, (unsigned int)sizeof(logEvent));
        Serial.println("DONE");
    }

    recordLimit();
    countRecords();
    createRecords(RECORDS_TO_CREATE);
    countRecords();
    selectAll();
    deleteOneRecord(RECORDS_TO_CREATE / 2);
    countRecords();
    selectAll();
    appendOneRecord(RECORDS_TO_CREATE + 1);
    countRecords();
    selectAll();
    insertOneRecord(RECORDS_TO_CREATE / 2);
    countRecords();
    selectAll();
    updateOneRecord(RECORDS_TO_CREATE);
    selectAll();
    countRecords();
    deleteAll();
    Serial.println("Use insertRec() and deleteRec() carefully, they can be slow");
    countRecords();
    for (int i = 1; i <= 20; i++) insertOneRecord(1);  // inserting from the beginning gets slower and slower
    countRecords();
    for (int i = 1; i <= 20; i++) deleteOneRecord(1);  // deleting records from the beginning is slower than from the end
    countRecords();

    dbFile.close();
}

void loop()
{
}

// utility functions

void recordLimit()
{
    Serial.print("Record Limit: ");
    Serial.println(db.limit());
}

void deleteOneRecord(int recno)
{
    Serial.print("Deleting recno: ");
    Serial.println(recno);
    db.deleteRec(recno);
}

void deleteAll()
{
    Serial.print("Truncating table... ");
    db.clear();
    Serial.println("DONE");
}

void countRecords()
{
    Serial.print("Record Count: ");
    Serial.println(db.count());
}

void createRecords(int num_recs)
{
    Serial.print("Creating Records... ");
    for (int recno = 1; recno <= num_recs; recno++)
    {
        logEvent.id = recno;
        logEvent.temperature = random(1, 125);
        EDB_Status result = db.appendRec(EDB_REC logEvent);
        if (result != EDB_OK) printError(result);
    }
    Serial.println("DONE");
}

void selectAll()
{
    for (int recno = 1; recno <= db.count(); recno++)
    {
        EDB_Status result = db.readRec(recno, EDB_REC logEvent);
        if (result == EDB_OK)
        {
            Serial.print("Recno: ");
            Serial.print(recno);
            Serial.print(" ID: ");
            Serial.print(logEvent.id);
            Serial.print(" Temp: ");
            Serial.println(logEvent.temperature);
        }
        else printError(result);
    }
}

void updateOneRecord(int recno)
{
    Serial.print("Updating record at recno: ");
    Serial.print(recno);
    Serial.print("... ");
    logEvent.id = 1234;
    logEvent.temperature = 4321;
    EDB_Status result = db.updateRec(recno, EDB_REC logEvent);
    if (result != EDB_OK) printError(result);
    Serial.println("DONE");
}

void insertOneRecord(int recno)
{
    Serial.print("Inserting record at recno: ");
    Serial.print(recno);
    Serial.print("... ");
    logEvent.id = recno;
    logEvent.temperature = random(1, 125);
    EDB_Status result = db.insertRec(recno, EDB_REC logEvent);
    if (result != EDB_OK) printError(result);
    Serial.println("DONE");
}

void appendOneRecord(int id)
{
    Serial.print("Appending record... ");
    logEvent.id = id;
    logEvent.temperature = random(1, 125);
    EDB_Status result = db.appendRec(EDB_REC logEvent);
    if (result != EDB_OK) printError(result);
    Serial.println("DONE");
}

void printError(EDB_Status err)
{
    Serial.print("ERROR: ");
    switch (err)
    {
        case EDB_OUT_OF_RANGE:
            Serial.println("Recno out of range");
            break;
        case EDB_TABLE_FULL:
            Serial.println("Table full");
            break;
        case EDB_OK:
        default:
            Serial.println("OK");
            break;
    }
}

### Debug Messages:
```
 Extended Database Library + External SD CARD storage demo

Openning current table... DONE
Record Limit: Guru Meditation Error: Core  1 panic'ed (IntegerDivideByZero)
. Exception was unhandled.
Core 1 register dump:
PC      : 0x400ecb42  PS      : 0x00060730  A0      : 0x800d0c5d  A1      : 0x3ffca6c0  
A2      : 0x00000000  A3      : 0x3f400ef0  A4      : 0x00000000  A5      : 0x00000000  
A6      : 0x3f401114  A7      : 0x00000000  A8      : 0xfffffff0  A9      : 0x3ffca6a0  
A10     : 0x0000000e  A11     : 0x3f400ef0  A12     : 0x00000004  A13     : 0x0000ff00  
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x0000000c  EXCCAUSE: 0x00000006  
EXCVADDR: 0x00000000  LBEG    : 0x400014fd  LEND    : 0x4000150d  LCOUNT  : 0xfffffffc  

Backtrace: 0x400ecb42:0x3ffca6c0 0x400d0c5a:0x3ffca6e0 0x400d10bf:0x3ffca700 0x400ec367:0x3ffca760

```

laercionit, 2018-04-17:
From what I've analyzed, the problem is that the file is created without the Header,
If you run the program the first time it will work, but the header of the file was not created, running the second time, you will see the error because it can not load the header.

I think it's something related to unsigned long

beegee-tokyo, 2018-04-17:
Not sure which library you are using, I guess [EDB by James Whiddon](https://github.com/jwhiddon/EDB?utm_source=platformio&utm_medium=piohome)
You need to address this issue to the library maintainer, not here.

laercionit, 2018-04-17:
@beegee-tokyo , Sorry.
The Library is: [](https://playground.arduino.cc/Code/ExtendedDatabaseLibrary)

beegee-tokyo, 2018-04-17:
@laercionit which is the same as I said. 

> Download
> 
> https://github.com/jwhiddon/EDB

laercionit, 2018-04-17:
Exact!
This library is fantastic, but unfortunately I could not identify what is happening.
By my tests the code is failing to create the Header of the file.

laercionit, 2018-04-18:
Hello everyone, any suggestion of where the problem might be? I really wanted to participate in the solution, do you need any additional information?

lbernstone, 2018-04-19:
esp32 uses "a" instead of "w+" for append.  Change your FILE_WRITEs to FILE_APPENDs.  The SPIFFS demo worked fine for me.

laercionit, 2018-04-19:
Hello @lbernstone,

Yes, by SPIFFS it works fine, but in this case I used it by SD, by SD I could not get it to work.

I did not understand your orientation to use SD, could you give me an example if it does not bother you?

closed 2019-08-15 by stale[bot]