espressif/arduino-esp32 #1325: RTC DS321

laercionit, 2018-04-17:
The RTCDS3231 works normally, after a few readings the error below occurs.

DEBUG VERBOSE

[W][esp32-hal-i2c.c:280] i2cRead(): Busy Timeout! Addr: 68
[E][esp32-hal-i2c.c:161] i2cWrite(): Busy Timeout! Addr: 68

everslick, 2018-04-17:
try the stickbreaker-i2c branch

laercionit, 2018-04-17:
Hello @everslick 

Thank you for participating in the post.

Yes, I tried using the stickbreaker-i2c version, it did not work.
Alternatively I also tried a master checkout, it also did not work.

The I2C after a few minutes begins to present a problem.

everslick, 2018-04-17:
do you have pullups on SDA and SCL lines to the 3.3V rail?

laercionit, 2018-04-18:
Hello @everslick,

Yes I am with 4.7k pullup connected to 3.3v.
The project works for a few minutes or hours, there is no default to present the problem.

laercionit, 2018-04-18:
I'm going to redo all the cabling, make sure you have no doubt about cabling.
I'm with RTC Cli for ESP32 and Core ESP32 stickbreaker-i2c branch

I will leave the project running all night long to follow if there is still a problem.

laercionit, 2018-04-19:
Hello @everslick
I made the exchange of all I2C cabling, reviewed the Pullups and all the connections.
I'm with RTC Cli for ESP32 and Core ESP32 stickbreaker-i2c branch

The system is running for 6 hours without any locking.

I'm having a question, what's the reason the master version does not work? I am afraid of being limited in the evolutions of CORE.

stickbreaker, 2018-04-19:
@laercionit The stickbreaker-i2c core is the result of 4 months of testing. It is a revision of my repo stickbreaker/arduino-esp32.  The maintainer of this repo @me-no-dev is in the process of converting my base code to match the style of this main branch.  stickbreaker-i2c is a staging branch.   I expect it to be merged into the main branch in the next few weeks.  We are still discussing the appropriate methods to support multi-threading/tasking.  Currently it does not support multiple tasks transacting i2c messages simultaneously.  We are debating how to support independent tasks interleaving communications. Basic Read/Write atomization is not enough to support i2c devices with internal register configurations.  A standard device read sequence requires both a write and a read operation.  If MUTEX synchronization is used in the `read()` and `write()` functions, one task could adjust a devices internal register pointer, then next task could reset it to another value, then the first task issues its `read()` expecting to read data from its where it set the register pointer.  But, the intervening second task interfered.  I am working on a task/thread level queue system where the atomic transaction can be built of multiple `read()` and `write()` calls.  I hope a working proof of concept will be convincing.  

This stickbreaker-i2c branch will probably be incorporated into the main repo with only single thread support. Most Arduino libraries and sketches are not thread safe, so this limitation should not impose any problems.  We expect to eventually achieve thread safe operations.  I am advocating each `Wire()` object be considered a single thread, and requiring the creation of multiple `Wire()` or equivalent objects where each one is single threaded, and the Hardware Abstraction Layer(HAL) is the MUTEX holder.

Chuck.

RudyFiero, 2018-04-20:
> I have a project that is practically all based on I2C, I have the following devices:
> 
> 4x PCA8574
> 2x PCA8574A
> 1x RTC DS3231
> 
> Yes I am with 4.7k pullup connected to 3.3v.
> 

What is the wire length? If this is distributed, with significant wire lengths, then that may be the cause of your problems. Have you looked at the waveform with an oscilloscope?

laercionit, 2018-04-21:
Hello @RudyFiero,

I had mentioned in the previous post that the problem was solved.
I'll take advantage and give some more details that have surprised me:

Usually I2C is not used for wiring larger than 30 cm, using USB type cabling (4-way and shielded) I did tests with 5 meters and without any problem, I just did not test with larger wires because I did not have. It worked very well.

I hope the next version of the core is also stable like this branch (stickbreaker)

RudyFiero, 2018-04-21:
Hi @laercionit ,

I saw that you it was working (for six hours) but I didn't take that as the problem being solved. I try and not read more into what is said but I'm glad to hear that you consider it solved. I use I2C a lot and I was a concerned about it for the ESP32. I have only had a little time to work with the ESP32 so far but I am planning on using it for projects in the future.

I design electronic products (hardware) at work. I was asked to extend the I2C bus for one product so that a radio could be installed on the outside of a metal walled building. Initially all that was required was a 2 meter cable. Later there were requests for distances of 6 meters. The engineering manager made the required cable length and tested his modification and it worked well for him so that is how they started to ship the product. 

Not too long afterwards they were asking me to look into the product because they were getting unreliable operation, transmission failures. I then learned about the cable extension. I told them that the failures would be because of the cable length. But I was told that it worked fin in his test. So I took scope captures of the signal waveforms with different lengths of wire and included that in a report that listed the rise time requirements of the I2C specifications. On those products we use 2k2 pull up resistors on a 5 volt system. (3.3 volt needs lower resistance to achieve the same rise time)

Because they wanted the extended length I had to use I2C bus buffers. The waveforms were in specification. It worked. Everyone was happy, including me. We use the same electrical interface for another control system to connect a slave box to the main panel. When they had a request for an extended length cable, now asking for for 15 meters, they came to me to check and see if it would be OK. Once bitten, twice shy.

I tested with 25 meters and the waveforms, and operation, was still OK. The length ended up being longer than the initial 15 meter request. The distance initially requested was how far the panels were from each other but they did not include the length needed to run up to the ceiling and back down. 

I read quite a number of times where people have extended the lines quite far (without extender buffering) and stated that it worked. And that may be OK for that particular situation but the possibility for corruption is is too high for a commercial product. And as I said, that's what I do.

Sorry for the extended post length. ;)

everslick, 2018-04-21:
@laercionit Please, close the issue.

closed 2018-04-21 by laercionit