espressif/arduino-esp32 #1327: Include md5.h throws compile error #include expects "FILENAME" or <FILENAME>

hreintke, 2018-04-17:
### Hardware:
Board:							Nano32
Core Installation/update date:		Latest git
IDE name:						Sloeber Arduino plugin V4.2

### Description:

When compiling I get an error on md5.h when compiling WebAuthentication.cpp
Error is: md5.h:29:10: error: #include expects "FILENAME" or \<FILENAME\> ->  #include MBEDTLS_CONFIG_FILE
The compile command contains : -DMBEDTLS_CONFIG_FILE="mbedtls/esp_config.h"

Updating within md5.h to #include "mbedtls/esp_config.h" then it works.

Google revealed that it is a combination of the compiler command and the include statement.

### Sketch:
```cpp

#include "esp2.h"
#include "ESPAsyncWebServer.h"

//The setup function is called once at startup of the sketch
void setup()
{
// Add your initialization code here
}

// The loop function is called in an endless loop
void loop()
{
//Add your repeated code here
}
```

william-ferguson-au, 2018-11-05:
The mbedtls files need better handling of the MBEDTLS_CONFIG_FILE define.

In Windows, ```#include MBEDTLS_CONFIG_FILE``` will fail with 
```Error is: md5.h:29:10: error: #include expects "FILENAME" or <FILENAME> -> #include MBEDTLS_CONFIG_FILE```

One solution is to modify each of the mbdetls files with:

```
#if !defined(MBEDTLS_CONFIG_FILE)
#include "config.h"
#else
// From https://github.com/Sloeber/arduino-eclipse-plugin/issues/1028
#define VAL(str) #str
#define TOSTRING(str) VAL(str)
#include TOSTRING(MBEDTLS_CONFIG_FILE)
#endif
```

But I'm sure a more refined solution exists.

See https://github.com/Sloeber/arduino-eclipse-plugin/issues/1028

mikemoretti3, 2019-05-26:
Even with the latest set of files that seem to all contain the above fix, it still happens.  I had to edit the Environment and change the A.COMPILER.CPREPROCESSOR.FLAGS where it says -DMBEDTLS_CONFIG_FILE\="mbedtls/esp_config.h" to -DMBEDTLS_CONFIG_FILE\="""mbedtls/esp_config.h"""   note the triple quotes.

closed 2019-08-14 by stale[bot]

smuellener, 2019-12-17:
Is this issue still open or resolved by now? esp 1.0.3 seems to have this bug for Windows still

atanisoft, 2019-12-17:
@smuellener open an issue against Sloeber if that is the IDE you are using. Also note that 1.0.4 is out already. This issue was never an issue in arduino-esp32 but instead the build system in Sloeber.

madhephaestus, 2020-07-12:
A workaround to this problem can be found in #4159 The workaround will work for both Sloeber and Arduino IDE