espressif/arduino-esp32 #1328: mkspiffs needs updating

JamesGKent, 2018-04-18:
when i tried to preload files into spiffs using the plugin I get an invalid filesystem error from `spiffs.begin`

I downloaded the latest mkspiffs (0.2.3) and replaced my installed version (0.2.2) and the error disappeared.
package_esp32_index.template.json needs updating with the urls, names and checksums of the newer packages.

schweini, 2018-04-23:
have you tried calling it as "spiffs.begin(true)" ? This made some examples that didn't work start working for me.

JamesGKent, 2018-04-23:
i haven't, as stated i updated my local mkspiffs executable and the issue was resolved, i'm not sure why this repo isn't using the latest mkspiffs as it is...

closed 2019-08-15 by stale[bot]