espressif/arduino-esp32 #1330: WiFiMulti does not work

smacyas, 2018-04-18:
Example from lib does not work:
#include <WiFi.h>
#include <WiFiMulti.h>

WiFiMulti wifiMulti;

void setup()
{
    Serial.begin(115200);
    delay(10);

    wifiMulti.addAP("ssid_from_AP_1", "your_password_for_AP_1");
    wifiMulti.addAP("ssid_from_AP_2", "your_password_for_AP_2");
    wifiMulti.addAP("domoton1", "nevotondomoton");

    Serial.println("Connecting Wifi...");
    if(wifiMulti.run() == WL_CONNECTED) {
        Serial.println("");
        Serial.println("WiFi connected");
        Serial.println("IP address: ");
        Serial.println(WiFi.localIP());
    }
}

void loop()
{
    if(wifiMulti.run() != WL_CONNECTED) {
        Serial.println("WiFi not connected!");
        delay(1000);
    }
}

E (302) wifi: esp_wifi_scan_start 967 wifi not start
WiFi not connected!
WiFi not connected!
WiFi not connected!

sticilface, 2018-04-18:
Try this pull and see if it works.
https://github.com/espressif/arduino-esp32/pull/1322

smacyas, 2018-04-19:
Yes, it works, thanks!

closed 2018-04-19 by smacyas

sserena, 2018-04-25:
Hi

I got the same error, but in this line:

`int networksLength = WiFi.scanNetworks();`

Specifically, it says this:

> E (132) wifi: esp_wifi_scan_start 967 wifi not start

I just pulled, but no change.