espressif/arduino-esp32 #1331: Can't get timer count variable out of function

liamt26, 2018-04-18:
### Hardware:
Board:							ESP32 Dev Module
Core Installation/update date:			April/2018
IDE name:							Arduino IDE
Flash Frequency:					         80Mhz
Upload Speed:						 115200

### Description:
I'm working on a project where I'm using two internal clocks to record the times between two interrupts. The ESP interrupts on a rising edge, and I have it lighting an LED so that I can see changes in time.

What I cant seem to do is pull timer1Value out of the function below. 

esp_err_t timer_get_counter_value(timer_group_t TIMER_GROUP_1 , timer_idx_t TIMER_1, uint64_t*timer1Value);

I cant seem to print/save to another variable or doing anything with the timer1Value?

Part of my code is below.

Thanks for all help in advance.  

### Sketch:
```cpp

//Change the code below by your sketch
hw_timer_t * timer0 = NULL;
portMUX_TYPE timerMux0 = portMUX_INITIALIZER_UNLOCKED;
volatile uint8_t led0state = 0;   //unsigned 8 bit int

void IRAM_ATTR count0 ()
{
  portENTER_CRITICAL_ISR(&timerMux0);
  esp_err_t timer_get_counter_value(timer_group_t TIMER_GROUP_0 , timer_idx_t TIMER_0, uint64_t*timer0Value); //saving counter value
  timer0Value = timer0Val;
  Serial.println("Timer 0 Value");
  Serial.println(timer0Value); //hopefully prints timer ticks.

  led0state = 1 - led0state;
  digitalWrite(12, led0state);    //write state change to LED1
  portEXIT_CRITICAL_ISR(&timerMux0);  //exit ISR
}

void setup() {

Serial.begin(115200);
  pinMode (12, OUTPUT);     //LEDs on board to react to chnages in input on next set of digital inputs.
  pinMode (14, OUTPUT);
  digitalWrite(12, LOW);
  digitalWrite(14, LOW);

 Serial.println("Starting timer0");
  timer0 = timerBegin(0, 80, true);  //timer 0, 80 as prescaler value. MWDT clock period 12.5nS * prescaler 
  value (80) = 1000ns = 1ms can change. count up
  esp_err_t timer_start(timer_group_t TIMER_GROUP_0 , timer_idx_t TIMER_0);
  attachInterrupt(digitalPinToInterrupt(26), count0, RISING);
  timerAlarmWrite(timer0, 18000, true); //reset timer afrter 18ms 1uS * 18,000 = 18ms
  timerAlarmEnable(timer0);
}

void loop() {
}
```

### Debug Messages:
```
exit status 1
'timer0Value' was not declared in this scope
```

lbernstone, 2018-04-18:
Is that really an asterisk between the uint64_t and timer0Value or is that a typo?

liamt26, 2018-04-18:
No, not a typo. from the documentation I was looking at, it looks like it should be there.  In any case, it the same error messages comes up with or without it. 

http://esp-idf.readthedocs.io/en/latest/api-reference/peripherals/timer.html

everslick, 2018-04-18:
```esp_err_t timer_get_counter_value(timer_group_t TIMER_GROUP_0 , timer_idx_t TIMER_0, uint64_t*timer0Value); //saving```

this does not make sense at all. you somehow just copied the function prototype into the timer service routine.

you are not calling a function in C this way. Without wanting to sound rude, I suggest visiting a C tutorial first.

liamt26, 2018-04-18:
Nope sounds like solid advice, I'll look into how to call and use C functions properly. Thanks. 
Any tutorial recommendations?

stickbreaker, 2018-04-18:
@liamt26 try [Beginning Arduino Programming](https://www.amazon.com/Beginning-Arduino-Programming-Technology-Action/dp/1430237775/ref=sr_1_1)

Chuck.

stickbreaker, 2018-04-18:
@liamt26 a more advanced [SAMS Teach Youself C++ in one Hour a Day](https://www.amazon.com/gp/offer-listing/0789757745/ref=dp_olp_all_mbc?ie=UTF8&condition=all)

Chuck.

everslick, 2018-04-18:
Please, don't forget to close this issue. When you have a more concrete question or search for help, visit the nice people in the [gitter channel](https://gitter.im/espressif/arduino-esp32).

closed 2018-04-19 by liamt26