espressif/arduino-esp32 #1333: WIFI_PS_MIN_MODEM "not defined" error in Arduino Environment

IraSch, 2018-04-19:
### Hardware:
Board:							Adafruit ESP32 Featuerh
Core Installation/update date:		Don't know where to find this
IDE name:						Arduino IDE
Flash Frequency:					80Mhz
Upload Speed:					921600

### Description:
I'm working on the ESP32 in the Arduino environment. 

There is apparently a function call to set the Wifi Modem to various power saving states via the calls

esp_wifi_set_ps(WIFI_PS_MIN_MODEM); 
esp_wifi_set_ps(WIFI_PS_MAX_MODEM); 

When I try this, WIFI_PS_MIN_MODEM is undefined. I've also searched the ".h" files and can't seem to find a definition anywhere. 

Does anyone have a way to set the MIN powersaving mode for the Wifi component? 

Thanks.

IraSch, 2018-04-19:
OK, I'm confused (nothing new).

On my computer, esp_wifi.h has two definitions: WIFI_PS_NONE and WIFI_PS_MODEM.  On Github, the version has 3 (the expected ones).

I installed the ESP32 package on 3-3-2018. Is this something that changed since 3-3? Doesn't seem like it since the articles that I found which referenced WIFI_PS_MIN_MODEM are prior to 3-3.

copercini, 2018-07-08:
It should work in the last core version https://github.com/espressif/arduino-esp32/blob/c77aed4ac4f343fb6e7e7b9bbaa36b3495eda4cc/tools/sdk/include/esp32/esp_wifi_types.h#L189-L195

@IraSch try to [update your SDK to the last version](https://github.com/espressif/arduino-esp32/blob/master/docs/arduino-ide/windows.md#how-to-update-to-the-latest-code) and try again, if the problem still happens, we can reopen this issue =)

closed 2018-07-08 by copercini