espressif/arduino-esp32 #1334: M5Stack: M5.BtnA.wasPressed() conflicts with WiFi

evanevery, 2018-04-19:
I reported this in the M5Stack community forum and also the M5Stack Github.  Someone else also found this problem and reported that if the IDF was rolled back it went away.  (http://forum.m5stack.com/topic/174/m5-btna-waspressed-conflicts-with-wifi)  So it looks like this is an issue with the current IDF:

It appears that M5.BtnA.wasPressed() will not work properly if the M5Stack is currently connected via WiFi.  It appears that BtnA can not be reliably used if the device is connected via WiFi. The pin appears to become unstable or maybe is being used for some other WiFi related function.

Leave code below as is to see unstable BtnA. Comment out just the three WiFi connect lines to see BtnA working as expected:

----- Code Below ---------------

#include <M5Stack.h>
#include <WiFi.h>

void setup(){
     M5.begin();

     // Comment out the next three lines for Button A to work properly
     WiFi.begin("SSID", "PWD");
     while (WiFi.status() != WL_CONNECTED) {
     }

     M5.Lcd.println("Ready!");
}

void loop() {

     if(M5.BtnA.wasPressed()) M5.Lcd.println("Button A");

     M5.update();
}

reaper7, 2018-04-19:
m5stack button A is connected to GPIO39

https://github.com/m5stack/M5Stack/issues/52

reaper7, 2018-04-19:
it may be important...
not everyone knows what is BtnA

drschlaumeier, 2018-05-06:
I have same problem.
Anyone has already solution?

evanevery, 2018-05-07:
Not so far!

If you use M5Stack WiFi, then you can not use Button A.  Only TWO buttons are usable!

drschlaumeier, 2018-05-07:
Hmm, seems to be in the espressif core and not the libs. There is a similar issue documented. Track down the root cause is difficult...

evanevery, 2018-05-07:
All I can do is lay the problem on the altar and hope someone fixes it...

We only have three buttons and to lose one of them just cause WiFi is being used seems to be a pretty big loss of functionality...

negativekelvin, 2018-05-07:
https://github.com/espressif/esp-idf/issues/1806

drschlaumeier, 2018-05-08:
Thanks for the hint. I understood RTC causes problems and/or ADC1 needs to be disabled. Correct?

drschlaumeier, 2018-05-09:
Hey, took me a while with Try&Error to disable ADC, ISR, DAC etc. until I recognized thats only a shit timing issue.
A smal delay(10) helps and btnA is reading normal !

```
void loop() {
  delay(10);
  if(M5.BtnA.wasPressed()) M5.Lcd.println("Button A");
  M5.update();
}
```

drschlaumeier, 2018-05-09:
sure

evanevery, 2018-05-09:
Yup... Just tested it with my very simple test code and it seemed to mitigate the issue!  Thanks!

Now I need to test it in my real world sketch to see if it helps....

evanevery, 2018-05-09:
Seems to mitigate the issue in my "real world" sketch as well.

Its a simple work-around, but it costs 10ms.  Hopefully it will help someone find the root cause and actually fix it!

Nice!  Thanks!

evanevery, 2018-05-09:
BTW - As I'm finding out - the 10ms delay helps but its not perfect.  I'm still getting intermittant fake ButtonA presses every couple of minutes...

I will try a longer delay...

closed 2019-08-15 by stale[bot]

tialm, 2020-04-24:
So I am having exactly the same problem. I am using an M5 grey unit. You can see with detail what is happening here https://forum.littlevgl.com/t/problems-with-interface-after-connecting-to-network/2130/5
When I turn on the wifi the A button is always being pressed.. If I turn it off it works fine..
M5stack support told me not to use IRQ events. I'll have to test it further.
I wonder if it is a software error or Hardware. And if it is hardware, can it be resolved with another board configuration?
Anyone got any other fix for this issue?

evanevery, 2020-04-24:
This is a very old issue and I honestly don't remember what the solution was.  However, I do seem to remember the solution was identified in this thread.

Although I'm not going to go back and read through this thread all over again (I'll leave that to you), I will tell you that I just loaded up my code that originally caused this issue to see if I embedded any notes to indicate some workaround.  The fact that I haven't left myself some notes on what needed to be done to resolve the issue is indicative that the solution resided somewhere outside my code.  I would suggest you check the dates and versions of your libraries to ensure everything is current.  My application has been running fine since May of 2018 and it doesn't appear that the resolution required any sort of a "workaround" in my code to accomplish this.

Good Luck!

MDAR, 2020-06-29:
Hi

I'm having very similar issues with a M5StickC.

Is there anything I can try that isn't listed above? (As I've tried each one and I'm not seeing any kind of improvement)

Many thanks

electricidea, 2021-11-28:
The problem can be solved with a delay(20); at the end of the loop() function. 10ms was not enough on my system, but 20 does not show any more ghost keystrokes.