espressif/arduino-esp32 #1337: WifiClientSecure Certificate verification failed on SSL/8883 handshakes

joey1442, 2018-04-20:
Board:							ESP32 Dev Module?
Core Installation/update date:			#1296 
IDE name:						Arduino IDE 1.8.5Flash 
Frequency:					        80Mhz
Upload Speed:						921600

### Description:
I am using wificlientsecure to connect to the test MQTT mosquito server and failing. I have taken the cert on the test.mosquitto.org site page and formatted it.  The Mosquito cert is signed.
I cannot see what the issue is , the same code successfully connects to another HTTPS site with their site's signed cert,   could there be problem with the Mbed SSL handling ?. 
I have also had problems in using unsigned certs in the past and gave up in favour of signed for https servers. 
Thanks in advance to anyone who can shed light on this.
### Sketch:
```cpp

//Change the code below by your sketch
#include <WiFiClientSecure.h>

const char* ssid = "xxxxx";
const char* password = "xxxxx";
//String Port = "443"; // HTTPS port
int Port = 8883;  // MQTT port

const char*  server = "test.mosquitto.org";  // Server URL

const char* root_ca = \
"-----BEGIN CERTIFICATE-----\n" \
"MIIC8DCCAlmgAwIBAgIJAOD63PlXjJi8MA0GCSqGSIb3DQEBBQUAMIGQMQswCQYD\n" \
"VQQGEwJHQjEXMBUGA1UECAwOVW5pdGVkIEtpbmdkb20xDjAMBgNVBAcMBURlcmJ5\n" \
"MRIwEAYDVQQKDAlNb3NxdWl0dG8xCzAJBgNVBAsMAkNBMRYwFAYDVQQDDA1tb3Nx\n" \
"dWl0dG8ub3JnMR8wHQYJKoZIhvcNAQkBFhByb2dlckBhdGNob28ub3JnMB4XDTEy\n" \
"MDYyOTIyMTE1OVoXDTIyMDYyNzIyMTE1OVowgZAxCzAJBgNVBAYTAkdCMRcwFQYD\n" \
"VQQIDA5Vbml0ZWQgS2luZ2RvbTEOMAwGA1UEBwwFRGVyYnkxEjAQBgNVBAoMCU1v\n" \
"c3F1aXR0bzELMAkGA1UECwwCQ0ExFjAUBgNVBAMMDW1vc3F1aXR0by5vcmcxHzAd\n" \
"BgkqhkiG9w0BCQEWEHJvZ2VyQGF0Y2hvby5vcmcwgZ8wDQYJKoZIhvcNAQEBBQAD\n" \
"gY0AMIGJAoGBAMYkLmX7SqOT/jJCZoQ1NWdCrr/pq47m3xxyXcI+FLEmwbE3R9vM\n" \
"rE6sRbP2S89pfrCt7iuITXPKycpUcIU0mtcT1OqxGBV2lb6RaOT2gC5pxyGaFJ+h\n" \
"A+GIbdYKO3JprPxSBoRponZJvDGEZuM3N7p3S/lRoi7G5wG5mvUmaE5RAgMBAAGj\n" \
"UDBOMB0GA1UdDgQWBBTad2QneVztIPQzRRGj6ZHKqJTv5jAfBgNVHSMEGDAWgBTa\n" \
"d2QneVztIPQzRRGj6ZHKqJTv5jAMBgNVHRMEBTADAQH/MA0GCSqGSIb3DQEBBQUA\n" \
"A4GBAAqw1rK4NlRUCUBLhEFUQasjP7xfFqlVbE2cRy0Rs4o3KS0JwzQVBwG85xge\n" \
"REyPOFdGdhBY2P1FNRy0MDr6xr+D2ZOwxs63dG1nnAnWZg7qwoLgpZ4fESPD3PkA\n" \
"1ZgKJc2zbSQ9fCPxt2W3mdVav66c6fsb7els2W2Iz7gERJSX\n" \
"-----END CERTIFICATE-----\n";

WiFiClientSecure client;

void setup() {
  //Initialize serial and wait for port to open:
  Serial.begin(115200);
  delay(100);

  Serial.print("Attempting to connect to SSID: ");
  Serial.println(ssid);
  WiFi.begin(ssid, password);

  // attempt to connect to Wifi network:
  while (WiFi.status() != WL_CONNECTED) {
    Serial.print(".");
    // wait 1 second for re-trying
    delay(1000);
  }

  Serial.print("Connected to ");
  Serial.println(ssid);

  client.setCACert(root_ca);
  //client.setCertificate(test_client_key); // for client verification
  //client.setPrivateKey(test_client_cert);  // for client verification

  Serial.println("\nStarting connection to server...");
  if (!client.connect(server, Port))
    Serial.println("Connection failed!");
  else {
    Serial.println("Connected to server!");
    // Make a HTTP request:
  //  client.println("GET https://www.howsmyssl.com/a/check HTTP/1.0");
    //client.println("Host: www.howsmyssl.com");
   // client.println("Connection: close");
    //client.println();

    while (client.connected()) {
      String line = client.readStringUntil('\n');
      if (line == "\r") {
        Serial.println("headers received");
        break;
      }
    }
    // if there are incoming bytes available
    // from the server, read them and print them:
    while (client.available()) {
      char c = client.read();
      Serial.write(c);
    }

    client.stop();
  }
}

void loop() {
  // do nothing
}
```

### Debug Messages:
```
Attempting to connect to SSID: xxxxxx
.[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 13 - AP_START
[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 4 - STA_CONNECTED
..[D][WiFiGeneric.cpp:293] _eventCallback(): Event: 7 - STA_GOT_IP
Connected to BTHub6-2C8G

Starting connection to server...
[V][ssl_client.cpp:48] start_ssl_client(): Free heap before TLS 150860
[V][ssl_client.cpp:50] start_ssl_client(): Starting socket
[V][ssl_client.cpp:86] start_ssl_client(): Seeding the random number generator
[V][ssl_client.cpp:95] start_ssl_client(): Setting up the SSL/TLS structure...
[V][ssl_client.cpp:108] start_ssl_client(): Loading CA cert
[V][ssl_client.cpp:143] start_ssl_client(): Setting hostname for TLS session...
[V][ssl_client.cpp:158] start_ssl_client(): Performing the SSL/TLS handshake...
[E][ssl_client.cpp:28] handle_error(): X509 - Certificate verification failed, e.g. CRL, CA or signature check failed
[E][ssl_client.cpp:30] handle_error(): MbedTLS message code: -9984
[E][WiFiClientSecure.cpp:108] connect(): lwip_connect_r: 11
[V][ssl_client.cpp:209] stop_ssl_socket(): Cleaning SSL connection.
Connection failed!
```

copercini, 2018-04-21:
This looks like a problem in the `test.mosquitto.org` server, because it's not connecting even using a desktop MQTT client like https://github.com/eclipse/paho.mqtt-spy

joey1442, 2018-04-21:
Thanks, I have been trying to connect to another MQTT server also running REST server, I can connect to the REST but not the MQTT. The test with Mosquito was a sanity check that the wifi client was ok , but obviously choose the wrong server to check. I'll find another and try. Thanks for that.

joey1442, 2018-04-21:
I have checked with another public MQTT secure broker and was successful in connecting. The problem was indeed on the test.mosquitto.org server. Thanks Copercini.

closed 2018-04-21 by joey1442