espressif/arduino-esp32 #1339: Reboot loop and I2C - csum error Falling back to built-in command interpreter

syelnmh, 2018-04-20:
### Hardware:
Board:							ESP32 WROOM (on breakout board)
Core Installation/update date:			Two weeks ago (Apr 2018)
IDE name:						Arduino IDE 1.8.5
Flash Frequency:					40Mhz
Upload Speed:						115200

### Description:
I'm trying to use I2C to talk to an ADXL345 accelerometer board, but ESP32 is getting stuck in a reboot loop. The behavior changes depending on whether I have the SDA/SCL lines connected when resetting the ESP32. What could be causing this?
Here's the debug (verbose) serial output of the ESP32 if the I2C lines are CONNECTED at boot:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13268
csum err:0x12!=0xb8
Falling back to built-in command interpreter.
OK
```
The csum errr values do change when you upload different scripts to the board.
After printing that block, the board seems to reboot, then continues printing the same message over and over again.

And here's the output with I2C lines DISCONNECTED at boot:
```
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13268
entry 0x40078a94
Accelerometer Test

[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 53
[W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 53
FF
Ooops, no ADXL345 detected ... Check your wiring!
[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 53
[W][esp32-hal-i2c.c:334] i2cRead(): Ack Error! Addr: 53
[W][esp32-hal-i2c.c:235] i2cWrite(): Ack Error! Addr: 53

```
Neither case is very nice.

Hardware setup:
- external 3v3 regulator supplies power to esp32, caps on supply line
- ADXL345 SDA -> SD0 (pin 21) 
- ADXL345 SCL -> SD1 (pin 22)
- Tried 10k pullups on SDA/SCL to no effect

### Sketch:
(example code for the ADXL345 from Arduino)
```
#include <Wire.h>
#include <Adafruit_Sensor.h>
#include <Adafruit_ADXL345_U.h>

/* Assign a unique ID to this sensor at the same time */
Adafruit_ADXL345_Unified accel = Adafruit_ADXL345_Unified(12345);

void displaySensorDetails(void)
{
  sensor_t sensor;
  accel.getSensor(&sensor);
  Serial.println("------------------------------------");
  Serial.print  ("Sensor:       "); Serial.println(sensor.name);
  Serial.print  ("Driver Ver:   "); Serial.println(sensor.version);
  Serial.print  ("Unique ID:    "); Serial.println(sensor.sensor_id);
  Serial.print  ("Max Value:    "); Serial.print(sensor.max_value); Serial.println(" m/s^2");
  Serial.print  ("Min Value:    "); Serial.print(sensor.min_value); Serial.println(" m/s^2");
  Serial.print  ("Resolution:   "); Serial.print(sensor.resolution); Serial.println(" m/s^2");  
  Serial.println("------------------------------------");
  Serial.println("");
  delay(500);
}

void displayDataRate(void)
{
  Serial.print  ("Data Rate:    "); 
  
  switch(accel.getDataRate())
  {
    case ADXL345_DATARATE_3200_HZ:
      Serial.print  ("3200 "); 
      break;
    case ADXL345_DATARATE_1600_HZ:
      Serial.print  ("1600 "); 
      break;
    case ADXL345_DATARATE_800_HZ:
      Serial.print  ("800 "); 
      break;
    case ADXL345_DATARATE_400_HZ:
      Serial.print  ("400 "); 
      break;
    case ADXL345_DATARATE_200_HZ:
      Serial.print  ("200 "); 
      break;
    case ADXL345_DATARATE_100_HZ:
      Serial.print  ("100 "); 
      break;
    case ADXL345_DATARATE_50_HZ:
      Serial.print  ("50 "); 
      break;
    case ADXL345_DATARATE_25_HZ:
      Serial.print  ("25 "); 
      break;
    case ADXL345_DATARATE_12_5_HZ:
      Serial.print  ("12.5 "); 
      break;
    case ADXL345_DATARATE_6_25HZ:
      Serial.print  ("6.25 "); 
      break;
    case ADXL345_DATARATE_3_13_HZ:
      Serial.print  ("3.13 "); 
      break;
    case ADXL345_DATARATE_1_56_HZ:
      Serial.print  ("1.56 "); 
      break;
    case ADXL345_DATARATE_0_78_HZ:
      Serial.print  ("0.78 "); 
      break;
    case ADXL345_DATARATE_0_39_HZ:
      Serial.print  ("0.39 "); 
      break;
    case ADXL345_DATARATE_0_20_HZ:
      Serial.print  ("0.20 "); 
      break;
    case ADXL345_DATARATE_0_10_HZ:
      Serial.print  ("0.10 "); 
      break;
    default:
      Serial.print  ("???? "); 
      break;
  }  
  Serial.println(" Hz");  
}

void displayRange(void)
{
  Serial.print  ("Range:         +/- "); 
  
  switch(accel.getRange())
  {
    case ADXL345_RANGE_16_G:
      Serial.print  ("16 "); 
      break;
    case ADXL345_RANGE_8_G:
      Serial.print  ("8 "); 
      break;
    case ADXL345_RANGE_4_G:
      Serial.print  ("4 "); 
      break;
    case ADXL345_RANGE_2_G:
      Serial.print  ("2 "); 
      break;
    default:
      Serial.print  ("?? "); 
      break;
  }  
  Serial.println(" g");  
}

void setup(void) 
{
  Serial.begin(115200);
  Serial.println("Accelerometer Test"); Serial.println("");
  Wire.begin(21, 22); // 21 & 22 are default on ESP32
  /* Initialise the sensor */
  if(!accel.begin())
  {
    /* There was a problem detecting the ADXL345 ... check your connections */
    Serial.println("Ooops, no ADXL345 detected ... Check your wiring!");
    delay(500);
  }

  /* Set the range to whatever is appropriate for your project */
  accel.setRange(ADXL345_RANGE_16_G);
  // displaySetRange(ADXL345_RANGE_8_G);
  // displaySetRange(ADXL345_RANGE_4_G);
  // displaySetRange(ADXL345_RANGE_2_G);
  
  /* Display some basic information on this sensor */
  displaySensorDetails();
  
  /* Display additional settings (outside the scope of sensor_t) */
  displayDataRate();
  displayRange();
  Serial.println("");
}

void loop(void) 
{
  /* Get a new sensor event */ 
  sensors_event_t event; 
  accel.getEvent(&event);
 
  /* Display the results (acceleration is measured in m/s^2) */
  Serial.print("X: "); Serial.print(event.acceleration.x); Serial.print("  ");
  Serial.print("Y: "); Serial.print(event.acceleration.y); Serial.print("  ");
  Serial.print("Z: "); Serial.print(event.acceleration.z); Serial.print("  ");Serial.println("m/s^2 ");
  delay(500);
}
```
Thanks.

stickbreaker, 2018-04-20:
@syelnmh try the stickbreaker-i2c branch of this repo, instead of the Main branch.

Chuck.

syelnmh, 2018-04-20:
@stickbreaker No joy, but things changed a bit.

```
>> git branch
* stickbreaker-i2c
```

With I2C lines plugged in on boot, it continues to give this:
```
 rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13268
csum err:0x30!=0xb8
Falling back to built-in command interpreter.
OK
>
```
When I booted the board without the I2C lines plugged in, it does fine for a while (reading 0's in over I2C. Then when plugged the lines in while the ESP32 is still powered on, the brownout detector gets triggered, even though the ADXL345 board is not drawing power from the ESP32, but from a 3V3 regulator off board:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:956
load:0x40078000,len:0
load:0x40078000,len:13268
entry 0x40078a94
Accelerometer Test

0
Ooops, no ADXL345 detected ... Check your wiring!
------------------------------------
Sensor:       ADXL345
Driver Ver:   1
Unique ID:    12345
Max Value:    -156.91 m/s^2
Min Value:    156.91 m/s^2
Resolution:   0.04 m/s^2
------------------------------------

Data Rate:    0.10  Hz
Range:         +/- 2  g

X: 0.00  Y: 0.00  Z: 0.00  m/s^2 
X: 0.00  Y: 0.00  Z: 0.00  m/s^2 
X: 0.00  Y: 0.00  Z: 0.00  m/s^2 

Brownout detector was triggered    <----note: I plugged wires in at this point

ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
```

When you remove the lines again and re-plug them back in:
```
rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
>ets Jun  8 2016 00:22:57

rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK
```
After reflash of code, same deal as above. The thing that changed is that the I2C Ack errors are gone.

I haven't played with the ESP32 for too long, but perhaps the ESP32 wants to access external flash on the I2C lines but is failing?

Is there a way to use the current Adafruit library with different pins for I2C to see if this problem is pin specific?

Thanks.

ElectronicBytes, 2018-04-21:
@stickbreaker 
I think I found the issue. From the datasheet, page 4:

[image]

So I changed the code to 
```
Wire.begin(18,19); //SDA on pin "IO18", SCL on pin "IO19"
```
and I'm getting values. The x and y seem reasonable, but the z is off.
```
X: -2.51 Y: -0.08 Z: 65.55 m/s^2
X: -2.51 Y: -0.08 Z: 65.51 m/s^2
X: -2.51 Y: -0.04 Z: 65.54 m/s^2
X: -2.51 Y: -0.00 Z: 65.59 m/s^2
X: -2.51 Y: -0.04 Z: 65.51 m/s^2
```
I also tried 
```
Wire.begin(4,15);
```
When I pull the I2C wires now, the output is:
```
X: 65.47 Y:65.47 Z:65.47 m/s^2
X: 65.47 Y:65.47 Z:65.47 m/s^2
X: 65.47 Y:65.47 Z:65.47 m/s^2
X: 65.47 Y:65.47 Z:65.47 m/s^2
X: 65.47 Y:65.47 Z:65.47 m/s^2
```
Looks like 65.47 is what the Arduino library returns if no data is received?

Also, it works on pins IO4 and IO15. But Z axis is still messed up. 
But I2C is now resolved. 

Thanks.

stickbreaker, 2018-04-21:
@syelnmh 
Here is a link to a simple I2C scan sketch, try it and see what it reports.
[i2c_scan.ino](https://github.com/stickbreaker/arduino-esp32/tree/master/libraries/Wire/examples/i2c_scan)

If you are using stickbreaker-i2c, then in `setup()` just do a `Wire.begin(SDA,SCL);` to assign pins.  The error message:
>rst:0x10 (RTCWDT_RTC_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
flash read err, 1000
Falling back to built-in command interpreter.
OK

is usually generated when you have some hardware attached to the Flash i/o pins GPIO6 .. GPIO11.

Or, you have one if the strapping io wrong.  Review this page at espressif [strapping ESP32](https://github.com/espressif/esptool/wiki/ESP32-Boot-Mode-Selection)

Chuck.

stickbreaker, 2018-04-21:
@syelnmh I would recommend pullups between 1k to 3.3k  the Weak builtin pullups are way too high for reliable I2C communications above 10khz.. `Wire()` defaults to 100kHz 

That adafruit library might be assuming 16bit ints instead of 32bit.. I would Either review the library or write your own.  

Chuck.

Chuck.