espressif/arduino-esp32 #1342: Add ESP8266 hardware abstraction layer

renne, 2018-04-21:
### Description:
The ESP32 has a CAN bus interface, ethernet interface, touch button support and much more ADCs and GPIOs than the ESP8266. There are tons of ESP8266 applications available while ESP32 applications are rare. Porting ESP8266 applications to ESP32 means a lot of re-coding. An ESP8266 hardware abstraction layer which allows to compile ESP8266 applications for ESP32 without re-coding would simplify the migration path to ESP32 and increase the demand for ESP32 a lot.

So I suggest to develop a hardware abstraction layer for ESP8266 applications or a virtual machine which allows to run ESP8266 applications on ESP32.

lbernstone, 2018-04-21:
That is what arduino is intended to be.  Unfortunately, the reality is that architectures vary between hardware, and many applications are timing and process dependent.  If you have some ideas about how to implement such a thing, particularly in making software designed for a PIC to run on a multithreaded real time OS, we'd love to hear them.

renne, 2018-04-22:
The ESP32 has two CPU-cores. Cutting corners I suggest to run the OS-processes/threads on one CPU-core and the PIC code in a single process exclusively on the other core. The hardware abstraction layer would have to provide headers for the ESP8266 API and conversion functions (e.g. 12-ADC to 10-bit ADC conversion) and some kind of shared memory to hand over data from the ESP32 hardware drivers/API to virtual ESP8266 drivers.

closed 2019-08-15 by stale[bot]