espressif/arduino-esp32 #1343: Timer Interrupt vs Delay

laercionit, 2018-04-22:
Hello everyone, I would like to share a question with you.

I currently have a metering pump that is controlled by the ESP32.
To apply the correct dosage, I use a delay (ms).

This is a big problem for larger dosages, it locks the system for a long time.

Studying a bit on the web, I realized that I could use the Timer Interrupt to avoid locking the system.

Unfortunately I could not implement it.

Can you guide me?

Basically I have a

digitalWrite (PIN, HIGH);
delay (ms);
digitalWrite (PIN, LOW);

ms is a variable in milliseconds, how to adapt to Timer Interrupt?

beegee-tokyo, 2018-04-22:
You could use the Ticker library Ticker.h
Something like
```
#include <Ticker.h>
...

Ticker closeTheValve;

void closeValve() {
  digitalWrite(PIN,LOW);
}

void setup() {
...
}

void loop() {
...
   digitalWrite(PIN, HIGH);
   closeTheValve.once_ms(10,closeValve);
...
}
```

beegee-tokyo, 2018-04-22:
Check this [example](https://github.com/beegee-tokyo/ESP32-MyLib/blob/master/src/myTimer.cpp) I used this code in the past before I switched to Ticker. This code gives you access to uo to 4 hardware timers.

laercionit, 2018-04-22:
Follow my code so I can analyze, sorry but I'm a beginner ....

void DosingStart(byte DosingID) {
  if ( (DosingID < DosingTotal) ) {
    if ((Dosing[DosingID].Remaining) >= (Dosing[DosingID].Amount)) {
      if (DebugCheck(Info)) DebugSerial.println( String(F("DosingStart: Dossing[")) + String(DosingID) + String(F("] >> Started")));
      DosingI2CWriter(DosingID, true);
      **delay((Dosing[DosingID].Rate) * (Dosing[DosingID].Amount));**
      DosingI2CWriter(DosingID, false);
      if (DebugCheck(Info)) DebugSerial.println( String(F("DosingStart: Dossing[")) + String(DosingID) + String(F("] >> Stoped")));
      Dosing[DosingID].Remaining = (  (Dosing[DosingID].Remaining) - (Dosing[DosingID].Amount) );
      Dosing[DosingID].Last.Hour = RTCNowHour();
      Dosing[DosingID].Last.Minute = RTCNowMinute();
      Dosing[DosingID].LastWeekday = RTCNowWeekday();
      Dosing[DosingID].Performed = true;
    } else if (DebugCheck(Error)) DebugSerial.println( String(F("DosingStart: Reservatory is insufficient")));
  } else if (DebugCheck(Error)) DebugSerial.println( String(F("DosingStart: Invalid parameters!")));
}

The bold delay I would like to remove, leaving the program free to run. This delay can be 1 min or hours depending on the dose required by the meter.

beegee-tokyo, 2018-04-22:
If your delay is between 1 minute and several hours, I think the Ticker library is accurate enough and easy to use. 
Beside of that I see at least three things in your code that you need to find a solution for:

1. How many parallel timers do you need? I see the function **`void DosingStart(byte DosingID)`** which indicates that it is called several times with different **`DosingID`**. There are only 4 HW timers available, not sure if this is enough for your requirements. I do not know about limitations of number of Tickers.
2. If several "dosings" are running in parallel, you have to pass the **`DosingID `** into the callback of the HW timer (or Ticker, whatever you decide to use). Neither the HW timers nor the Ticker library directly support this. You might need to define **`DosingTotal`** numbers of callbacks for the Ticker or HW timers.
3. To stop the "dosing" you seem to use an I2C function **`DosingI2CWriter(DosingID, false);`**. I don't think it is wise to call this from within an ISR (interrupt service routine) that would be called from Ticker or a HW timer. But you can try.

If you decide to go with the HW timers and my example, I attached two files that show the principal of use from ArduinoIDE.

[timer-test.zip](https://github.com/espressif/arduino-esp32/files/1935275/timer-test.zip)

laercionit, 2018-04-22:
Hello @ beegee-tokyo, first of all thanks for the willingness and willingness to help.

I currently have 6 dosing machines, each dispenser can only be run once, two dosing machines never run at the same time, the method does not allow it.

Is a single timer enough?

Basically it would be a count down.

What I need is to spend a unit of time in ms and the interruption stops the meter when the time is zero.

beegee-tokyo, 2018-04-22:
@laercionit , If you need only one dosing machine at a time and if (as you said earlier) your activation time is in the range of minutes to a few hours, what is the accuracy you need for switching off the dosing machine? Is a few seconds variance acceptable?
If a few seconds/milliseconds are ok, I would go for a Ticker solution. What I said earlier was actually wrong, you can start a Ticker and give the called function an argument, which would be in your case the DosingID. Check the [arguments example](https://github.com/espressif/arduino-esp32/blob/master/libraries/Ticker/examples/Arguments/Arguments.ino) how to do it.

laercionit, 2018-04-22:
I got it! Thanks guys, now I need to create some functions to prevent more than one meter being triggered at the same time.

void DosingStop(byte DosingID) {
  if (DebugCheck(Info)) DebugSerial.println( String(F("DosingStop: Dossing[")) + String(DosingID) + String(F("] >> Stoped")));
  DosingI2CWriter(DosingID, LOW);
  DosingTicker.detach();
}

void DosingStart(byte DosingID) {
  if ( (DosingID < DosingTotal) ) {
    if ((Dosing[DosingID].Remaining) >= (Dosing[DosingID].Amount)) {
      if (DebugCheck(Info)) DebugSerial.println( String(F("DosingStart: Dossing[")) + String(DosingID) + String(F("] >> Started")));
      DosingI2CWriter(DosingID, HIGH);
      DosingTicker.once_ms((Dosing[DosingID].Rate) * (Dosing[DosingID].Amount), DosingStop, DosingID);
      Dosing[DosingID].Remaining = (  (Dosing[DosingID].Remaining) - (Dosing[DosingID].Amount) );
      Dosing[DosingID].Last.Hour = RTCNowHour();
      Dosing[DosingID].Last.Minute = RTCNowMinute();
      Dosing[DosingID].LastWeekday = RTCNowWeekday();
      Dosing[DosingID].Performed = true;
      //DosingSaveSettings(DosingID) ;
    } else if (DebugCheck(Error)) DebugSerial.println( String(F("DosingStart: Reservatory is insufficient")));
  } else if (DebugCheck(Error)) DebugSerial.println( String(F("DosingStart: Invalid parameters!")));
}

laercionit, 2018-04-22:
@beegee-tokyo 

Unfortunately this function is causing problems in I2C. Randomly generates communication failure with I2C.

lbernstone, 2018-04-22:
I'm a little late on this, but I think the Arguments.ino in the Ticker library is actually a very good example that shows how you could turn on and off each pump.  Rather than using the argument for on/off, you could send the pump number to turn off.

beegee-tokyo, 2018-04-23:
@laercionit are you using the Arduino-ESP32 I2C as it is or are you using @stickbreaker improvements? Many people reported issues with I2C, but there is a solution. Search the issues for I2C problems. And in the Wiki I wrote how to apply the [I2C patch](https://desire.giesecke.tk/index.php/2018/04/20/how-to-use-stickbreakers-i2c-improved-code/)

laercionit, 2018-04-23:
Hello @beegee-tokyo beegee-tokyo ...

I'm using the ESP32 Wroom board through the Arduino IDE with the Core @stickbreaker.
This was requested by some forum members.
Apparently this branch has more stability for I2C.

In this post they mentioned this: https://github.com/espressif/arduino-esp32/issues/1325

Branch Core: https://github.com/espressif/arduino-esp32/tree/stickbreaker-i2c

copercini, 2018-07-01:
The I2C core was changed to @stickbreaker code officially after https://github.com/espressif/arduino-esp32/commit/13dcfe5c13bc882f8923d137ed604aa65b6db057 (thanks @stickbreaker), Let's close issues with old code and if have problems with the new code, open new issues =)

closed 2018-07-01 by copercini